espressif/esp-idf #16778: Issue : External 32.768 kHz crystal oscillator

gajananmaske1234, 2025-07-11:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

[image]
[image]

Hello ESP32 community, 

I'm facing an issue with an external 32.768 kHz crystal oscillator on an ESP32-WROOM 32EU board. Upon booting, I consistently receive the following warning in the serial monitor: W (1735) clk: 32 kHz XTAL not found, switching to internal 150 kHz oscillator .

This suggests the ESP32 is unable to detect or start the external 32kHz crystal. I'm using a YXC 447GO 32.768 kHz through-hole crystal connected to the RTC slow clock pins (32K_XP and 32K_XN). The crystal is connected as follows: Load capacitors: 15 pF, 50V ceramic capacitors on both sides. Capacitor to GND from each crystal pin. Crystal is placed close to the ESP32 module to reduce parasitic capacitance. Here is my crystal connection diagram as given below: Troubleshooting Done Verified the soldering – all connections are intact. Checked capacitor values (15pF, 50V). Tried replacing the crystal with another identical part – same issue. Confirmed correct crystal orientation (although not polarized). Crystal is not shorted or open as per multimeter check. Additional Info: ESP-IDF version: v5.3.2 Project boot log indicates fallback to internal oscillator. Application works, but I need the 32kHz XTAL for accurate RTC and low-power sleep modes. 

Question: 

Are there any additional steps or configurations in ESP-IDF needed to enable external 32kHz XTAL?

 Does this warning indicate a hardware failure, or could it be a software initialization issue? 
Could capacitor values or layout affect startup of the crystal? 

Appreciate any guidance or experience anyone can share..!

songruo, 2025-07-16:
Hi @gajananmaske1234, are you missing a parallel resistor? See reference: https://docs.espressif.com/projects/esp-hardware-design-guidelines/en/latest/esp32/schematic-checklist.html#rtc-clock-source-optional

Alvin1Zhang, 2025-07-30:
@gajananmaske1234 Thanks for reporting, would you please help share if any further updates? Thanks.

Alvin1Zhang, 2025-08-22:
Thanks for reporting, will close due to inactivity, feel free to reopen.

closed 2025-08-22 by Alvin1Zhang