espressif/esp-idf #16779: Minimal MIPI-CSI example hangs on second call to esp_cam_ctlr_receive() (by @Bruce297)

brianmichalk1, 2025-07-11:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

@Bruce297 
I have taken a minimal approach using your example at:
examples/peripherals/camera/mipi_isp_dsi/main/mipi_isp_dsi_main.c
Hardware:
Waveshare P4 Nano
Raspberry Pi Camera B
ESP-IDF master branch as of July 11.

The second call to esp_cam_ctlr_receive() hangs, but with debug statements added to the callbacks, I can see them running up until I place the second call.  I have a suspicion that the display interface does something with the frame buffer, but I don't have a display.
Here's my example code:
`
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "sdkconfig.h"
#include "esp_attr.h"
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "esp_lcd_mipi_dsi.h"
#include "esp_lcd_panel_ops.h"
#include "esp_ldo_regulator.h"
#include "esp_cache.h"
#include "driver/i2c_master.h"
#include "driver/isp.h"
#include "esp_cam_ctlr_csi.h"
#include "esp_cam_ctlr.h"
#include "example_sensor_init.h"
#include "example_config.h"

static const char *TAG = "simple_mipi";
void *frame_buffer;

static bool IRAM_ATTR s_camera_get_new_vb(esp_cam_ctlr_handle_t handle, esp_cam_ctlr_trans_t *trans, void *user_data){
    esp_cam_ctlr_trans_t new_trans = *(esp_cam_ctlr_trans_t *)user_data;
    trans->buffer = new_trans.buffer;
    trans->buflen = new_trans.buflen;
    return false;
}
static bool IRAM_ATTR s_camera_get_finished_trans(esp_cam_ctlr_handle_t handle, esp_cam_ctlr_trans_t *trans, void *user_data){
    return false;
}
void app_main(void){
    ESP_LOGI(TAG, "");
    esp_log_set_level_master(ESP_LOG_VERBOSE);
    esp_log_level_set("*", ESP_LOG_VERBOSE);
    esp_cam_ctlr_handle_t handle = NULL;
    esp_cam_ctlr_csi_config_t csi_config = {
        .ctlr_id = 0,
        .h_res = CONFIG_EXAMPLE_MIPI_CSI_DISP_HRES,
        .v_res = CONFIG_EXAMPLE_MIPI_CSI_DISP_VRES,
        .lane_bit_rate_mbps = EXAMPLE_MIPI_CSI_LANE_BITRATE_MBPS,
        .input_data_color_type = CAM_CTLR_COLOR_RAW8,
        .output_data_color_type = CAM_CTLR_COLOR_RGB565,
        .data_lane_num = 2,
        .byte_swap_en = false,
        .queue_items = 1,
    };
    esp_cam_ctlr_evt_cbs_t cbs = {
        .on_get_new_trans = s_camera_get_new_vb,
        .on_trans_finished = s_camera_get_finished_trans,
    };
    size_t frame_buffer_size = CONFIG_EXAMPLE_MIPI_CSI_DISP_HRES * CONFIG_EXAMPLE_MIPI_CSI_DISP_VRES * EXAMPLE_RGB565_BITS_PER_PIXEL / 8;
    frame_buffer = (uint8_t *)malloc(frame_buffer_size);
    if (frame_buffer == NULL) {
        ESP_LOGE(TAG, "Error allocating frame buffer");
    }
    esp_cam_ctlr_trans_t new_trans = {
        .buffer = frame_buffer,
        .buflen = frame_buffer_size,
    };
    ESP_ERROR_CHECK(esp_cam_new_csi_ctlr(&csi_config, &handle));
    ESP_ERROR_CHECK(esp_cam_ctlr_register_event_callbacks(handle, &cbs, &new_trans));
    ESP_ERROR_CHECK(esp_cam_ctlr_enable(handle));
    ESP_ERROR_CHECK(esp_cam_ctlr_start(handle));
    ESP_LOGI("0", "");
    ESP_ERROR_CHECK(esp_cam_ctlr_receive(handle, &new_trans, ESP_CAM_CTLR_MAX_DELAY));ESP_LOGI("1", "");vTaskDelay(100 / portTICK_PERIOD_MS);
    // This never executes
    ESP_ERROR_CHECK(esp_cam_ctlr_receive(handle, &new_trans, ESP_CAM_CTLR_MAX_DELAY));ESP_LOGI("2", "");vTaskDelay(100 / portTICK_PERIOD_MS);
}
`

Bruce297, 2025-07-15:
Hi @brianmichalk1, Your `esp_cam_ctlr_csi_config_t.queue_items` is 1, that means the queue only hold 1 transaction :  Only one transaction can wait for processing at a time. 

The print been stuck, but the camera and display should be worked.

GingerLukas, 2025-09-20:
I had a look at the source and it seems to me that all the examples that uses receive and callback at the same time are wrong. I say this because in the CSI driver there is this if/else if which causes the queue to fill up and never be emptied if a callback is set.
 https://github.com/espressif/esp-idf/blob/master/components%2Fesp_driver_cam%2Fcsi%2Fsrc%2Fesp_cam_ctlr_csi.c#L341

So it's clear that only one method, either setting a callback or calling the receive function in a loop should be used. At least with the current version of the driver.

jdsalmonson, 2025-09-20:
I just confronted this issue.  The OP's intuition that the display interface in the [example code](https://github.com/espressif/esp-idf/tree/v5.5.1/examples/peripherals/camera/mipi_isp_dsi) is influencing the camera's frame buffer is spot on.  I was able to get my OV5647 camera to work in the camera port of my esp32P4-nano without a display by adding a call to `esp_cam_ctlr_get_frame_buffer()` after starting the controller with 'esp_cam_ctlr_start()' and before the `esp_cam_ctlr_receive()` call.  Internally, this sets `bk_buffer_exposed = true` and so the camera controller thinks an external system is consuming frames (even though i don't have an external system such as a display).

Taking the end of the OP's example code from above, this is how I would implement my fix.  (I didn't test this particular example, but my version is functionally very similar.)
```
    // ...
    ESP_ERROR_CHECK(esp_cam_ctlr_start(handle));

    // Add these lines to enable continuous callbacks:
    const void *dummy_buffer = NULL;
    ESP_ERROR_CHECK(esp_cam_ctlr_get_frame_buffer(handle, 1, &dummy_buffer));

    ESP_LOGI("0", "");
    ESP_ERROR_CHECK(esp_cam_ctlr_receive(handle, &new_trans, ESP_CAM_CTLR_MAX_DELAY));
}
 ```

Note that `esp_cam_ctlr_receive()` is blocking and does not return; the callbacks configured above handle everything.  Also note that I am otherwise following Espressif's example: implementing both the callbacks and the `esp_cam_ctlr_receive()` call.

There are likely other ways to do this, but this is a pretty simple fix.  The documentation should certainly explain how to configure a camera w/o a display.

ghpille, 2025-09-21:
I tried jds' advice, but: `
E (2069) CSI: s_csi_ctlr_get_internal_buffer(285): driver don't initialized or back_buffer not available
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4000ab38
--- 0x4000ab38: app_main at /home/ghp/ESP/mipi_isp_dsi/main/mipi_isp_dsi_main.c:175
file: "./main/mipi_isp_dsi_main.c" line 175
func: app_main
expression: esp_cam_ctlr_get_frame_buffer(cam_handle, 1, &dummy_buffer)
`

ghpille, 2025-09-24:
Hi Jay, about the the code you highlighted, did you think I took it out?  I didn't, but still got the error above.  What a PITA.  Couple of days ago, I received a pair of esp32-p4-wifi6 and a pair of esp32-s3 wroom.  Couldn't start testing with the P4's, the MIPI connectors are of a smaller type, should get some 22-15 cables today.  With the S3's, esp-idf is unaware that there are S3's with a MIPI connector.  So I returned to the arduino-ide with which I have a lot more experience.  Two hours later I had a working piece of code, "fiat lux", as they say.  Most of the two hours was spent learning about multitasking on the esp32's.  I ordered the P4's to make sure nothing was wrong with the esp32-p4-nano, ie. if the code I have would work on the new ones, there was something amiss with the nano.  In the mean time that turned out to be not the case.  By switching from userptr memory buffers to mmapped, everything is as it should be.  So there may be a problem with the driver code when using userptr.  I said that esp-idf doesn't know about S3 with MIPI, but arduino-ide doesn't know about P4's with a camera.  Such is life.

ghpille, 2025-09-24:
Lukáš, thanks for the pointer to the c-code, it may allow me to understand what is expected from the callbacks.

AfoxSa1d, 2025-12-05:
Bro, you only need to call `ESP_ERROR_CHECK(esp_cam_ctlr_receive(handle, &new_trans, ESP_CAM_CTLR_MAX_DELAY));` once. Then in the loop, you can read `new_trans.buffer` in real time. The data stored in `new_trans.buffer` is the binary data of RGB565 or other formats output by the camera through CSI-ISP, and you can use them.

zzy1099207684, 2025-12-11:
#### If you use 'esp_cam_ctlr_receive', you don't need to implement 's_camera_get_new_vb'. If you use 's_camera_get_new_vb' and 's_camera_get_finished_trans', you don't need 'esp_cam_ctlr_receive', as they conflict.

#### Yes, in your code, remove the two `esp_cam_ctlr_receive(handle, &new_trans, ESP_CAM_CTLR_MAX_DELAY)` calls, and then use `new_trans.buffer` to get the data.

simeondmr, 2025-12-15:
Hello, I have the same issue.
@zzy1099207684 I removed the esp_cam_ctlr_receive(handle, &new_trans, ESP_CAM_CTLR_MAX_DELAY), but the callback s_camera_get_new_vb is only called once.
How can I fix this?