espressif/esp-idf #16780: WiFi Mesh stop and restart causes <IP_EVENT_STA_GOT_IP> intermittantly not to fire

michaelsimp, 2025-07-12:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am having a problem where after I restart WiFi and WiFi Mesh, the <MESH_EVENT_PARENT_CONNECTED> event fires but the <IP_EVENT_STA_GOT_IP> intermittently does not.  I use this event to start MQTT etc so it is important that if fires or I need some other trigger for this.

On ESP32 first startup, on WiFiMesh connect to parent the following events ALWAYS fire <MESH_EVENT_PARENT_CONNECTED> and then <IP_EVENT_STA_GOT_IP>

I have a requirement to be able to support a backup WiFi Access Point. If the primary AP goes down and my MESH_ROOT gets a <MESH_EVENT_PARENT_DISCONNECTED> event, it uses this as a trigger where outside the mesh_event_handler I call the following function to stop WiFi and restart it with a different WiFi AP.

```
void wifiBothRestart() {
    parentDisconnectCount = 0;
    wifiMeshStop(true);
    wifiStop();
    sleepMs(5000); // delay to try and ensure <IP_EVENT_STA_GOT_IP> fires
    wifiStart(false);
    wifiMeshStart(false);
}

void wifiMeshStop(bool destroy) {
    wifiMeshStarted = false;
    MQTTDisconnect();
    wifiMeshStatus = Wifi_Mesh_Status::Stopping; // temporary work around to stop mesh breaking by complete Mesh restart
    ESP_LOGW(TAG, "Wifi Mesh Stopping");
    // mesh_netifs_stop();
    connectionStatusLed(Provisioned); // update LED flash pattern
    // ESP_ERROR_CHECK(esp_event_handler_unregister(MESH_EVENT, ESP_EVENT_ANY_ID, &mesh_event_handler));
    ESP_LOGW(TAG, "pre mesh_netifs_destroy");
    if (destroy) ESP_ERROR_CHECK(mesh_netifs_destroy());
    ESP_ERROR_CHECK(esp_mesh_stop());
    ESP_ERROR_CHECK(esp_mesh_deinit());
    ESP_LOGW(TAG, "Wifi Mesh Stopped");
}

void wifiStop() {
    connectionStatusLed(Provisioned); // update LED flash pattern
    ESP_LOGI(TAG, "Stopping WiFi");
    esp_wifi_stop();
    // ESP_ERROR_CHECK(esp_event_handler_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, &ip_event_handler));
    // ESP_ERROR_CHECK(esp_event_loop_delete_default()); //  event initialization
    // esp_wifi_deinit();
    // esp_netif_deinit();
}

void wifiStart(bool firstStart) {
    ESP_LOGI(TAG, "Starting WiFi");
    connectionStatusLed(Provisioned); // update LED flash pattern

    // ESP_ERROR_CHECK(esp_netif_init()); //  tcpip initialization
    // ESP_ERROR_CHECK(esp_event_loop_create_default()); //  event initialization
    //  create network interfaces for mesh (only station instance saved for further manipulation, soft AP instance ignored

    wifi_init_config_t config = WIFI_INIT_CONFIG_DEFAULT(); //  wifi initialization
    ESP_ERROR_CHECK(esp_wifi_init(&config));

    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &ip_event_handler, NULL));

    // ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_FLASH)); // don't overwrite provisioning data for MESH_ROOT to Access Point
    ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM)); // so instead write to RAM
    ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_NONE));
    ESP_ERROR_CHECK(esp_wifi_start());
}

void wifiMeshStart(bool firstStart) {
    wifiMeshStarted = true;
    wifiMeshStatus = Wifi_Mesh_Status::Restarting;
    ESP_LOGW(TAG, "WiFi Mesh Start");

    /*  mesh initialization */
    ESP_LOGW(TAG, "Wifi pre esp_mesh_init");
    ESP_ERROR_CHECK(esp_mesh_init());
    ESP_ERROR_CHECK(esp_event_handler_register(MESH_EVENT, ESP_EVENT_ANY_ID, &mesh_event_handler, NULL));
    ESP_ERROR_CHECK(esp_mesh_set_max_layer(4)); //(CONFIG_MESH_MAX_LAYER));
    ESP_ERROR_CHECK(esp_mesh_set_vote_percentage(1));
    ESP_ERROR_CHECK(esp_mesh_set_ap_assoc_expire(10));
    /* set blocking time of esp_mesh_send() to 30s, to prevent the esp_mesh_send() from permanently for some reason */
    ESP_ERROR_CHECK(esp_mesh_send_block_time(5000)); // was 30 seconds
    mesh_cfg_t cfg = MESH_INIT_CONFIG_DEFAULT();
#if !MESH_IE_ENCRYPTED
    cfg.crypto_funcs = NULL;
#endif
    /* mesh ID */
    memcpy((uint8_t *) &cfg.mesh_id, MESH_ID, MAC_SIZE);
    /* router */
    cfg.channel = CONFIG_MESH_CHANNEL;

    // here I use apIndex to switch between my Primary WiFi AP ssid and the backup AP which is ssid with a "1" appended to the end
    cfg.router.ssid_len = strlen(meshProvisionData.ssid);
    memcpy((uint8_t *) &cfg.router.ssid, meshProvisionData.ssid, cfg.router.ssid_len);
    memcpy((uint8_t *) &cfg.router.password, meshProvisionData.password, strlen(meshProvisionData.password));
    if (apIndex > 0) {
        cfg.router.ssid[cfg.router.ssid_len] = 0x31 + apIndex;
        cfg.router.ssid_len++;
        cfg.router.ssid[cfg.router.ssid_len] = 0x00;
    }
    memcpy((uint8_t *) &ssid, &cfg.router.ssid, cfg.router.ssid_len+1);
    ESP_LOGI(TAG, "Connecting to SSID %d: %s", 1+apIndex, ssid);
    ESP_ERROR_CHECK(esp_mesh_set_ap_authmode((wifi_auth_mode_t) CONFIG_MESH_AP_AUTHMODE));
    cfg.mesh_ap.max_connection = CONFIG_MESH_AP_CONNECTIONS;
    cfg.mesh_ap.nonmesh_max_connection = CONFIG_MESH_NON_MESH_AP_CONNECTIONS;
    memcpy((uint8_t *) &cfg.mesh_ap.password, CONFIG_MESH_AP_PASSWD, strlen(CONFIG_MESH_AP_PASSWD));
    ESP_LOGW(TAG, "Wifi pre esp_mesh_set_config");
    ESP_ERROR_CHECK(esp_mesh_set_config(&cfg));
    /* mesh start */
    esp_mesh_allow_root_conflicts(false); // MBS added 29 Oct 2024
    ESP_LOGW(TAG, "Wifi pre esp_mesh_start");
    ESP_ERROR_CHECK(esp_mesh_start());

    esp_read_mac(myMac, ESP_MAC_WIFI_STA);
    ESP_LOGI(TAG, "This node MAC:" MACSTR "", MAC2STR(myMac));
}

```
See comms trace attached with following points of interest:

```
WiFi AP powered off
I (16:24:38.879) aWifiMesh: <MESH_EVENT_PARENT_DISCONNECTED>reason:200
..2 seconds later...
W (16:24:40.706) aWifiMesh: Switching to AP 2:
W (16:24:40.710) aWifiMesh: Wifi Mesh Stopping
I (16:24:42.964) aWifiMesh: Stopping WiFi
... 5 second delay...
I (16:24:47.966) aWifiMesh: Starting WiFi
W (16:24:47.980) aWifiMesh: WiFi Mesh Start
I (16:24:52.451) aWifiMesh: <MESH_EVENT_FIND_NETWORK>new channel:7, router BSSID:00:00:00:00:00:00
I (16:24:55.966) aWifiMesh: <MESH_EVENT_PARENT_CONNECTED>layer:1-->1, parent:00:17:13:21:3f:f0<ROOT>, ID:77:77:77:77:77:76
I (16:24:55.971) aWifiMesh: <MESH_EVENT_TODS_REACHABLE>state:0
I (16:24:55.973) aWifiMesh: <MESH_EVENT_ROOT_ADDRESS>root address:24:58:7c:de:8e:e9
E (16:25:06.241) aWifiMesh: ip_event_handler did not fire. Rebooting...

```

I think my understanding of the correct procedure for stopping and restarting WiFi and WiFI Mesh is potentially my problem. Can you confirm the correct proceedure for this please.

I have tried commenting in and out the following lines to deal with this but I still have the problem.  
```
// ESP_ERROR_CHECK(esp_event_handler_unregister(MESH_EVENT, ESP_EVENT_ANY_ID, &mesh_event_handler));
// mesh_netifs_stop(); 
// ESP_ERROR_CHECK(esp_event_handler_unregister(MESH_EVENT, ESP_EVENT_ANY_ID, &mesh_event_handler));

// ESP_ERROR_CHECK(esp_netif_init()); //  tcpip initialization
// ESP_ERROR_CHECK(esp_event_loop_create_default()); //  event initialization

```

[RestartTrace.txt](https://github.com/user-attachments/files/21192974/RestartTrace.txt)

michaelsimp, 2025-07-15:
_Do you mean that when the root node disconnects from AP, it will try to connect to another AP, but it didn't have the <IP_EVENT_STA_GOT_IP event?_ Yes correct

I added the diagnostic to .../components/esp_netif/esp_netif_handlers.c#L49
```
        ESP_LOGW(TAG, "Pre calling esp_netif_dhcpc_start(esp_netif)"); // MBS
        esp_netif_dhcpc_start(esp_netif);
        ESP_LOGI(TAG, "Post calling esp_netif_dhcpc_start(esp_netif)"); // MBS
```
I ran a test run powering off AP VONETS2 and switching to AP SSID VONETS
Here is the coms trace showing the end of the scan on the attempt to connect to SSID VONETS.
It shows events <MESH_EVENT_PARENT_CONNECTED>, <MESH_EVENT_TODS_REACHABLE, MESH_EVENT_ROOT_ADDRESS> and MESH_NODES being added <MESH_EVENT_ROUTING_TABLE_ADD>
But no <IP_EVENT_STA_GOT_IP> and my diagnostics didn't print

```
I (749382) mesh: [SCAN][ch:1]AP:2, other(ID:0, RD:0), MAP:0, idle:0, candidate:1, root:0, topMAP:0[c:0,i:1][00:17:13:20:bd:74]router found<>
I (749382) mesh: 1328[SCAN]init rc[48:ca:43:9b:53:d9,-27], mine:0, voter:0
I (749382) mesh: [SCAN:9/10]rc[128][48:ca:43:9b:53:d9,-27], self[48:ca:43:9b:53:d8,-27,reason:0,votes:1,idle][mine:1,voter:1(1.00)percent:1.00][128,1,48:ca:43:9b:53:d9]

I (749692) mesh: [SCAN][ch:1]AP:2, other(ID:0, RD:0), MAP:0, idle:0, candidate:1, root:0, topMAP:0[c:0,i:1][00:17:13:20:bd:74]router found<>
I (749692) mesh: 1328[SCAN]init rc[48:ca:43:9b:53:d9,-27], mine:0, voter:0
I (749692) mesh: [SCAN:10/10]rc[128][48:ca:43:9b:53:d9,-27], self[48:ca:43:9b:53:d8,-27,reason:0,votes:1,idle][mine:1,voter:1(1.00)percent:1.00][128,1,48:ca:43:9b:53:d9]

I (749702) mesh: [DONE]connect to router:VONETS, channel:1, rssi:-27, 00:17:13:20:bd:74[layer:0, assoc:0], my_vote_num:1/voter_num:1, rc[48:ca:43:9b:53:d9/-27/1]
I (749712) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1, snd_ch_cfg:0x0
I (749712) wifi:state: init -> auth (0xb0)
I (749712) wifi:state: auth -> assoc (0x0)
I (749722) wifi:state: assoc -> run (0x10)
I (749992) wifi:<ba-add>idx:0 (ifx:0, 00:17:13:20:bd:74), tid:0, ssn:2, winSize:64
I (749992) wifi:connected with VONETS, aid = 1, channel 1, 40U, bssid = 00:17:13:20:bd:74
I (749992) wifi:security: WPA2-PSK, phy: bgn, rssi: -29
I (750002) wifi:pm start, type: 0

I (750002) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
I (750002) mesh: [scan]new scanning time:600ms, beacon interval:300ms
I (750002) mesh: <flush>upstream packets, connections(max):6, waiting:0, upQ:0
I (16:33:49.883) aWifiMesh: <MESH_EVENT_PARENT_CONNECTED>layer:1-->1, parent:00:17:13:20:bd:74<ROOT>, ID:77:77:77:77:77:76
I (750002) mesh: <flush>root
I (16:33:49.886) mesh_netif: Already wifi station, no need to do anything
I (750012) mesh: [TXQ]<max:32>up(0, be:0), down(0, be:0), mgmt:0, xon(req:0, rsp:0), bcast:0, wnd(0, parent:00:00:00:00:00:00)
I (16:33:49.889) aWifiMesh: <MESH_EVENT_TODS_REACHABLE>state:0
I (750012) mesh: [RXQ]<max:32 = cfg:32 + extra:0>self:0, <max:32 = cfg:32 + extra:0>tods:0
I (16:33:49.891) aWifiMesh: <MESH_EVENT_ROOT_ADDRESS>root address:48:ca:43:9b:53:d9
I (750032) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (758442) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1, snd_ch_cfg:0x0
I (758442) wifi:station: 48:ca:43:9b:5d:20 join, AID=1, bgn, 40U
I (758452) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1, snd_ch_cfg:0x0
I (758452) wifi:station: 48:ca:43:9b:5c:68 join, AID=2, bgn, 40U
I (758462) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1, snd_ch_cfg:0x0
I (758462) wifi:station: 94:a9:90:0d:7f:14 join, AID=3, bgn, 40U
I (16:33:58.345) aWifiMesh: unknown id:25
I (16:33:58.346) aWifiMesh: unknown id:25
I (758472) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1, snd_ch_cfg:0x0
I (758472) wifi:station: 24:58:7c:de:8e:e8 join, AID=4, bgn, 40U
I (758472) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1, snd_ch_cfg:0x0
I (758472) wifi:station: 94:a9:90:08:51:08 join, AID=5, bgn, 40U
I (758482) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1, snd_ch_cfg:0x0
I (758482) wifi:station: 94:a9:90:0d:7f:a4 join, AID=6, bgn, 40U
I (16:33:58.365) aWifiMesh: unknown id:25
I (16:33:58.398) aWifiMesh: unknown id:25
I (16:33:58.399) aWifiMesh: unknown id:25
I (16:33:58.401) aWifiMesh: unknown id:25
W (16:33:58.414) aWifiMesh: <MESH_EVENT_ROUTING_TABLE_ADD>add 1, new:2
I (16:33:58.422) aWifiMesh: <MESH_EVENT_CHILD_CONNECTED>aid:1, 48:ca:43:9b:5d:20
W (16:33:58.425) aWifiMesh: <MESH_EVENT_ROUTING_TABLE_ADD>add 1, new:3
I (16:33:58.429) aWifiMesh: <MESH_EVENT_CHILD_CONNECTED>aid:3, 94:a9:90:0d:7f:14
W (16:33:58.432) aWifiMesh: <MESH_EVENT_ROUTING_TABLE_ADD>add 1, new:4

```
So I restarted the ESP32 and here is the comms showing my diagnostics:
```
[... 50 lines cut ...]
I (16:34:17.453) aWifiMesh: Vote: 100%
I (16:34:17.454) aWifiMesh: Network is: self organized
I (16:34:17.457) aWifiMesh: Mesh Root: 48:ca:43:9b:53:d8 is not fixed
I (16:34:17.459) aWifiMesh: Number of nodes: 1
I (16:34:17.460) aWifiMesh: Max layers: 4
I (16:34:17.462) aWifiMesh: Mesh Parent IP: 0.23.19.32: 29885  MAC: 00:17:13:20:bd:74  RSSI: -29
I (16:34:17.463) aWifiMesh: Mesh Parent MAC: 00:17:13:20:bd:74  RSSI: -29  Channel: 255
I (16:34:17.465) aWifiMesh: Mesh routing Table records: 1   Record size: 6   Table size: 6
I (16:34:17.467) aWifiMesh:   MAC: 48:ca:43:9b:53:d8
I (16:34:18.265) aMQTT: Pre esp_mqtt_client_init
W (16:34:18.266) aMQTT: Connecting to Broker: 192.168.1.2
I (16:34:18.269) aMQTT: mqtt5_cfg.task.priority: 0
[... cut ...]
I (16:34:18.272) aMQTT: mqtt5_cfg.buffer.out_size: 10240
I (16:34:18.273) aMQTT: Post esp_mqtt_client_init
I (16:34:18.275) aMQTT: Broker pre Connect
I (16:34:18.278) aMQTT: Other event id:7
I (16:34:18.280) aMQTT: Broker post connect
I (16:34:18.281) aMQTT: triggerMqttSubscribe
W (16:34:18.310) aMQTT: Broker connected
```

Here is a Wireshark log filtering on dhcp. It is a .pcapng file renamed as .txt so I could upload it.
[Switch VONETS2 to VONETS.txt](https://github.com/user-attachments/files/21225831/Switch.VONETS2.to.VONETS.txt)

zhangyanjiaoesp, 2025-07-15:
@michaelsimp 
Your packet capture looks like it's from a wired interface, but what we expect is a wireless capture, like the following. The DHCP Discover packet should be sent immediately after the Wi-Fi connection is successfully established. 
[image]

However, your log shows that `esp_netif_dhcpc_start(esp_netif)` was not called, you can further trace whether it's due to `esp_netif_action_connected()` not being called, or because the condition `status == ESP_NETIF_DHCP_INIT` is not met. All the related code is open source in the ESP-IDF, so you can trace through it yourself.

michaelsimp, 2025-07-15:
Thanks. I will delve into this tomorrow.  But can you please confirm if it is ok to stop Wifi Mesh and then restart it with a different SSID.  The initial reset of my ESP32 which starts WiFi Mesh always seems to work properly and fire the <IP_EVENT_STA_GOT_IP> event. I only seem to have a problem with **NOT** getting the <IP_EVENT_STA_GOT_IP> event after a WiFiMesh stop and restart. 

I tried setting the SSID on both of my WiFi Access points the same but it didn't help.  Is this because the WiFi-AP MAC address or BSSID is being used for the reconnection rather than the SSID?  If I was to stop WiFi Mesh and then restart it with the same SSID, to resolve the MESH_ROOT scanning using BSSID, would this be a better solution than having 2 WiFi-APs with different SSIDs? Your advice appreciated

michaelsimp, 2025-07-15:
I had another go at this tonight 

[Switch VONETS2 to VONETS.txt](https://github.com/user-attachments/files/21228878/Switch.VONETS2.to.VONETS.txt)

as I am keen for a solution:

Here is my modified code - changes with MBS on the line
```
void esp_netif_action_connected(void *esp_netif, esp_event_base_t base, int32_t event_id, void *data)
{

    // ESP_LOGD(TAG, "esp_netif action connected with netif%p from event_id=%" PRId32, esp_netif, event_id); // MBS removed
    ESP_LOGI(TAG, "esp_netif action connected with netif%p from event_id=%" PRId32, esp_netif, event_id); // MBS replace
    esp_netif_up(esp_netif);

    if (!(esp_netif_get_flags(esp_netif) & ESP_NETIF_DHCP_CLIENT)) {
        // No more actions for interfaces without DHCP client flag
        return;
    }
#if CONFIG_LWIP_IPV4
    esp_netif_dhcp_status_t status;
    ESP_NETIF_CALL_CHECK("connected action: dhcpc failed", esp_netif_dhcpc_get_status(esp_netif, &status), ESP_OK);
    ESP_LOGI(TAG, "esp_netif_action_connected status: %d", status); // MBS add

    if (status == ESP_NETIF_DHCP_INIT) {
        ESP_LOGW(TAG, "Pre calling esp_netif_dhcpc_start(esp_netif)"); // MBS add
        esp_netif_dhcpc_start(esp_netif);
        ESP_LOGI(TAG, "Post calling esp_netif_dhcpc_start(esp_netif)"); // MBS add

```

```
When it fails after a WiFi Mesh stop and restart with the new SSID, I don't see the diagnostics I added:
I (1444802) mesh: [SCAN][ch:1]AP:3, other(ID:0, RD:0), MAP:0, idle:0, candidate:1, root:0, topMAP:0[c:0,i:1][00:17:13:20:bd:74]router found<>
I (1444812) mesh: 1328[SCAN]init rc[48:ca:43:9b:5c:69,-31], mine:0, voter:0
I (1444812) mesh: [SCAN:10/10]rc[128][48:ca:43:9b:5c:69,-29], self[48:ca:43:9b:5c:68,-31,reason:0,votes:1,idle][mine:1,voter:1(1.00)percent:1.00][128,1,48:ca:43:9b:5c:69]

I (1444812) mesh: [DONE]connect to router:VONETS, channel:1, rssi:-31, 00:17:13:20:bd:74[layer:0, assoc:0], my_vote_num:1/voter_num:1, rc[48:ca:43:9b:5c:69/-29/1]
I (1444822) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1, snd_ch_cfg:0x0
I (1444832) wifi:state: init -> auth (0xb0)
I (1444832) wifi:state: auth -> assoc (0x0)
I (1444842) wifi:state: assoc -> run (0x10)
I (1445052) wifi:<ba-add>idx:0 (ifx:0, 00:17:13:20:bd:74), tid:0, ssn:2, winSize:64
I (1445052) wifi:connected with VONETS, aid = 1, channel 1, 40U, bssid = 00:17:13:20:bd:74
I (1445052) wifi:security: WPA2-PSK, phy: bgn, rssi: -30
I (1445052) wifi:pm start, type: 0

I (1445062) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
I (1445062) mesh: [scan]new scanning time:600ms, beacon interval:300ms
I (1445062) mesh: <flush>upstream packets, connections(max):6, waiting:0, upQ:0
I (20:20:09.325) aWifiMesh: <MESH_EVENT_PARENT_CONNECTED>layer:1-->1, parent:00:17:13:20:bd:74<ROOT>, ID:77:77:77:77:77:76
I (1445062) mesh: <flush>root
I (20:20:09.327) mesh_netif: Already wifi station, no need to do anything
I (1445062) mesh: [TXQ]<max:32>up(0, be:0), down(0, be:0), mgmt:0, xon(req:0, rsp:0), bcast:0, wnd(0, parent:00:00:00:00:00:00)
I (20:20:09.330) aWifiMesh: <MESH_EVENT_PARENT_CONNECTED>layer:1-->1, parent:00:17:13:20:bd:74<ROOT>, ID:77:77:77:77:77:76
I (1445072) mesh: [RXQ]<max:32 = cfg:32 + extra:0>self:0, <max:32 = cfg:32 + extra:0>tods:0
I (20:20:09.333) mesh_netif: Already wifi station, no need to do anything
I (20:20:09.336) aWifiMesh: <MESH_EVENT_TODS_REACHABLE>state:0
I (20:20:09.337) aWifiMesh: <MESH_EVENT_TODS_REACHABLE>state:0
I (20:20:09.339) aWifiMesh: <MESH_EVENT_ROOT_ADDRESS>root address:48:ca:43:9b:5c:69
I (20:20:09.341) aWifiMesh: <MESH_EVENT_ROOT_ADDRESS>root address:48:ca:43:9b:5c:69
I (1445102) wifi:AP's beacon interval = 102400 us, DTIM period = 1
20:20:13 [HOST] Entered Standalone Mode
I (20:20:13.531) aStoreIDF: Storage Header Write ESTOP
20:20:15 [PUMP] Standalone mode AutoAuthorise
20:20:15 [PUMP] Standalone mode AutoAuthorise
20:20:15 [PUMP] Standalone mode AutoAuthorise
E (20:20:19.764) aWifiMesh: ip_event_handler did not fire. Rebooting...
```
So it seems `void esp_netif_action_connected(void *esp_netif, esp_event_base_t base, int32_t event_id, void *data)
[... 18 lines cut ...]
I (12862) wifi:pm start, type: 0

I (12862) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (12862) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
I (20:20:35.726) esp_netif_handlers: esp_netif action connected with netif0x3fccddf4 from event_id=4
I (20:20:35.729) esp_netif_handlers: esp_netif_action_connected status: 0
W (20:20:35.731) esp_netif_handlers: Pre calling esp_netif_dhcpc_start(esp_netif)
I (20:20:35.735) esp_netif_handlers: Post calling esp_netif_dhcpc_start(esp_netif)
I (12872) mesh: [scan]new scanning time:600ms, beacon interval:300ms
I (12872) mesh: <flush>upstream packets, connections(max):6, waiting:0, upQ:0
I (20:20:35.740) aWifiMesh: <MESH_EVENT_PARENT_CONNECTED>layer:0-->1, parent:00:17:13:20:bd:74<ROOT>, ID:77:77:77:77:77:76
I (12882) mesh: <flush>root
I (20:20:35.742) mesh_netif: Already wifi station, no need to do anything
I (12882) mesh: [TXQ]<max:32>up(0, be:0), down(0, be:0), mgmt:0, xon(req:0, rsp:0), bcast:0, wnd(0, parent:00:00:00:00:00:00)
I (20:20:35.746) aWifiMesh: <MESH_EVENT_TODS_REACHABLE>state:0
I (12882) mesh: [RXQ]<max:32 = cfg:32 + extra:0>self:0, <max:32 = cfg:32 + extra:0>tods:0
I (20:20:35.748) aWifiMesh: <MESH_EVENT_ROOT_ADDRESS>root address:48:ca:43:9b:5c:69
I (12922) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (20:20:37.370) esp_netif_handlers: sta ip: 192.168.1.112, mask: 255.255.255.0, gw: 192.168.1.1
I (20:20:37.372) aWifiMesh: <IP_EVENT_STA_GOT_IP>IP:192.168.1.112
I (20:20:37.378) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 10.0.0.1

```
I disconnnected my Ethernet cable and connected to the WiFi AP I am swithcing my nodes to and captured the attached Wireshark log.  I hope this is what you need.
Thanks for your help

zhangyanjiaoesp, 2025-07-18:
@michaelsimp 
Sorry, I have higher-priority tasks to handle at the moment and won’t be able to help you debug this issue right now. Please provide the complete code that can reproduce the problem. I might be able to offer some debug patches for you next week.

michaelsimp, 2025-07-24:
I made a test app from "ip_internal_network"
In the source file mesh_main.c I added two functions to:

When the `<IP_EVENT_STA_GOT_IP>` event fires, I set a boolean `restartMesh `read by the mainline task which calls2 new functions I added to stop and restart WiFiMesh as follows:
```
    while (true) {
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        if (restartMesh) {
            restartMesh = false;
            vTaskDelay(3000 / portTICK_PERIOD_MS);
            wifiMeshStop();
            wifiMeshStart();
        }
    }
```

I found this works very reliably with a very rare exception where `<MESH_EVENT_PARENT_CONNECTED`> fires but`<IP_EVENT_STA_GOT_IP>` does not fire. See log below,

I am wondering if the WiFi Access point is still in the process of starting up and has enabled a connection but DHCP may not have started yet?

If the DHCP request process fails and the ESP32 node doesn't get an IP address, is it supposed to retry and if so, after what delay?
Is there any way for me to programmatically request an IP address again?
Is there any way to view the DHCP progress in the logs:

- DHCPDiscover
- DHCPOFFER
- DHCPREQUEST
- DHCPACK
- DHCPNAK

From the Forum and Stack Overflow, I dont seem to be the only one who has seen this issue? 

```
I (142497) mesh: [SCAN][ch:1]AP:2, other(ID:0, RD:0), MAP:0, idle:0, candidate:1, root:0, topMAP:0[c:0,i:1][00:17:13:20:bd:74]router found<>
I (142497) mesh: 1328[SCAN]init rc[48:ca:43:9b:53:d9,-28], mine:0, voter:0
I (142497) mesh: [SCAN:10/10]rc[128][48:ca:43:9b:53:d9,-23], self[48:ca:43:9b:53:d8,-28,reason:0,votes:1,idle][mine:1,voter:1(1.00)percent:1.00][128,1,48:ca:43:9b:53:d9]

I (142507) mesh: [DONE]connect to router:VONETS, channel:1, rssi:-28, 00:17:13:20:bd:74[layer:0, assoc:0], my_vote_num:1/voter_num:1, rc[48:ca:43:9b:53:d9/-23/1]
I (142517) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1, snd_ch_cfg:0x0
I (142517) wifi:state: init -> auth (0xb0)
I (142517) wifi:state: auth -> assoc (0x0)
I (142527) wifi:state: assoc -> run (0x10)
WIN>
                                                                                                                                                                                                                                                    I (142797) wifi:<ba-add>idx:0 (ifx:0, 00:17:13:20:bd:74), tid:0, ssn:2, winSize:64
I (142797) wifi:connected with VONETS, aid = 1, channel 1, 40U, bssid = 00:17:13:20:bd:74
I (142797) wifi:security: WPA2-PSK, phy: bgn, rssi: -22
I (142797) wifi:pm start, type: 0

I (142797) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
I (01:33:04.363) esp_netif_handlers: esp_netif action connected with netif0x3fcb850c from event_id=4
I (01:33:04.365) esp_netif_handlers: esp_netif_action_connected status: 0
W (01:33:04.367) esp_netif_handlers: Pre calling esp_netif_dhcpc_start(esp_netif)
I (01:33:04.370) esp_netif_handlers: Post calling esp_netif_dhcpc_start(esp_netif)
I (142817) mesh: [scan]new scanning time:600ms, beacon interval:300ms
I (01:33:04.375) aWifiMesh: <MESH_EVENT_PARENT_CONNECTED>layer:1-->1, parent:00:17:13:20:bd:74<ROOT>, ID:77:77:77:77:77:76
I (01:33:04.377) mesh_netif: Already wifi station, no need to do anything
I (01:33:04.380) aWifiMesh: <MESH_EVENT_TODS_REACHABLE>state:0
I (142817) mesh: <flush>upstream packets, connections(max):4, waiting:0, upQ:0
I (142827) mesh: <flush>root
I (142827) mesh: [TXQ]<max:32>up(0, be:0), down(0, be:0), mgmt:0, xon(req:0, rsp:0), bcast:0, wnd(0, parent:00:00:00:00:00:00)
I (01:33:04.383) aWifiMesh: <MESH_EVENT_ROOT_ADDRESS>root address:48:ca:43:9b:53:d9
I (142827) mesh: [RXQ]<max:32 = cfg:32 + extra:0>self:0, <max:32 = cfg:32 + extra:0>tods:0
WIN> I (142867) wifi:AP's beacon interval = 102400 us, DTIM period = 1
E (01:33:15.041) aWifiMesh: ************ ip_event_handler did not fire  Rebooting...****************

```