espressif/esp-idf #16868: In Windows pwsh, when importing export.ps1, detect_shell failed due to uppercase/lowercase issues.

a3750, 2025-07-14:
My shell is installed by scoop, named pwsh.

When I importing `export.ps1`, it will.
```pwsh
$ & $env:IDF_PATH/export.ps1                                                                                         
Activating ESP-IDF 5.4                                                                                               
warning:  IDF_PATH is changed from 'E:\scoop\apps\esp-idf\current' to 'E:\scoop\apps\esp-idf\5.4.2'.
* Checking python version ... 3.13.5
* Checking python dependencies ... OK
* Deactivating the current ESP-IDF environment (if any) ... OK
* Establishing a new ESP-IDF environment ... OK
* Identifying shell ... pwsh.EXE
* Detecting outdated tools in system ... OK - no outdated tools found
error:  "pwsh.EXE" shell is not among the supported options: "bash zsh fish sh ksh dash nu pwsh.exe pwsh powershell.exe powershell cmd.exe cmd"

ERROR: Activation script failed
Command '['E:\\scoop\\persist\\esp-idf\\.espressif\\python_env\\idf5.4_py3.13_env\\Scripts\\python.exe', 'E:\\scoop\\apps\\esp-idf\\5.4.2\\tools\\export_utils\\activate_venv.py', '--export']' returned non-zero exit status 1.
To view detailed debug information, set ESP_IDF_EXPORT_DEBUG=1 and run the export script again.
InvalidOperation: E:\scoop\apps\esp-idf\current\export.ps1:27:3
Line |
  27 |  . $idf_exports
     |    ~~~~~~~~~~~~
     | The expression after '.' in a pipeline element produced an object that was not valid. It must result in a     
     | command name, a script block, or a CommandInfo object.
```

The reason is that in the `$env:ESP_IDF/tools/export_utils/activate_venv.py`, the function `detect_shell` detects the shell name as `pwsh.EXE`. However, `SHELL_CLASSES` is like this
```python
SHELL_CLASSES = {
    'bash': BashShell,
    'zsh': ZshShell,
    'fish': FishShell,
    'sh': UnixShell,
    'ksh': UnixShell,
    'dash': UnixShell,
    'nu': UnixShell,
    'pwsh.exe': PowerShell,
    'pwsh': PowerShell,
    'powershell.exe': PowerShell,
    'powershell': PowerShell,
    'cmd.exe': WinCmd,
    'cmd': WinCmd
}
```
Here is `pwsh.exe` so `pwsh.EXE` does not match, which causes the following code segment to terminate abnormally.
```python
if detected_shell not in SHELL_CLASSES:
    die(f'"{detected_shell}" shell is not among the supported options: "{SUPPORTED_SHELLS}"')
```

mfialaf, 2025-07-14:
Hi @a3750,

Thank you for opening this issue. We haven’t yet encountered `powershell.EXE` with this casing. Since changing the `detected_shell` name to lowercase could introduce execution problems, would adding `powershell.EXE` to the SHELL_CLASSES list resolve the issue for you?

``` python
SHELL_CLASSES = {
    'bash': BashShell,
    'zsh': ZshShell,
    'fish': FishShell,
    'sh': UnixShell,
    'ksh': UnixShell,
    'dash': UnixShell,
    'nu': UnixShell,
    'pwsh.exe': PowerShell,
    'pwsh': PowerShell,
    'powershell.exe': PowerShell,
    'powershell.EXE': PowerShell,
    'powershell': PowerShell,
    'cmd.exe': WinCmd,
    'cmd': WinCmd
}
```

a3750, 2025-07-15:
In fact, the detected name is `pwsh.EXE`, so it's necessary to add both of them.

closed 2025-07-24 by espressif-bot