espressif/esp-idf #16872: assert failed: esp_sleep_pd_config sleep_modes.c:1983 (refs >= 0)

inbarab, 2025-07-15:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.3-11-gb3ea8d979b

### Espressif SoC revision.

esp32-s3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-s3 wroom-1U

### Power Supply used.

Battery

### What is the expected behavior?

light sleep configuration should complete and light sleep should start

### What is the actual behavior?

System crashes:

assert failed: esp_sleep_pd_config sleep_modes.c:1983 (refs >= 0)

Backtrace: 0x403762c6:0x3fceaab0 0x40381b95:0x3fceaad0 0x4038bd19:0x3fceaaf0 0x42061c63:0x3fceac10 0x4203b3c4:0x3fceac30 0x4038278d:0x3fceac90

ELF file SHA256: 11bc3dd0e

Rebooting...

### Steps to reproduce.

light sleep configuration:
err = esp_sleep_enable_timer_wakeup(sleep_to_us);
if (err != ESP_OK) {
    ESP_LOGE(LOG_TAG, "Failed setting sleep timer, Skip Sleep");
    return err;
}

/* override auto power down option */
 esp_sleep_pd_config(ESP_PD_DOMAIN_VDDSDIO, ESP_PD_OPTION_ON);
/* initialize gpio wakeup */
 err = gpio_wakeup_enable(gpio_17,  GPIO_INTR_HIGH_LEVEL);
if (err != ESP_OK) {
    ESP_LOGE(LOG_TAG, "Enable gpio wakeup failed");
    return err;
}

// make sure gpio is not high - read level and wait if high - omitting this code, taken from example 

/* Enter sleep mode */
ESP_LOGI(LOG_TAG, "Starting sleep");
err = esp_light_sleep_start();

### Debug Logs.

```plain
assert failed: esp_sleep_pd_config sleep_modes.c:1983 (refs >= 0)

Backtrace: 0x403762c6:0x3fceaab0 0x40381b95:0x3fceaad0 0x4038bd19:0x3fceaaf0 0x42061c63:0x3fceac10 0x4203b3c4:0x3fceac30 0x4038278d:0x3fceac90

PS C:\dev\ESP\wispot3\release\wispot-3> xtensa-esp32s3-elf-addr2line -pfiaC -e build/ws3_5_2_811.elf  0x403762c6:0x3fceaab0 0x40381b95:0x3fceaad0 0x4038bd19:0x3fceaaf0 0x42061c63:0x3fceac10 0x4203b3c4:0x3fceac30 0x4038278d:0x3fceac90
0x403762c6: panic_abort at C:/dev/ESP/esp-idf523/components/esp_system/panic.c:466
0x40381b95: esp_system_abort at C:/dev/ESP/esp-idf523/components/esp_system/port/esp_system_chip.c:93
0x4038bd19: __assert_func at C:/dev/ESP/esp-idf523/components/newlib/assert.c:81
0x42061c63: esp_sleep_pd_config at C:/dev/ESP/esp-idf523/components/esp_hw_support/sleep_modes.c:1983 (discriminator 1)
0x4203b3c4: ws3_sleep_task_entry at C:/dev/ESP/wispot3/release/wispot-3/components/ws3_sleep/ws3_sleep.c:349
0x4038278d: vPortTaskWrapper at C:/dev/ESP/esp-idf523/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: 11bc3dd0e

Rebooting...
```

### More Information.

The error does not occur every call to esp_sleep_pd_config
It happens once every ~24 hours

esp-wzh, 2025-07-16:
Hi, I can't reproduce the problem you encountered with only the above code, but from the backtrace given, it seems that `esp_sleep_pd_config` is called at `ws3_sleep.c:349` in your project to configure a power domain that has been configured to `ESP_PD_OPTION_OFF` during sleep. The `pd_config` of the power domain is managed using reference counting, which is already in the `ESP_PD_OPTION_OFF` state, disable it again causes its count to be reduced to a negative value, triggering the assert.

inbarab, 2025-07-16:
Thanks for the quick reply
I am trying to ENABLE the option, not disable it (see code snippet included above)
Please advise,
Cheers,
Inbar

inbarab, 2025-07-16:
> I'd have to increment the refs every 40 microseconds to overflow in 24
> hours
>
>
> https://github.com/espressif/esp-idf/blob/v5.2.3/components/esp_hw_support/sleep_modes.c#L191
>
> The reference count type is int16_t, so the overflow time may not be that
> long.
>
> Is there an API to GET the domain state?
>
> Sorry, this API does not exist currently.

refs is an int, not int16_t. This is from the code you linked:

esp_err_t esp_sleep_pd_config(esp_sleep_pd_domain_t domain,
esp_sleep_pd_option_t option)
{
    if (domain >= ESP_PD_DOMAIN_MAX || option > ESP_PD_OPTION_AUTO) {
        return ESP_ERR_INVALID_ARG;
    }
    portENTER_CRITICAL_SAFE(&s_config.lock);

    int refs = (option == ESP_PD_OPTION_ON)  ?
s_config.domain[domain].refs++ \
             : (option == ESP_PD_OPTION_OFF) ?
--s_config.domain[domain].refs \
             : s_config.domain[domain].refs;
    if (refs == 0) {
        s_config.domain[domain].pd_option = option;
    }
    portEXIT_CRITICAL_SAFE(&s_config.lock);
    assert(refs >= 0);
    return ESP_OK;
}

inbarab, 2025-07-16:
> Yes, once is enough.

Thanks