espressif/esp-idf #16882: USB host appears to leak HDC resources when closing and reopening connections

safepassglobal, 2025-07-16:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.2

### Espressif SoC revision.

ESP32-S3 Chip is ESP32-S3 (QFN56) (revision v0.2) Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3) Crystal is 40MHz MAC: cc:8d:a2:ed:cc:a4 Uploading stub... Running stub... Stub running... Warning: ESP32-S3 has no Chip ID. Reading MAC instead. MAC: cc:8d:a2:ed:cc:a4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

I have multiple USB devices (up to 20) connected via a hub.  I can enumerate and handle the plug/unplug events communicate with each device separately.  Due to the 8 HCD limit on the S3, I have opted to handle each device in turn in a round robin fashion (serial data CDC).  I expect to be able to close a device and reopen at a later time to continue processing where I left off.

### What is the actual behavior?

closing and reopening appears to cause a leak of HCD resources and other errors on the reopen.

### Steps to reproduce.

1) plug in 3+ devices into the hub connected to the USB-OTG.
2) Open a device
3) Transact serial data of the CDC endpoints
4) close the device
5) repeat in round robin fashion for all connected devices.

### Debug Logs.

```plain
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2f (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x1564
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd24
load:0x403cb700,len:0x2ed4
entry 0x403c8928
I (27) boot: ESP-IDF v5.4.2 2nd stage bootloader
I (27) boot: compile time Jul 14 2025 18:29:15
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (30) boot: efuse block revision: v1.3
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
[... 529 lines cut ...]
I (17862) BADGE_VCP: Port 7: Session START, opening device addr 4
I (17872) BADGE_VCP: Port 7: Device open SUCCESS, handle=0x3FCAB434
I (17872) BADGE_VCP: Port 7: Claiming interface...
E (17882) HCD DWC: No more HCD channels available
E (17882) USBH: EP Alloc error: ESP_ERR_NOT_SUPPORTED
E (17892) USB HOST: EP allocation error ESP_ERR_NOT_SUPPORTED
E (17892) USB HOST: Claiming interface error: ESP_ERR_NOT_SUPPORTED
E (17902) BADGE_VCP: Port 7: Failed to claim interface.
I (17902) BADGE_VCP: Port 7: Session END, closing device handle 0x3FCAB434
I (17912) BADGE_VCP: ---- Top of Service Loop ----
I (19302) MAIN_APP: Main task found no active ports.
I (20302) MAIN_APP: Main task found no active ports.
I (21302) MAIN_APP: Main task found no active ports.
I (22302) MAIN_APP: Main task found no active ports.
I (23302) MAIN_APP: Main task found no active ports.
I (24302) MAIN_APP: Main task found no active ports.
```

### Diagnostic report archive.

[diag-4c364019-c7b8-41ce-902a-fe15cc6294f9.zip](https://github.com/user-attachments/files/21262090/diag-4c364019-c7b8-41ce-902a-fe15cc6294f9.zip)

### More Information.

The devices stay physically connected and fully functional.  Communications works fine if I only open 2 (HCD limit) and leave the devices open.  If the system can not support round robin servicing of the devices, then enough HCD resources to open and transact with 20 devices could be a work around.

roma-jam, 2025-07-16:
Hi @safepassglobal, 

thanks for the report and log.

Can you share the part of the code, how you open the device and how you close the device on `Session END`?

safepassglobal, 2025-07-16:
[badgeCommunications.zip](https://github.com/user-attachments/files/21262370/badgeCommunications.zip)
Zipped the source that I am using.

safepassglobal, 2025-07-16:
Yeah, I thought I was releasing the bulk transfers etc. correctly.  I have tried several various ways of doing it.  

I (4147) BADGE_VCP: Port 1: Found START delimiter.
I (4147) BADGE_VCP: Port 1: Found END delimiter.
I (4147) BADGE_VCP: Port 1: Handshake 'LOGOFF' successful. Port is now IDLE.
I (4157) BADGE_VCP: Port 1: Releasing interface 1...
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4200b6c6
--- 0x4200b6c6: device_manager_task at C:/Projects/SafePass/esp32/esp32idf/SmartDock/components/badgeCommunications/badgeVCP.c:417

file: "./components/badgeCommunications/badgeVCP.c" line 417
func: device_manager_task
expression: usb_host_device_close(client_hdl, dev_hdl)

abort() was called at PC 0x4037a453 on core 1
--- 0x4037a453: _esp_error_check_failed at C:/Users/aps/esp/v5.4.2/esp-idf/components/esp_system/esp_err.c:49

Backtrace: 0x40375b35:0x3fca8890 0x4037a45d:0x3fca88b0 0x40380d91:0x3fca88d0 0x4037a453:0x3fca8940 0x4200b6c6:0x3fca8970 0x4037aec1:0x3fca89d0
--- 0x40375b35: panic_abort at C:/Users/aps/esp/v5.4.2/esp-idf/components/esp_system/panic.c:469
--- 0x4037a45d: esp_system_abort at C:/Users/aps/esp/v5.4.2/esp-idf/components/esp_system/port/esp_system_chip.c:87
--- 0x40380d91: abort at C:/Users/aps/esp/v5.4.2/esp-idf/components/newlib/abort.c:38
--- 0x4037a453: _esp_error_check_failed at C:/Users/aps/esp/v5.4.2/esp-idf/components/esp_system/esp_err.c:49
--- 0x4200b6c6: device_manager_task at C:/Projects/SafePass/esp32/esp32idf/SmartDock/components/badgeCommunications/badgeVCP.c:417
--- 0x4037aec1: vPortTaskWrapper at C:/Users/aps/esp/v5.4.2/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: 62d1f6eb5

Rebooting...
���ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2f (SPI_FAST_FLASH_BOOT)
Saved PC:0x42007bbf
--- 0x42007bbf: uart_ll_get_txfifo_len at C:/Users/aps/esp/v5.4.2/esp-idf/components/hal/esp32s3/include/hal/uart_ll.h:418
---  (inlined by) uart_tx_char at C:/Users/aps/esp/v5.4.2/esp-idf/components/esp_driver_uart/src/uart_vfs.c:190

safepassglobal, 2025-07-16:
Also, what is the issue then that is causing it to look like a resource leak?  Incomplete close?

roma-jam, 2025-08-05:
Hi @safepassglobal,

there is a USB Host CDC driver is available in the ESP component registry: [USB Host CDC-ACM Class Driver](https://components.espressif.com/components/espressif/usb_host_cdc_acm/versions/2.1.0)

It is opening and closing the CDC device, so it is might be helpful to your goal.