espressif/esp-idf #16884: Cert Phy test(Task watchdog got triggered)

alexsvn, 2025-07-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

2abf7c6ec595ed5cf948f0d8b77f6b0ab069dbf1

### Espressif SoC revision.

ESP32 chip revision: v3.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32

### Power Supply used.

External 3.3V

### What is the expected behavior?

normal

### What is the actual behavior?

1.I am integrating cert test related programs; the following loop error occurred

cert_wifi_tx xTaskCreate task
```log
E (50139) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (50139) task_wdt:  - IDLE0 (CPU 0)
E (50139) task_wdt: Tasks currently running:
E (50139) task_wdt: CPU 0: cert_wifi_tx
E (50139) task_wdt: CPU 1: IDLE1
E (50139) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x401628BF:0x3FFC03A0 0x40162CF8:0x3FFC03C0 0x40086A0D:0x3FFC03F0 0x401BE445:0x3FFBB4D0 0x401BC8F8:0x3FFBB4F0 0x401BF0BE:0x3FFBBC00 0x401BE015:0x3FFBBC20 0x401BC1C5:0x3FFBBC40 0x400E2B2E:0x3FFBBC60 0x4009A35D:0x3FFBBC80

E (55139) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (55139) task_wdt:  - IDLE0 (CPU 0)
E (55139) task_wdt: Tasks currently running:
E (55139) task_wdt: CPU 0: cert_wifi_tx
E (55139) task_wdt: CPU 1: IDLE1
E (55139) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x401628BF:0x3FFC03A0 0x40162CF8:0x3FFC03C0 0x40086A0D:0x3FFC03F0 0x40083041:0x3FFBB4D0 0x401BC901:0x3FFBB4F0 0x401BF0BE:0x3FFBBC00 0x401BE015:0x3FFBBC20 0x401BC1C5:0x3FFBBC40 0x400E2B2E:0x3FFBBC60 0x4009A35D:0x3FFBBC80
...
```

2.The Readme describes
```txt
* Enable RF certification test functions.
* Disable Interrupt Watchdog.
* Disable Task Watchdog Timer.
```

3.Is it possible to port this in production version?
The watchdog is disabled, and the following error occurs after startup
```log
assert failed: esp_startup_start_app app_startup.c:72 (!soc_has_cache_lock_bug() && "ESP32 Rev 3 + Dual Core + PSRAM requires INT WDT enabled in project config!")
```

### Steps to reproduce.

call esp_phy_wifi_rx_func
call esp_phy_wifi_tx_func

### Debug Logs.

```plain

```

mhdong, 2025-07-17:
Hi, there are two approaches we can consider. If the task we're testing won't take particularly long, we can choose to enable the task watchdog and simply set a longer timeout—for example, 60 seconds.
If the duration of the test task is uncertain, you can control the task watchdog using code within the specific task. Refer to the example below for how to do this. https://github.com/espressif/esp-idf/tree/master/examples/system/task_watchdog

alexsvn, 2025-07-17:
Dear @mhdong ,
Cert_wifi_tx is a library function. When called in a thread, it will block. I tried to use the watchdog example program, but it failed to achieve prefetching. It still gets stuck in the cert_wifi_tx call, and cannot feed the dog.

Method 1: How to Set Temporary Extra Long Feeding for Dogs
Thanks.

alexsvn, 2025-07-18:
Dear @mhdong ,
Thanls,
The implementation of temporarily closing Watchdog using esp_task_wdt_stop() and esp_task_wdt_restart() achieved the goal

closed 2025-07-18 by alexsvn