espressif/esp-idf #16886: usb uac not working as expected with esp32s3

Kristian8606, 2025-07-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Is there more comprehensive documentation for usb uac? the basic example in the repository is not working properly and a whistling sound is heard approximately every 2 seconds. If anyone has a working example and can share it, I would be grateful.

peter-marcisovsky, 2025-07-17:
Hi @Kristian8606 

Can you provide some more information about the issues you are facing?
- Can you explain your setup? Are you using UAC as a USB Host or as a USB Device?
- Which example are you using?
- What are the issues you are facing?
- What is the idf version, and the version of the UAC Device/Host component you are using?

Thank you.

peter-marcisovsky, 2025-07-28:
Hi @L-KAYA could you PTAL, regarding the I2S driver example?

L-KAYA, 2025-07-28:
Hi @Kristian8606 @peter-marcisovsky, it was confirmed by @leeebo that the UAC is handled differently on MacOS compare to the Windows. But it should work correctly on windows.

leeebo, 2025-07-29:
@Kristian8606 The hissing sound on Windows may be related to  https://github.com/espressif/esp-iot-solution/pull/512

For MacOS, please try with enable UAC_SUPPORT_MACOS https://github.com/espressif/esp-iot-solution/blob/master/components/usb/usb_device_uac/Kconfig.uac#L41

Kristian8606, 2025-07-29:
I tried this example.
````
/*
 * SPDX-FileCopyrightText: 2025 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 */

#include <stdio.h>
#include <string.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/i2s_std.h"
#include "driver/i2c_master.h"
#include "driver/gpio.h"
#include "esp_system.h"
#include "usb_device_uac.h"
#include "esp_codec_dev_defaults.h"
#include "esp_codec_dev.h"
#include "esp_codec_dev_vol.h"
#include "i2s_usb_example.h"
#include "esp_log.h"

static const char *TAG = "i2s_usb";

static i2s_chan_handle_t i2s_tx_handle = NULL;
static esp_codec_dev_handle_t codec_handle = NULL;
static uint32_t init_mclk_freq_hz = 0;

static void i2s_driver_init(void)
{
    i2s_chan_config_t chan_cfg = I2S_CHANNEL_DEFAULT_CONFIG(0, I2S_ROLE_MASTER);
    chan_cfg.auto_clear = true;
    chan_cfg.dma_desc_num = 17;
    chan_cfg.dma_frame_num = 384;

    i2s_std_config_t std_cfg = {
        .clk_cfg = I2S_STD_CLK_DEFAULT_CONFIG(44100),
        .slot_cfg = I2S_STD_PHILIPS_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_STEREO),
        .gpio_cfg = {
            .mclk = -1,
            .bclk = 7,
            .ws = 6,
            .dout = 4,
            .din = -1,
            .invert_flags = {
                .mclk_inv = false,
                .bclk_inv = false,
                .ws_inv = false,
            },
        },
    };
    std_cfg.clk_cfg.mclk_multiple = EXAMPLE_CODEC_MCLK_MULTIPLE;
    std_cfg.slot_cfg.slot_bit_width = I2S_SLOT_BIT_WIDTH_32BIT;

    ESP_ERROR_CHECK(i2s_new_channel(&chan_cfg, &i2s_tx_handle, NULL));
    ESP_ERROR_CHECK(i2s_channel_init_std_mode(i2s_tx_handle, &std_cfg));
    ESP_ERROR_CHECK(i2s_channel_enable(i2s_tx_handle));

    i2s_tuning_info_t tune_info = {};
    ESP_ERROR_CHECK(i2s_channel_tune_rate(i2s_tx_handle, NULL, &tune_info));
    init_mclk_freq_hz = tune_info.curr_mclk_hz;
    printf("initial MCLK: %"PRIu32" Hz\n", init_mclk_freq_hz);
}

static esp_err_t usb_uac_device_output_cb(uint8_t *buf, size_t len, void *arg)
{
    size_t bytes_written = 0;
    int ret = i2s_channel_write(i2s_tx_handle, buf, len, &bytes_written, portMAX_DELAY);

#if EXAMPLE_DYNAMIC_TUNING_EN
    static uint32_t overflow_cnt = 0;
    static uint32_t underflow_cnt = 0;
    static uint32_t normal_cnt = 0;
    i2s_tuning_info_t tune_info = {};
    ESP_ERROR_CHECK(i2s_channel_tune_rate(i2s_tx_handle, NULL, &tune_info));

    if (tune_info.water_mark >= 90) {
        overflow_cnt++;
        if (overflow_cnt >= 5000) {
            i2s_tuning_config_t tune_cfg = EXAMPLE_I2S_DEFAULT_TUNING_CFG(init_mclk_freq_hz, true);
            ESP_ERROR_CHECK(i2s_channel_disable(i2s_tx_handle));
            ESP_ERROR_CHECK(i2s_channel_tune_rate(i2s_tx_handle, &tune_cfg, &tune_info));
            ESP_ERROR_CHECK(i2s_channel_enable(i2s_tx_handle));
            printf("overflow detected, current MCLK: %"PRIu32" Hz, %"PRId32" Hz changed\n", tune_info.curr_mclk_hz, tune_info.delta_mclk_hz);
            overflow_cnt = 0;
            normal_cnt = 0;
        }
    } else if (tune_info.water_mark < 40) {
        underflow_cnt++;
        if (underflow_cnt >= 5000) {
            i2s_tuning_config_t tune_cfg = EXAMPLE_I2S_DEFAULT_TUNING_CFG(init_mclk_freq_hz, false);
            ESP_ERROR_CHECK(i2s_channel_disable(i2s_tx_handle));
            ESP_ERROR_CHECK(i2s_channel_tune_rate(i2s_tx_handle, &tune_cfg, &tune_info));
            ESP_ERROR_CHECK(i2s_channel_enable(i2s_tx_handle));
            printf("underflow detected, current MCLK: %"PRIu32" Hz, %"PRId32" Hz changed\n", tune_info.curr_mclk_hz, tune_info.delta_mclk_hz);
            underflow_cnt = 0;
            normal_cnt = 0;
        }
    } else {
        normal_cnt++;
        if (normal_cnt >= 1000) {
            overflow_cnt = 0;
            underflow_cnt = 0;
            normal_cnt = 0;
        }
    }
#endif

    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "i2s write failed");
        return ESP_FAIL;
    }
    return ESP_OK;
}

static void usb_uac_device_set_mute_cb(uint32_t mute, void *arg)
{
    esp_codec_dev_set_out_mute(codec_handle, !!mute);
    ESP_LOGI(TAG, "uac-device %s muted", mute ? "is" : "cancel");
}

static void usb_uac_device_set_volume_cb(uint32_t volume, void *arg)
{
    esp_codec_dev_set_out_vol(codec_handle, volume);
    ESP_LOGI(TAG, "set uac-device volume to: %"PRIu32, volume);
}

static void usb_uac_device_init(void)
{
    uac_device_config_t config = {
        .output_cb = usb_uac_device_output_cb,
        .input_cb = NULL,
        .set_mute_cb = usb_uac_device_set_mute_cb,
        .set_volume_cb = usb_uac_device_set_volume_cb,
        .cb_ctx = NULL,
    };
    ESP_ERROR_CHECK(uac_device_init(&config));
}

void app_main(void)
{
    i2s_driver_init();
    usb_uac_device_init();

    i2s_tuning_info_t tune_info = {};
    while (1) {
        ESP_ERROR_CHECK(i2s_channel_tune_rate(i2s_tx_handle, NULL, &tune_info));
        ESP_LOGI(TAG, "MCLK: %"PRId32", delta: %"PRId32", buffer water mark: %"PRIu32"%%",
                 tune_info.curr_mclk_hz, tune_info.delta_mclk_hz, tune_info.water_mark);
        vTaskDelay(pdMS_TO_TICKS(2000));
    }
}

````

I activated UAC_SUPPORT_MACOS.
I added the suggested correction..

```
#if CFG_TUD_AUDIO_FUNC_1_N_CHANNELS_RX
static void usb_mic_task(void *pvParam)
{
    TickType_t xLastWakeTime = xTaskGetTickCount();
    while (1) {
        if (s_uac_device->mic_active == false) {
            // clear the notification
            ulTaskNotifyTake(pdTRUE, portMAX_DELAY);
            xLastWakeTime = xTaskGetTickCount();
            continue;
        }
        // clear the notification
@@ -479,6 +481,8 @@ static void usb_mic_task(void *pvParam)
            s_uac_device->mic_data_size = bytes_read;
            UAC_EXIT_CRITICAL();
        }

        vTaskDelayUntil(&xLastWakeTime, pdMS_TO_TICKS(MIC_INTERVAL_MS));
    }
}
```

now the sound works without interruption
thank you for your help