espressif/esp-idf #16887: xTaskCreate return out of memory

alexsvn, 2025-07-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

2abf7c6ec595ed5cf948f0d8b77f6b0ab069dbf1

### Espressif SoC revision.

ESP32 V3.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

normal

### What is the actual behavior?

xTaskCreate return out of memory
The task creation failed, returning -1 out of memory
xTaskCreate(task, "task", 1024, NULL, 10, &s_task_handle);

(1)The memory information is as follows
I (1293) heap_init: Initializing. RAM available for dynamic allocation:
I (1300) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1306) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1312) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1318) heap_init: At 3FFD8500 len 00007B00 (30 KiB): DRAM
I (1324) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1331) heap_init: At 3FFE4350 len 00019398 (100 KiB): D/IRAM
I (1338) esp_psram: Adding pool of 2048K of PSRAM memory to heap allocator
Heap summary for capabilities 0x00000004:
  At 0x3ffe46bc len 32767 free 591 allocated 31104 min_free 591
    largest_free_block 480 alloc_blocks 84 free_blocks 2 total_blocks 86
  At 0x3f800000 len 2097152 free 2094536 allocated 544 min_free 2092548
    largest_free_block 2064384 alloc_blocks 9 free_blocks 2 total_blocks 11
  At 0x3ffaff10 len 240 free 4 allocated 28 min_free 4
    largest_free_block 0 alloc_blocks 2 free_blocks 0 total_blocks 2
  At 0x3ffb6388 len 7288 free 4 allocated 6684 min_free 4
    largest_free_block 0 alloc_blocks 55 free_blocks 0 total_blocks 55
  At 0x3ffb9a20 len 16648 free 4 allocated 15716 min_free 4
    largest_free_block 0 alloc_blocks 48 free_blocks 0 total_blocks 48
  At 0x3ffd8500 len 31488 free 4 allocated 30532 min_free 4
    largest_free_block 0 alloc_blocks 54 free_blocks 0 total_blocks 54
  At 0x3ffe0440 len 15072 free 4 allocated 14308 min_free 4
    largest_free_block 0 alloc_blocks 86 free_blocks 0 total_blocks 86
  At 0x3ffe4350 len 103320 free 408 allocated 101892 min_free 4
    largest_free_block 184 alloc_blocks 37 free_blocks 3 total_blocks 40
  Totals:
    free 2095555 allocated 200808 min_free 2093163 largest_free_block 2064384

(2)Main task stack size
I (189250) TASK: Remaining Stack: 3116 bytes

Why does memory shortage occur at this time, where does xTaskCreation() occupy memory, and how to debug this issue; How to analyze the above memory information (contents of memory info)

### Steps to reproduce.

Start without creating, external selection triggers call to create task

### Debug Logs.

```plain

```

KonstantinKondrashov, 2025-07-22:
Hi @alexsvn!

You did correctly debug this issue by using `heap_caps_print_heap_info(MALLOC_CAP_8BIT)`, which prints the summary of the heap. Here you can see that most of the blocks (from internal RAM) do not have much space, 480 - 0 (see `largest_free_block = ` in the log). 
The block at 0x3f800000 has a significant amount of free memory (2064384 bytes), but as you can see, it is PSRAM. By default, for esp32 [it is not recommended to use this memory for tasks](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html#failure-to-initialize) due to some limitations (can be enabled [CONFIG_FREERTOS_TASK_CREATE_ALLOW_EXT_MEM](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig-reference.html#config-freertos-task-create-allow-ext-mem)). So, you need to free up a bit an internal memory to be able to create that task.

Some useful links:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/mem_alloc.html#stack-and-heap
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html#configuring-external-ram

alexsvn, 2025-07-24:
Dear @KonstantinKondrashov ,
Thanks for your reply.

closed 2025-07-24 by alexsvn