espressif/esp-idf #16891: Problem with ulp_riscv_delay_cycles

gestalth, 2025-07-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.5.0

### Espressif SoC revision.

S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom

### Power Supply used.

USB

### What is the expected behavior?

ULP runs without freezing.

### What is the actual behavior?

It freezes after several minutes of working correctly using I2C and I have to restart the ULP.

### Steps to reproduce.

1. Setup a RISCV ULP program that continuously reads from I2C.
2. Setup a heartbeat signal to monitor on main core if the ULP is still running.
3. Let run for quite a while. Maybe up to 30 or so minutes. It doesn't happen at the same time each time.
4. Probably the ULP will have frozen at some point in that time.
...

### Debug Logs.

```plain

```

### More Information.

I used trace variables to find that it was freezing due to ulp_riscv_delay_cycles(1) function called from ulp_riscv_i2c_wait_for_interrupt(). 

Currently it looks like this:
void static inline ulp_riscv_delay_cycles(uint32_t cycles)
{
    uint32_t start = ULP_RISCV_GET_CCOUNT();
    /* Off with an estimate of cycles in this function to improve accuracy */
    uint32_t end = start + cycles - 20;

    while (ULP_RISCV_GET_CCOUNT()  < end) {
        /* Wait */
    }
}

I think that the line "while (ULP_RISCV_GET_CCOUNT()  < end)" is creating problems near the roll over. I haven't seen the freeze since changing it to:
void static inline ulp_riscv_delay_cycles(uint32_t cycles)
{    
    uint32_t start = ULP_RISCV_GET_CCOUNT();

    /* Off with an estimate of cycles in this function to improve accuracy */
    if(cycles < 21)
        return;

    cycles = cycles-20;

    while ((ULP_RISCV_GET_CCOUNT() - start)  < cycles) {
        /* Wait */
    }
}

I have no idea if the offset value would still be accurate with this change though. Also, because of the subtraction in the while loop, the resolution would probably be reduced.

KonstantinKondrashov, 2025-07-18:
Hi @gestalth! Thank you for noticing this issue. We will fix it.

closed 2025-08-19 by espressif-bot

gestalth, 2025-08-22:
Yeah, looking at how it is in master, it's still bugged. It will be hard to track down later because it would freeze less now. Going back to the 8 bit example, if the loop can check the ULP_RISCV_GET_CCOUNT() every 5 cycles, then what you can get is this: 

end=255
ULP_RISCV_GET_CCOUNT()<end //246 <-Not there yet
ULP_RISCV_GET_CCOUNT()<end //251 <-Not there yet
ULP_RISCV_GET_CCOUNT()<end //0     <-Not there yet (missed it)

This can happen any time 'end' is calculated to a value near the overflow. You don't control what 'end' gets calculated to. It depends on when the function is called. If you keep the subtraction in the loop (ULP_RISCV_GET_CCOUNT()-start < cycles), you can't check as often, but you don't miss it unless you choose cycles to be a value near the overflow.

m-mcgowan, 2025-08-25:
Thinking about need for accuracy, especially for short delays, I would imagine using some unrolled loops might help reduce error.

For the lowest overhead and most precise timings, the unrolled loops could be defined as a macro so the preprocessor is inlining as many NOPs as needed based on the value of the argument.

Something like this:
```c
#define REPEAT_1(expr) expr;
#define REPEAT_2(expr) REPEAT_1(expr) expr;
#define REPEAT_3(expr) REPEAT_2(expr) expr;
#define REPEAT_4(expr) REPEAT_3(expr) expr;
// ... define up to the desired maximum unroll factor

// A macro to call the appropriate REPEAT_N macro
#define UNROLL_LOOP(N, expr) REPEAT_##N(expr)

#define ULP_RISCV_DELAY_CCOUNT(cycles) UNROLL_LOOP(cycles, //         __asm__ volatile ("nop"))
```

I can appreciate having a separate macro for short delays is not ideal, but when needed and accuracy is critical, it would do the job. If the delay is too large for the macro, then you'll get a compiler error about REPEAT_<N> not being defined.

Similarly, the delay function itself could be modified to handle the cases of small values, avoiding the need for subtraction. 

```c
void static inline ulp_riscv_delay_cycles(uint32_t cycles) {
    switch(cycles) {
       case 20: __asm__ volatile ("nop");
       case 19: __asm__ volatile ("nop");
       case 18: __asm__ volatile ("nop");
       // etc...

       case 5:   // for very small delays, just return. The exact minimum achievable delay count would be found empirically
       case 4:
       case 3:
       case 2:
       case 1:
       case 0: return
       default:
           <loop, waiting for difference between the current cycle counter and the start cycle counter to become greater or equal to the requested delay count>
}
```