espressif/esp-idf #16893: [IDF 5.4.2] - RGB Buffers in BB mode  desyncing on WiFI/SPI interrupts

Zer0-bit, 2025-07-18:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi - i've  been running my project on the stable arduino-esp32 release (IDF 5.3.2.1 based) and all runs well no bounce buffer desyncing happening, now i updated to the latest (IDF 5.4.2 based) release and i observe heavy buffer desyncing  during wifi or spi operations.

I'm running the LVGL stack on Core0 and Wifi stack on Core1 using tasks and all, my project is in a private github repo  so can't really share a github link but here's my lcd config:

```
esp_err_t EspNativeDisplayController::initializeLCD(lv_disp_drv_t* disp_drv) {
 inintPanelLed(10U);
 // LCD panel configuration
 esp_lcd_rgb_panel_config_t panel_config = {
     .clk_src = LCD_CLK_SRC_PLL240M,
     .timings = {
      .pclk_hz = LCD_PIXEL_CLOCK_HZ,
      .h_res = H_RES,
      .v_res = V_RES,
      .hsync_pulse_width = 4,
      .hsync_back_porch = 8,
      .hsync_front_porch = 8,
      .vsync_pulse_width = 4,
      .vsync_back_porch = 8,
      .vsync_front_porch = 8,
      .flags = {
       .pclk_active_neg = true,
      }
     },
     .data_width = 16,
     #if (ESP_IDF_VERSION > ESP_IDF_VERSION_VAL(5, 0, 0))
     .num_fbs = 2,
     .bounce_buffer_size_px = (15 * LCD_H_RES),
     .dma_burst_size = 64,
     #endif
  .hsync_gpio_num = LCD_PIN_HSYNC,
  .vsync_gpio_num = LCD_PIN_VSYNC,
  .de_gpio_num = LCD_PIN_DE,
  .pclk_gpio_num = LCD_PIN_PCLK,
  .data_gpio_nums = {
      LCD_PIN_DATA0, LCD_PIN_DATA1, LCD_PIN_DATA2, LCD_PIN_DATA3, LCD_PIN_DATA4, LCD_PIN_DATA5, LCD_PIN_DATA6, LCD_PIN_DATA7,
      LCD_PIN_DATA8, LCD_PIN_DATA9, LCD_PIN_DATA10, LCD_PIN_DATA11, LCD_PIN_DATA12, LCD_PIN_DATA13, LCD_PIN_DATA14, LCD_PIN_DATA15
  },
  #if (ESP_IDF_VERSION < ESP_IDF_VERSION_VAL(5, 0, 0))
  .disp_gpio_num = LCD_PIN_DISP_EN,
  #endif
  .flags = {
   .fb_in_psram = 1,
   .bb_invalidate_cache = 1,
  }
 };
```

Also can share a video of the new  behaviour:
https://github.com/user-attachments/assets/968029b1-ffd0-4bb0-b17f-09a070e0ed0d

Zer0-bit, 2025-07-23:
Also tried using this just as a testing point, issue still persists

[image]

Returning back to the below version and no issues
https://github.com/espressif/arduino-esp32/releases/tag/3.1.3

Zer0-bit, 2025-07-26:
Since my last post i saw release v.3.3.0 got promoted to stable, tried that, oc without hope of fix just to have table as a entry point and problem still persists.

Perimec, 2025-08-13:
Same problem here :(

closed 2025-08-20 by Zer0-bit

suda-morris, 2025-08-21:
Hi @Zer0-bit  I'm sorry for the late response. May I know if increase the `bounce_buffer_size_px` can help?