espressif/esp-idf #16910: example to use https_request with a single custom certificate.

greenpdx, 2025-07-18:
### Is your feature request related to a problem?

I am trying to get https_request work with a custom single https server certificate.  I keep getting 0x2700.  

### Describe the solution you'd like.

I would like an "How To" get a single  custom server certificate working with https_request. 

### Describe alternatives you've considered.

I have read everything about CA_bundles and tried to get that to work.

Ashish285, 2025-08-04:
Hi @greenpdx 

We do have an example which shows how to use a custom server certificate, check the [esp_http_client_example.c](https://github.com/espressif/esp-idf/blob/4e036983a751e4667ade94c8f6f6bf1e7f78eff0/examples/protocols/esp_http_client/main/esp_http_client_example.c#L453)

Let us know if this is sufficient for your test.

greenpdx, 2025-08-08:
I have created a github repo that Connects to WiFi and GET a web page from the RPi.
https://github.com/greenpdx/IoTRPiHTTPS

closed 2025-08-08 by greenpdx