espressif/esp-idf #16923: 开启psram后出现crush

Solar914, 2025-07-19:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF:          67c1de1eeb-dirty

### Espressif SoC revision.

esp32s3_n16r8

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp32s3 n16r8

### Power Supply used.

USB

### What is the expected behavior?

能够正常开启psram

### What is the actual behavior?

烧录完毕后出现crush，在反复重启，同样的硬件，使用小智ai的固件，psram并不会出现异常

### Steps to reproduce.

1. 编译后烧录
2. 烧录后启动
3. 不断重启
...

### Debug Logs.

```plain

```

### More Information.

[xiaozhi_config.txt](https://github.com/user-attachments/files/21327386/xiaozhi_config.txt)
[my_config.txt](https://github.com/user-attachments/files/21327387/my_config.txt)

[crash_log.txt](https://github.com/user-attachments/files/21327389/crash_log.txt)

Solar914, 2025-07-19:
ESP-IDF版本-v5.4.0

fatcatkk, 2025-07-20:
我也是同样的问题，有解决方法了吗

Solar914, 2025-07-20:
> 我也是同样的问题，有解决方法了吗

没有，我已经把SPIRAM的配置调整的和小智一样了，还是不行

ginkgm, 2025-07-21:
请先测试一下 v5.4.2 或者最新版本的master看问题是否还存在

ginkgm, 2025-07-21:
@fatcatkk  @Solar914

Solar914, 2025-07-21:
> [@fatcatkk](https://github.com/fatcatkk) [@Solar914](https://github.com/Solar914)

@ginkgm 你好，我这边网络出了点问题，我会尽快更新到5.4.2及以上进行复测，我再补充下，编译小智固件的时候，也是使用的我本地的ESP-IDF v5.4.0，所以应该是哪里配置不同，导致了异常，根据我提供的crash log有一些发现吗？

Solar914, 2025-07-22:
> [@fatcatkk](https://github.com/fatcatkk) [@Solar914](https://github.com/Solar914)

@ginkgm 使用esp-idf v5.5还是会出现crash

Solar914, 2025-07-23:
@ginkgm @methodmissing Could you please help take a look at this issue?

ginkgm, 2025-07-23:
Hi @Solar914 

我使用hello world example编译运行，还是不能复现你的问题，你能不能提供一下完整型号，比如说 ESP32-S3-WROOM-1-N16R8 或者 ESP32-S3-WROOM-2-N16R8V？

我使用了idf提供的工具 idf.py save-defconfig，对小智的和你的config文件做了提取，并对提取的文件做了diff，结果如下。另外一个解决的方法是你再针对这些小智使用的config去做测试，看看使能哪个选项能解决你的问题，这样有助于我们定位问题。

5,9d4
< CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y
< CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF=y
< CONFIG_BOOTLOADER_LOG_LEVEL_NONE=y
< CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS=y
< CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
11,12d5
< CONFIG_COMPILER_CXX_EXCEPTIONS=y
< CONFIG_COMPILER_CXX_EXCEPTIONS_EMG_POOL_SIZE=1024
17a11
> CONFIG_SPIRAM_MEMTEST=n
20d13
< CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=49152
22,31d14
< CONFIG_ESP32S3_INSTRUCTION_CACHE_32KB=y
< CONFIG_ESP32S3_DATA_CACHE_64KB=y
< CONFIG_ESP32S3_DATA_CACHE_LINE_64B=y
< CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=4096
< CONFIG_ESP_MAIN_TASK_STACK_SIZE=8192
< CONFIG_ESP_TASK_WDT_TIMEOUT_S=10
< CONFIG_FREERTOS_GENERATE_RUN_TIME_STATS=y
< CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC=y
< CONFIG_MBEDTLS_DYNAMIC_BUFFER=y
< CONFIG_MBEDTLS_SSL_KEEP_PEER_CERTIFICATE=n

Solar914, 2025-07-24:
我使用的模组如下图

是CONFIG_ESPTOOLPY_FLASHMODE_QIO=y的配置差异导致的，hello world例程中默认配置CONFIG_ESPTOOLPY_FLASHMODE_DIO=y。不开启spi ram例程可以正常跑，开启spi ram之后就异常了，此时配置CONFIG_ESPTOOLPY_FLASHMODE_QIO=y，例程可以正常运行。
这个原因可以帮忙看下吗？

[image]

ginkgm, 2025-07-26:
Hi @Solar914 

我看到你使用了一个第三方的模组。之前确实见过有一些flash存在一些不配置4线则会与psram有冲突的情况。

能提供这个模组的详细信息吗？尤其是他里面使用的flash型号及手册。以及以下命令的输出：

esptool.py flash_id

Solar914, 2025-07-26:
模组是从淘宝买的，问了客服暂时没回复。我理解这个flash应该是合封在esp32s3内部的吧，是不是从乐鑫出厂的时候就确定了，模组厂不会涉及到flash的修改吧？下面是esptool.py flash_id的输出:
esptool.py v4.9.0
Found 1 serial ports
Serial port COM4
Connecting.....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: d0:cf:13:15:94:94
Uploading stub...
Running stub...
Stub running...
Manufacturer: c2
Device: 2018
Detected flash size: 16MB
Flash type set in eFuse: quad (4 data lines)
Flash voltage set by eFuse to 3.3V
Hard resetting via RTS pin...

Solar914, 2025-07-26:
请教一下，我升级到esp-idf 5.5之后，hello world example默认开启了idf_build_set_property(MINIMAL_BUILD ON)，在menuconfig中无法看到esp-psram组件的配置选项，我要如何修改，才可以在最小化编译的情况下，将esp-psram组件的配置显示在menuconfig中呢？

ginkgm, 2025-07-26:
flash 并不是一定合封的，具体需要查阅该模组对应的手册。乐鑫的芯片有合封版本的也有非合封版本的，你可以查阅此处： https://products.espressif.com/#/product-selector?names=&filter={%22Products%22:[%22SoC%22],%22Series%22:[%22ESP32-S3%22]}

该 flash 很可能是旺宏提供的，据我所知，该公司产品有一些型号在非四线（QE=0）模式下，IO3用作复位脚，当使能psram之后，与psram的通信可能会意外造成 flash 的复位。这个解释和你的描述比较像。但在 QE=1 的时候， IO3将不再作为复位输入使用。

请问一下你的芯片在设置完 4 线后再回到 2 线以后还有问题吗？

ginkgm, 2025-07-28:
@Solar914  

因为 flash 的QE位是一个non-volatile位，IDF驱动中对该位的写入也没有撤销的动作，如果我的猜想正确，一旦运行过一次四线的模式，这一位被置1以后，问题就会消失。我想依据这个来佐证我的想法。 由于IDF 驱动不会清除这一位，所以也不需要运行时测试，只需要先烧一次QIO的程序，再烧回DIO的观察现象即可。

Solar914, 2025-07-28:
这是厂家给的模组图片，他们说原理图和你们n16r8一样
[image]

Solar914, 2025-07-28:
flash型号我问了客服，暂时没回复，我在网上没搜到是哪家的片子

kaidegit, 2025-07-29:
顺手帮你找了下，奇奇怪怪的片子增加了（

http://www.fs-rank.com/zh-Hant/products

[T25S128.pdf](https://github.com/user-attachments/files/21478433/T25S128.pdf)

ginkgm, 2025-08-08:
对于非官方支持的 flash，可能需要你们自己进行适配，文档在这： https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_flash/spi_flash_override_driver.html

但对于你现在这个情况，我觉得应该是直接切QIO问题就解决了，是什么原因让你不想用QIO呢？

Solar914, 2025-08-08:
> 对于非官方支持的 flash，可能需要你们自己进行适配，文档在这： https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_flash/spi_flash_override_driver.html
> 
> 但对于你现在这个情况，我觉得应该是直接切QIO问题就解决了，是什么原因让你不想用QIO呢？

用QIO可以的，我只是想搞清楚这是什么原因导致的，可以理解为开启PSRAM后，由于flash和psram共用SPI0的引脚，SPI0在操作psram时，错误的导致flash复位了？

Solar914, 2025-08-08:
另外，这个flash的驱动是包含在ESP-IDF5.4的，毕竟DIO和QIO都能正常工作，只是DIO模式和PSRAM有冲突，是不是可以理解为flash驱动有bug？

ginkgm, 2025-09-24:
@Solar914  请先提供准确的flash手册吧，解释需要从手册上出。

IDF 无法适配所有第三方 flash，建议你联系 https://www.espressif.com/en/contact-us/sales-questions 获取更直接的支持。

ginkgm, 2025-11-19:
Closed due to late reply. Feel free to reopen if you have further comments.