espressif/esp-idf #16949: unit-test-app can't build

nopnop2002, 2025-07-20:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.2

### Espressif SoC revision.

Chip is ESP32D0WDQ6 (revision 1)

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32 development board

### Power Supply used.

USB

### What is the expected behavior?

unit-test-app can build

### What is the actual behavior?

unit-test-app can't build

### Steps to reproduce.

```
cd esp-idf/tools/unit-test-app
idf.py -T all build

/home/nop/esp-idf/components/esp_partition/test/test_partitions.c:75:37: error: format '%x' expects argument of type 'unsigned int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
   75 |         printf("App partition at 0x%x doesn't hold a valid app\n", part->address);
      |                                    ~^                              ~~~~~~~~~~~~~
      |                                     |                                  |
      |                                     unsigned int                       uint32_t {aka long unsigned int}
      |                                    %lx
/home/nop/esp-idf/components/esp_partition/test/test_partitions.c: In function 'test_func_99':
/home/nop/esp-idf/components/esp_partition/test/test_partitions.c:109:45: error: format '%x' expects argument of type 'unsigned int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  109 |         printf("Hashing app partition at 0x%x\n", part->address);
      |                                            ~^     ~~~~~~~~~~~~~
      |                                             |         |
      |                                             |         uint32_t {aka long unsigned int}
      |                                             unsigned int
      |                                            %lx
/home/nop/esp-idf/components/esp_partition/test/test_partitions.c: In function 'test_func_119':
/home/nop/esp-idf/components/esp_partition/test/test_partitions.c:136:27: error: format '%d' expects argument of type 'int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  136 |     printf("flash size = %d bytes\n", size_flash_chip);
      |                          ~^           ~~~~~~~~~~~~~~~
      |                           |           |
      |                           int         uint32_t {aka long unsigned int}
      |                          %ld
/home/nop/esp-idf/components/esp_partition/test/test_partitions.c:139:36: error: format '%d' expects argument of type 'int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  139 |     printf("available page pool = %d pages\n", page_reservation_count);
      |                                   ~^           ~~~~~~~~~~~~~~~~~~~~~~
      |                                    |           |
      |                                    int         uint32_t {aka long unsigned int}
      |                                   %ld
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.
HINT: The issue is better to resolve by replacing format specifiers to 'PRI'-family macros (include <inttypes.h> header file).
ninja failed with exit code 1, output of the command is in the /home/nop/esp-idf/tools/unit-test-app/build/log/idf_py_stderr_output_7775 and /home/nop/esp-idf/tools/unit-test-app/build/log/idf_py_stdout_output_7775
```

### Debug Logs.

```plain

```

ESP-Marius, 2025-07-21:
The unit-test app is a bit of a legacy system, we dont use it internally anymore for testing IDF.

But seems like we missed a test that should have been migrated to the new test-apps approach. THanks for letting us know!

ESP-Marius, 2025-08-27:
@dpnebert we use a more modular approach where every component has their own test-app(s) with their own sdkconfigs. Then tests are run automatically with pytest-embedded. 

We also recently introduced [idf-ci](https://github.com/espressif/idf-ci) to make it easier to set-up and do this. E.g. with a simple command you can build every single test-app with all its sdkconfig configuration for all desired targets. This then easily integrates with pytest-embedded for test-case runing which can automatically run all these apps.

More docs on how to do this is on our plan for the near future.

ESP-Marius, 2025-08-29:
It will build apps based on `sdkconfig.defaults` and `sdkconfig.ci.X` files. So you need to set any needed configs in those.

It will do a clean build everytime, so your sdkconfig values doesnt take effect

dpnebert, 2025-08-29:
Thank you!

I'll have to open a issue for this because I am now getting Windows errors trying to build:
```
idf-ci build run -t esp32
[08/29/25 10:50:01] INFO     Building all applications
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\Dan\.espressif\python_env\idf5.4_py3.11_env\Scripts\idf-ci.exe\__main__.py", line 7, in <module>
  File "C:\Users\Dan\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\click\core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Dan\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\click\core.py", line 1082, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\Dan\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\click\core.py", line 1697, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Dan\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\click\core.py", line 1697, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Dan\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\click\core.py", line 1443, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Dan\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\click\core.py", line 788, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Dan\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Dan\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\idf_ci\cli\build_group.py", line 73, in run
    click.echo(f'Built the following apps in {time.time() - start_time:.2f} seconds:')
  File "C:\Users\Dan\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\click\utils.py", line 318, in echo
    file.write(out)  # type: ignore
    ^^^^^^^^^^^^^^^
  File "C:\Users\Dan\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\click\_compat.py", line 542, in _safe_write
    return _write(s)
           ^^^^^^^^^
  File "C:\Users\Dan\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\colorama\ansitowin32.py", line 47, in write
    self.__convertor.write(text)
  File "C:\Users\Dan\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\colorama\ansitowin32.py", line 177, in write
    self.write_and_convert(text)
  File "C:\Users\Dan\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\colorama\ansitowin32.py", line 205, in write_and_convert     
    self.write_plain_text(text, cursor, len(text))
  File "C:\Users\Dan\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\colorama\ansitowin32.py", line 210, in write_plain_text      
    self.wrapped.write(text[start:end])
  File "C:\Users\Dan\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\click\_winconsole.py", line 192, in write
    return self._text_stream.write(x)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Dan\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\click\_winconsole.py", line 177, in write
    raise OSError(self._get_error_message(GetLastError()))
OSError: Windows error 6
```