espressif/esp-idf #16969: ESP32-C5: Large BSS objects placed at incorrect memory addresses causing overlap

sgorsh, 2025-07-20:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-rc1

### Espressif SoC revision.

ESP32-C5

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-C5-DevKitC-v1.2

### Power Supply used.

USB

### What is the expected behavior?

Large static objects in `.dram0.bss` section are being placed at incorrect memory addresses, causing severe memory overlap with other sections. A 8KB decompressor object is placed 0x113C bytes *before* the `.dram0.bss` section start, overlapping multiple memory regions.

### What is the actual behavior?

8KB decompressor object is placed outside and before its section, causing memory corruption.

### Steps to reproduce.

1. Clone the repository
2. Build with `idf.py build`
3. Run `memorymap.cmd` to generate memory analysis
4. Check `memory.txt` - decompressor at `0x4080c064` overlaps with `.dram0.bss` section starting at `0x4080b1a0`

### Debug Logs.

```plain
Waiting to prevent fast boot loop...
Starting decompression...
Decompressor address: 0x4080c064
Decompression finished
Starting decompression...
Decompressor address: 0x4080c064
Decompression finished
Starting decompression...
Decompressor address: 0x4080c064
Decompression
ESP-ROM:esp32c5-eco2-20250121
Build:Jan 21 2025
rst:0x1a (CPU_LOCKUP),boot:0x18 (SPI_FAST_FLASH_BOOT)
Core0 Saved PC:0x42000c14
--- 0x42000c14: panic_print_str at C:/Users/SimonG/esp/v5.5/esp-idf/components/esp_system/panic.c:144
SPI mode:DIO, clock div:1
load:0x408556b0,len:0x1754
load:0x4084bba0,len:0xd9c
load:0x4084e5a0,len:0x3090
entry 0x4084bbaa
```

igrr, 2025-07-20:
> extends to 0x4080e110, which is 0x18 bytes past the .dram0.bss section boundary (0x4080e4f8),

I'm sorry, isn't it the other way around? 0x4080e4f8 is larger than 0x4080e110 by 0x3e8 bytes? Seems like the `decompressor` object is well within the .dram0.bss section.

sgorsh, 2025-07-20:
Sorry, I've messed up address calculations. That's what I get for relying on LLMs to do calculations 😅
Problem is actually with `miniz` decompression code. Closing this issue.

In case anyone is interested:  memory after decompressor object gets corrupted and next function after decompress throws error or instruction pointer gets corrupted.
Here's the test output:
```
Starting decompression...
Decompressor address: 0x4080a0b0
Decompressor size: 0x20ac (8364 bytes)
Decompressor end: 0x4080c15c
Guard address: 0x4080c19c, value: 0xdeadbeef
Output buffer address: 0x4080df54
Input data length: 43
Output buffer size: 1024
Calling tinfl_decompress...
tinfl_decompress returned: 0
Bytes consumed: 43
Bytes produced: 96
Guard check: 0x06120602 (should be 0xDEADBEEF)
OVERFLOW DETECTED: 64 bytes written past decompressor end
First 16 padding bytes: 02 06 12 06 05 04 03 06 00 06 11 06 05 04 04 06
Decompression finished
```

closed 2025-07-20 by sgorsh