espressif/esp-idf #16970: how is the CONFIG_SPIRAM_XIP_FROM_PSRAM supposed to work?

kdschlosser, 2025-07-20:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

This is a question but it may also possibly be a bug. There isn't any real documentation that explains fully what this config option does. The only information that is given is that when using the ESP-LCD component is that it might need to be used to get the display to run properly.

`CONFIG_SPIRAM_XIP_FROM_PSRAM=y`

There are some RGB displays that will not run properly without this command. These displays are displays that have a minimum pixel clock of 16mhz. Having a higher pixel clock is always a beneficial thing to have as the display is able to update faster the cost of using that config option is HUGE. Turning it on causes 2mb of SPIRAM to be used. It's as if the entire firmware is being loaded into SPIRAM. The project that I am working on has a compiled binary size of roughly 2.5mb and with that setting turned off all SPIRAM is available for use but if it needs to be tuned on if a user is using an MCU that only has 2MB of SPIRAM available it crashes when the code is ran on the MCU because it runs out of memory.

So what exactly is getting loaded into SPIRAM that it would take up 2mb? Is there a way to control what gets loaded into SPIRAM when using that config option? Is it supposed to be taking up that much SPIRAM?

Using that config option does allow for setting higher pixel clock speeds and that is a fantastic thing to have but the cost is simply too high to have turned on for anyone using an MCU that has less then 8MB of SPIRAM. That is going to remove a lot of prebuilt RGB displays that have an ESP32 onboard if the ESP32 has 4mb and lower SPIRAM

BitsForPeople, 2025-07-20:
https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/external-ram.html#external-ram-config-xip

kdschlosser, 2025-07-20:
I know that the config option is a macro for CONFIG_SPIRAM_FETCH_INSTRUCTIONS and CONFIG_SPIRAM_RODATA but It states that when having issues with  an RGB bus display to enable CONFIG_SPIRAM_XIP_FROM_PSRAM options. It looks like it is more of a global option across the board instead of only moving the bits related to the RGB driver in SPIRAM it's moving every piece of read only code and all of the IRAM functions into SPIRAM. That's an off the wall thing to do if only needing to have the rgb bus driver related bits be faster. The memory use becomes massive by moving all the read only data and all of the IRAM functions into SPIRAM when it doesn't need to be done.

kdschlosser, 2025-07-20:
so my question is how is it that there is 2mb of memory use when the compiled binary is 2.5mb in size. That would mean that the compiled size of by binary is 80% of the binary is read only and ISR functions. So basically it is moving the entire executable code into SPIRAM?

igrr, 2025-07-20:
> So basically it is moving the entire executable code into SPIRAM?

Yes, all except the parts already loaded into the internal RAM (code placed into IRAM as well as the .data segment).

kdschlosser, 2025-07-21:
Oh well that just completely silly.

kdschlosser, 2025-07-21:
For the purposes of the rgb display only the portions of the code that deal with writing the data to the display should be moved into RAM, not the entire firmware.

kdschlosser, 2025-07-21:
The issue with the RGB displays is not how fast the user code it able to write to a buffer but how fast the code that transfers the buffer to the display is able to be executed so the problem lies in the driver and not in user code. When using 2 frame buffers to send to the display regardless of where the program that is responsible for filling is sitting the filling of that frame buffer is not going to effect how fast the driver is able to send the second buffer to the display. the only thing that is going to effect that is going to be the driver code.

igrr, 2025-07-21:
You are right that it would be nice to be able to move just part of the code into PSRAM while executing the rest from Flash. However, IDF doesn't have such capability yet at framework level. This might be something we could add in the future. There is no concrete plan in place to do this, though. For now you do need to pick a chip with larger PSRAM size if you rely on this feature in your product.

Another thing to note is that the 2nd benefit of the "XIP from PSRAM" feature mentioned [in this part of the docs](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/external-ram.html#execute-in-place-xip-from-psram) — that tasks and ISRs don't have to be suspended for the duration of Flash read/write operations — can only be achieved if the _entire_ code and .rodata segments are moved from Flash to PSRAM. If only part of the code is moved into PSRAM, Flash cache still has to be disabled for the duration of Flash read/write operations. For GUI applications this means that the UI updates may lag or stutter while Flash operations (such as NVS or filesystem operations) are performed. Only by moving all the code into PSRAM can this issue be eliminated.

kdschlosser, 2025-07-21:
Or if you use double buffering and render to the buffer that is not being transmitted. There would be no apparent studdering or any other anomalies if it is done this way.

I actually use 4 frame buffers when dealing with an RGB display. 2 are full frame buffers and the other 2 are partials that are about 1/10th the size of the full buffers. one core handles copying the data from a partial buffer while the other core is rendering to the second partial buffer. When a stream of partial buffers that makes up the changes needed for a single fame that first core then swaps the 2 buffers with the RGB driver. as the buffer is being transmitted the data is also being copied out of it and written to the now idle full frame buffer.. once that is done then that core goes back to doing the copying of the partial buffers. 

There are RGB displays that simply will not operate at lower pixel clock speeds and currently the only way to get those displays to work is by using that config parameter. It's not all of the code that is causing the issue. It is only the driver code that is not able to do what it needs to do. It's almost like the RGB driver is a bitbang driver and not a hardware driver which is limited by the available processor time. This should not be the case when dealing with frame buffers that are placed in DMA memory. it should literally going through the data exactly how it is stored in memory so it should be really fast because it should not be impacted by available processor time. 

I am not sure why it is not able to keep up.

ginkgm, 2025-07-28:
Hi @kdschlosser ,

Do you have other request to the XIP? or we will treat this as a request to the LCD driver. Current title doesn't help people to recognize this as a request to the driver.

kdschlosser, 2025-07-28:
No that would basically it.

Thanks for the clarification on what it does. I feel it is a little bit overboard to move the entire program from flash into PSRAM for the sole purpose of speeding up access to only a small handful of functions in the RGB LCD IO driver. I guess that is how it's made there is nothing I can do about it. 🤯

suda-morris, 2025-08-05:
Hi @kdschlosser 

The limitation is in the hardware. When you're rendering the frame buffer A, you're writing to the PSRAM by CPU cache.
When the LCD controller is flushing the frame buffer B, it is reading from the PSRAM by DMA. Read and write from/to PSRAM cannot be executed in parallel, because they're on the same SPI bus.

Meanwhile, the SPI bus has another device, the main flash. CPU will fetch data and instruction from there, by the cache. The flash and PSRAM on the same SPI bus has the same priority, the SPI bus bandwidth are 1/2 for each (this is not configurable, sadly.)

If we disable the cache from fetching instruction from the main flash, then the SPI bus is almost consumed by the PSRAM only (although the filesystem API can still occupy the SPI bus, but that not happen frequently). Instruction, data are all read from PSRAM. The speed is faster than fetching from the flash.

That's why we recommend enabling the XIP_PSRAM feature when you use RGB LCD driver. Because the RGB LCD needs a large amount of bandwidth. It's not a must for the LCD driver, but a suggestion, on esp32s3. And the XIP PSRAM feature is also available on esp32s2 (and later chips). It's not designed for LCD driver. It's general feature.

BTW, Placing functions/data into PSRAM selectively is still a good feature to have.

kdschlosser, 2025-08-05:
I could have sworn I came across something in the ESP-IDF that does selectively load functions into PSRAM. Now I don't remember where I came across it. I probably saw something and read it wrong.

ginkgm, 2025-09-24:
I think there is no further discussion here. Issue closed. Please create a new issue if you have other requests.