espressif/esp-idf #16973: High-level command for full restart of the microcontroller.

alex000090, 2025-07-21:
### Is your feature request related to a problem?

I need make software restart of the ESP32-S3 chip with all periphery.
For now I use I2C and after restart this one doesn't work.

### Describe the solution you'd like.

Fully restart of the chip with all periphery.

### Describe alternatives you've considered.

I tried to use esp_restart() function as well as tried to restart via WatchDog ( while (true) ).

### Additional context.

I found the solution for ESP32-S3. According the ESP32-S3 TDS ("Reset and Clock" section), full restart of the chip with periphery can be made via set RTC_CNTL_SW_SYS_RST bit of RTC_CNTL_RTC_OPTIONS0_REG register by the command REG_WRITE(0x60008000, 1ul<<31).

ginkgm, 2025-08-08:
Hi @alex000090 ,

The reason why it's not described anywhere is that it is served as an internal API of ESP-IDF and not recommend to be used as a general reset. 

Let's look into the i2c issue -- what is left over after a CPU reset. Could you provide the code you use to reproduce the issue?

alex000090, 2025-08-21:
> What version did you use? I tried on master with following code, there is nothing wrong.
> 
> ```
> void app_main(void)
> {
>     i2c_master_bus_config_t i2c_bus_config = {
>         .clk_source = I2C_CLK_SRC_DEFAULT,
>         .i2c_port = PORT_NUMBER,
>         .scl_io_num = SCL_IO_PIN,
>         .sda_io_num = SDA_IO_PIN,
>         .glitch_ignore_cnt = 7,
>     };
>     i2c_master_bus_handle_t bus_handle;
>     i2c_new_master_bus(&i2c_bus_config, &bus_handle);
> 
>     esp_err_t ret = ESP_OK;
>     for (uint8_t i = 0x01; i < 0x7F; i++) {
>         ret = i2c_master_probe(bus_handle, i, -1);
>         if (ret == ESP_OK) {
>             printf("The slave has been found, the address is %x\n", i);
>             break;
>         }
>     }
> 
>     esp_restart();
> 
> }
> ```

Dear mythbuster5, thanks for the comment!

This code works well and it closely resembles my code. So the problem is in some other place of my code I think. Unfortunately, I can't show it.

Thanks all for assistance!

closed 2025-08-21 by alex000090