espressif/esp-idf #16975: PHY Testing of ESP32-C5 devkit-1-V1.2 ECO2

loraadh, 2025-07-21:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi ,

I am testing C5 using the iperf project (in examples). while running RX PER tests, I've noticed the following issues:
1. IEEE802.11ax (HE) is not functioning with LDPC. I need to set my AP (tester) to BCC to make the PER test work. knowing that, according to IEEE802.1, LDPC must replace BCC.
2. 11n is not functioning at 40MHz bandwidth unless I configure my router to force the 40MHz mode (instead of 20MHz). otherwise, it keeps connecting to the router at 20MHz. I configure the device using the commands wifi_protocol and wifi_bandwidth. 

I am using this version:
iperf> version
IDF Version:v5.5-rc1
Chip info:
        model:ESP32-C5
        cores:1
        feature:/802.11bgn/BLE/External-Flash:2 MB
        revision number:100

BR

QingzhaoYin, 2025-08-07:
@loraadh  Hi～ 

Thank you for your feedback.

1. The ESP32-C5 only supports BCC; LDPC is not currently supported.

2. The actual connection bandwidth between the ESP32-C5 and the AP is determined through capability negotiation. Even if the ESP32-C5 is configured to use 40 MHz via esp_wifi_set_bandwidths(), it will still fall back to 20 MHz if the AP only supports or operates in 20 MHz mode.

loraadh, 2025-08-07:
Hi @QingzhaoYin 
thanks for the clarification.
1. however, according to WiFi6 11ax, BCC remains only for legacy, for 11ax LDPC must replace BCC. therefore, having 11ax with BCC is like not having 11ax at all. do you agree with that?
2. yes, you are right. But my point is that even if the router supports both 40M and 20MHz, i need to configure the router to be in a "force 40Mz" state. this is not the case of many wifi wifi sta devices that i've tested using the same router.
BR

loraadh, 2025-08-08:
thanks a lot for your response @QingzhaoYin . i will close this ticket

closed 2025-08-08 by loraadh