espressif/esp-idf #16979: LEDC's ledc_set_duty() processing time is long in ESP-IDF V5.4.2

choejoongho, 2025-07-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF V5.4.2

### Espressif SoC revision.

ESP32-C6

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DEVKITM-1

### Power Supply used.

USB

### What is the expected behavior?

I want to generate a 38kHz IR signal using LEDC of ESP32-C6.

### What is the actual behavior?

I used LEDC to generate IR signals in ESP-IDF V5.4.1. I generated a 38kHz PWM signal without any problems. However, when I upgraded to ESP-IDF V5.4.2, I encountered a problem where the timing of the IR signal was not correct. In the function that maintains a 38kHz PWM signal for 560usec, I found that the period of 560usec was extended to about 750usec.

The same symptom occurs in ESP-IDF V5.4.2 and V5.5. I have attached a code and a picture using a simple basic example for generating LEDC signals.

Please let me know if there is anything that needs to be changed or added to the LEDC settings.

Example Code ledc_test.cpp

```
#include <stdio.h>
#include <string.h>
#include <inttypes.h>

#include "esp_timer.h"          // for esp_timer_get_time
#include "esp_log.h"
#include "esp_err.h"

#include "driver/gpio.h"
#include "driver/ledc.h"
#include "esp_err.h"

#define LEDC_TIMER              LEDC_TIMER_0
#define LEDC_MODE               LEDC_LOW_SPEED_MODE
#define LEDC_OUTPUT_IO          GPIO_NUM_21 // Define the output GPIO
#define LEDC_OUTPUT_INVERT      false  // IR LED Pin Invert
#define LEDC_CHANNEL            LEDC_CHANNEL_0
#define LEDC_DUTY_RES           LEDC_TIMER_11_BIT // Set duty resolution to 11 bits
#define LEDC_DUTY               (682) // Set duty to 33.33%. (2 ** 11) * 33.33% = 682
#define LEDC_FREQUENCY          (38000) // Frequency in Hertz. Set frequency at 38 kHz

static const char *TAG = "app_main";

void delayMicroseconds(uint32_t us)
{
    uint64_t m = (uint64_t)esp_timer_get_time();
    if (us)
    {
        uint64_t e = (m + us);
        if (m > e)
        {
            // overflow
            while ((uint64_t)esp_timer_get_time() > e)
            {
                asm volatile("nop");
            }
        }
        while ((uint64_t)esp_timer_get_time() < e)
        {
            asm volatile("nop");
        }
    }
}

static void example_ledc_init(void)
{
    // Prepare and then apply the LEDC PWM timer configuration
    ledc_timer_config_t ledc_timer = {
        .speed_mode       = LEDC_MODE,
        .duty_resolution  = LEDC_DUTY_RES,
        .timer_num        = LEDC_TIMER,
        .freq_hz          = LEDC_FREQUENCY,  // Set output frequency at 4 kHz
        .clk_cfg          = LEDC_AUTO_CLK
    };
    ESP_ERROR_CHECK(ledc_timer_config(&ledc_timer));

    // Prepare and then apply the LEDC PWM channel configuration
    ledc_channel_config_t ledc_channel = {
        .gpio_num       = LEDC_OUTPUT_IO,
        .speed_mode     = LEDC_MODE,
        .channel        = LEDC_CHANNEL,
        .intr_type      = LEDC_INTR_DISABLE,
        .timer_sel      = LEDC_TIMER,
        .duty           = 0, // Set duty to 0%
        .hpoint         = 0,
        .flags = {
            .output_invert = LEDC_OUTPUT_INVERT
        }
    };
    ESP_ERROR_CHECK(ledc_channel_config(&ledc_channel));
}

extern "C" void app_main(void)
{
    // Set the LEDC peripheral configuration
    example_ledc_init();

    uint64_t now = (uint64_t)esp_timer_get_time();
    ESP_ERROR_CHECK(ledc_set_duty(LEDC_MODE, LEDC_CHANNEL, LEDC_DUTY));
    ESP_LOGI(TAG, "S: %" PRIi64, (uint64_t)esp_timer_get_time() - now);
    ESP_ERROR_CHECK(ledc_update_duty(LEDC_MODE, LEDC_CHANNEL));

    uint64_t delay = (uint64_t)esp_timer_get_time();
    delayMicroseconds(560);
    ESP_LOGI(TAG, "D: %" PRIi64, (uint64_t)esp_timer_get_time() - delay);

    uint64_t curr = (uint64_t)esp_timer_get_time();
    ESP_ERROR_CHECK(ledc_set_duty(LEDC_MODE, LEDC_CHANNEL, 0));
    ESP_ERROR_CHECK(ledc_update_duty(LEDC_MODE, LEDC_CHANNEL));
    ESP_LOGI(TAG, "R: %" PRIi64, (uint64_t)esp_timer_get_time() - curr);
}

```

ESP-IDF V5.4.1
[image]

[image]

ESP-IDF V5.4.2 and V5.5 without log print
The ledc_set_duty() function used to take about 20usec of processing time, but it takes about 180usec in V5.4.2.

[image]

[image]

### Steps to reproduce.

Build -> Download -> check GPIO 21 pin signal

### Debug Logs.

```plain

```

songruo, 2025-07-23:
Hi @choejoongho I believe this is due to this bugfix: https://github.com/espressif/esp-idf/commit/a6b47fac12e3bf185683f72e1446a9e60cacdddc. This will increase the processing time of `ledc_set_duty` function. However, why would you care the time spent in `ledc_set_duty`? We never make guarantee to the time it takes to process the APIs will always be the same. As a better choice, you can use `ledc_stop` to pause the signal.

choejoongho, 2025-07-23:
Hi @songruo Thank you for your answer

For general LED control, 560usec or 1000usec is not important because people cannot perceive it, but the reason why the timing of LEDC is important is because LEDC is used to control IR LED and transmit IR commands. According to the NEC IR transmission characteristics, a 38kHz pulse is turned on and off for a set period of time and displayed as 1 or 0. However, various products such as air conditioners and TVs use On and Off times according to their own standards, so precise time intervals such as 480usec, 500usec, and 560usec,... are required.

[image]

Of course, it seems appropriate to use the RMT block for IR signal transmission, but since the length of the IR command varies from company to company such as 28bit, 48bit, 56bit, 128bit, and 256bit,... it is difficult to match the timing with the RMT block with a limited buffer length, so LEDC, which is convenient for PWM signal generation, has been used.

Depending on the home appliance, the time interval between On and Off states can have a tolerance of about 10%, but the home appliance does not recognize the 35% change from 560usec to 750usec. Is it possible to unilaterally deduct 200usec? Or is there a way to transmit IR commands of various lengths, such as 28bit to 256bit, without discontinuity using the RMT block?

Kainarx, 2025-07-24:
BTW, theoretically the buffer length shouldn't be a limitation as we can use ping pong transfers.

closed 2025-07-24 by songruo