espressif/esp-idf #16980: heap_trace fails to trace memory allocated by malloc

robcholz, 2025-07-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Espressif SoC revision.

ESP32-S3 rev v0.2

### Operating System used.

macOS

### How did you build your project?

CLion IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-N8R2

### Power Supply used.

USB

### What is the expected behavior?

heap trace component should trace all the heap memory including ones allocated by malloc as the example described in [https://github.com/espressif/esp-idf/blob/release/v5.4/examples/system/sysview_tracing_heap_log/README.md](url).

### What is the actual behavior?

the heap trace dump shows that it only tracks the memory explicitly allocated by heap_caps_malloc.

### Steps to reproduce.

#include <unity.h>

#include <esp_heap_caps.h>
#include <esp_heap_trace.h>

static void testHeapTracer() {
    static heap_trace_record_t record[100];
    heap_trace_init_standalone(record,100);
    heap_trace_start(HEAP_TRACE_LEAKS);
    printf("%d\n",heap_caps_get_free_size(MALLOC_CAP_INTERNAL));

    void* ptr1 = malloc(128);
    void* ptr2 = heap_caps_malloc(256, MALLOC_CAP_DEFAULT);
    auto leaked = malloc(150);
    leaked = malloc(100);
    free(ptr1);
    heap_caps_free(ptr2);

    printf("%d\n",heap_caps_get_free_size(MALLOC_CAP_INTERNAL));

    heap_trace_stop();
    heap_trace_dump();
}

extern "C" void app_main() {
    UNITY_BEGIN();

    RUN_TEST(testHeapTracer);

    UNITY_END();
}

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2f00
entry 0x403c8920
I (27) boot: ESP-IDF 5.4.1 2nd stage bootloader
I (27) boot: compile time Jul 22 2025 12:29:41
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (30) boot: efuse block revision: v1.3
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 8MB
I (45) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (65) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (71) boot:  2 app0             OTA app          00 10 00010000 002e0000
I (78) boot:  3 littlefs         Unknown data     01 83 002f0000 00400000
I (84) boot:  4 coredump         Unknown data     01 03 006f0000 00010000
I (91) boot: End of partition table
I (94) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0d898h ( 55448) map
I (111) esp_image: segment 1: paddr=0001d8c0 vaddr=3fc93d00 size=02758h ( 10072) load
I (114) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=2a6d4h (173780) map
I (147) esp_image: segment 3: paddr=0004a6fc vaddr=3fc96458 size=00668h (  1640) load
I (148) esp_image: segment 4: paddr=0004ad6c vaddr=40374000 size=0fc48h ( 64584) load
I (165) esp_image: segment 5: paddr=0005a9bc vaddr=600fe100 size=0001ch (    28) load
I (171) boot: Loaded app from partition at offset 0x10000
I (171) boot: Disabling RNG early entropy source...
I (183) esp_psram: Found 2MB PSRAM device
I (183) esp_psram: Speed: 40MHz
I (183) cpu_start: Multicore app
I (602) esp_psram: SPI SRAM memory test OK
I (611) cpu_start: Pro cpu start user code
I (611) cpu_start: cpu freq: 160000000 Hz
I (611) app_init: Application information:
I (611) app_init: Project name:     Pixu
I (615) app_init: App version:      d05c181-dirty
I (619) app_init: Compile time:     Jul 22 2025 12:29:23
I (624) app_init: ELF file SHA256:  ff9a741da...
I (629) app_init: ESP-IDF:          5.4.1
I (632) efuse_init: Min chip rev:     v0.0
I (636) efuse_init: Max chip rev:     v0.99 
I (640) efuse_init: Chip rev:         v0.2
I (644) heap_init: Initializing. RAM available for dynamic allocation:
I (650) heap_init: At 3FC98490 len 00051280 (324 KiB): RAM
I (655) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (660) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (666) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (671) esp_psram: Adding pool of 2048K of PSRAM memory to heap allocator
I (679) spi_flash: detected chip: boya
I (681) spi_flash: flash io: dio
I (684) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (690) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (697) coexist: coex firmware version: e727207
I (721) coexist: coexist rom version e7ae62f
I (722) main_task: Started on CPU0
I (723) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (725) main_task: Calling app_main()
371035
371035
====== Heap Trace: 0 records (100 capacity) ======
====== Heap Trace Summary ======
Mode: Heap Trace Leaks
0 bytes 'leaked' in trace (0 allocations)
records: 0 (100 capacity, 1 high water mark)
total allocations: 1
total frees: 1
================================
test/test_pixu.cpp:339:testHeapTracer:PASS
```

BitsForPeople, 2025-07-22:
`auto leaked = malloc(150);` is likely optimized away by the compiler.

BitsForPeople, 2025-07-22:
Try _using_ the 'leaked' pointer in a way that prevents the compiler from omitting the malloc() call to see the difference. Pass it to some external function (e.g. prtintf...) or assing it to a global volatile variable.

KonstantinKondrashov, 2025-07-24:
Thanks for the suggestion. We can try to add `__attribute__((malloc))` for heap_caps_malloc. I will make the changes.

KonstantinKondrashov, 2025-07-24:
The `malloc` attribute does not help the compiler recognize unused values and optimize them away in the same way as malloc.

```c
__attribute__((malloc)) void* my_malloc(size_t size)  {
    return malloc(size);
}

static void testHeapTracer(void) {
    static heap_trace_record_t record[100];
    heap_trace_init_standalone(record, 100);
    heap_trace_start(HEAP_TRACE_ALL);

    void* ptr1 = my_malloc(128);
    free(ptr1);
    
    heap_trace_stop();
    heap_trace_dump();
}
```

BitsForPeople, 2025-07-24:
Yeah, didn't really expect it would. However, supposedly the `malloc` attribute helps the compiler with alias analysis by letting it know that the returned pointer is never an alias of any other live object. And the `NODISCARD` should issue a warning if someone manages to call `heap_caps_malloc()` without 'capturing' its result in some way.