espressif/esp-idf #16982: UART0 Output Data Loss at 115200 bps (ESP_LOG Corruption)

italocjs, 2025-07-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### Problem Statement
When using UART0 (GPIO1/GPIO3) for output on ESP-IDF, printed data intermittently loses segments of output. This occurs at 115200 bps, regardless of whether the UART driver is explicitly started for port 0 or if default IDF printing is used (i.e., no driver start for printing).

#### Example Output
- **Expected:** `I (1293) testTag: 123465789ABCDEF123465789ABCDEF123465789ABCDEF`
- **Actual:**   `I (1293) testTag: 123465789ABCDEF123465789123465789ABCDEF`

I also print data as json using printf, example:
- **Expected:** 
{
  "timestamp": 1293,
  "tag": "testTag",
  "data": "123465789ABCDEF123465789ABCDEF123465789ABCDEF"
}

- **Actual:** `
{
  "timestamp": 1293,
  "t: "testTag",
  "data":BCDEF123465789ABCDEF123465789ABCDEF"
}

No other peripherals are enabled, only main task exists.

Missing segments appear consistently in the output stream, using both printf and ESP_LOGx.
This behavior is not new to IDF5, i also seen this on IDF4.4

### Scope of the Issue
- Reproducible across 20+ development modules (mix of CH340 and CP2102 USB-UART bridges) and proprietary boards (original CH340, sourced reliably).
- Issue does **not** occur on UART1 or UART2 (any pins other than GPIO1/GPIO3).
- Multiple USB cables tested, same behavior.
- Issue is intermittent.
- Tested on both Windows and Linux, issue is consistent.

### Additional Observations
- Flashing the board over UART0 at 921000 bps works flawlessly every time.
- However, ESP_LOG output at 115200 bps is consistently corrupted/missing segments.
- Power is STABLE, monitored via oscilloscope to be sure.  both the ESP32 and CH340 have decoupling capacitors and a reliable power source.

### Request for Suggestions
I find it odd that flashing at much higher speeds (921000) works reliably, but printing/logging is corrupted at a lower baud rate. Any suggestions on how to further diagnose or resolve this issue?  At the same time i don't think is a software issue as surely others would notice, but i'm not sure what in hardware could be causing this, as both common esp32 dev modules and proprietary boards have the same behavior.

---

If additional logs or test scenarios are needed, please let me know.

ginkgm, 2025-07-23:
Hi @italocjs ,

Have you tried latest IDF master to see if the issue exists?

And could ou provide a piece of code to reproduce the issue?

kurozz, 2025-09-22:
@italocjs As a workaround, try using another UART, such as UART1, for log output.
You should be able to set it to the same pins, but may have some corruption during the change of UART ports.

You can also avoid using oneshot ADC and use continuous ADC instead.

agzzzz, 2026-01-05:
Hello @ginkgm, thank you for reviewing this problem.

I do not have an official board with the ESP32-WROOM to test with. My tests were with a generic development board and one of our products.
I do have an ESP32_LyraT board (ESP32-WROVER-E).

The problem also occurs with the LyraT, using ESP-IDF v5.4.3. You do have to change the core affinity to 1, as my test code defaults to core 0, but you said that was done. Unfortunately, I do not know what else could be affecting the test.