espressif/esp-idf #16984: Access fault in r_ble_lll_conn_txbuf_insert_after() on ESP32-C6

kenh-sq, 2025-07-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

commit f10ac3eec22efb1b892f0c8a9f5e871940185736

### Espressif SoC revision.

ESP32-C6, chip rev: v0.0

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

No faults/crashes. Does not occur with `libble_app.a` from [this comment](https://github.com/espressif/esp-idf/issues/15696#issuecomment-2908841270) at commit fd37cd46b203786114b6f2793d62fba6da437a62

### What is the actual behavior?

Hard to reproduce, but when it does occur:

Load access fault when performing `lhu a5, 0(a4)` within `r_lll_ble_conn_txbuf_insert_after()` following call to `r_ble_lll_sched_get_realtime_info()`.

Localized disassembly within the function:
```
...
0x4080bcd8  addi a2,sp,8 
0x4080bcda  addi a1,sp,12 
0x4080bcdc  addi a0,sp,4 
0x4080bcde  jal 0x4080483c <r_ble_lll_sched_get_realtime_info> 
0x4080bce2  lw a5,4(sp) 
0x4080bce4  beqz a5,0x4080bc20 <r_ble_lll_conn_txbuf_insert_after+78> 
0x4080bce6  lw a4,584(s0) 
0x4080bcea  bne a5,a4,0x4080bc20 <r_ble_lll_conn_txbuf_insert_after+78> 
0x4080bcee  lw a4,8(sp) 
0x4080bcf0  lhu a5,0(a4) 
0x4080bcf4  andi a5,a5,1 
0x4080bcf6  beqz a5,0x4080bc5a <r_ble_lll_conn_txbuf_insert_after+136> 
0x4080bcf8  bne a4,s1,0x4080bc5a <r_ble_lll_conn_txbuf_insert_after+136> 
...
```

With a breakpoint at 0x4080bcf0, a4 is always 0x00000000 when the breakpoint is hit. And always with the below call stack:
```
#0  0x4080bcf0 in r_ble_lll_conn_txbuf_insert_after ()
#1  0x4080be48 in r_ble_lll_conn_insert_txbuf_ahead ()
#2  0x42062ed2 in r_ble_lll_conn_append_tx_buffer ()
#3  0x42041e96 in r_ble_ll_conn_prepare_tx_pdu.part.1 ()
#4  0x42042fc0 in r_ble_ll_conn_enqueue_pkt ()
#5  0x42048758 in r_ble_ll_ctrl_proc_init ()
#6  0x42048b1e in r_ble_ll_ctrl_proc_start ()
#7  0x42046308 in r_ble_ll_conn_hci_rd_rem_ver_cmd ()
```
Registers:
```
zero           0x0	0
ra             0x4080bce2	0x4080bce2 <r_ble_lll_conn_txbuf_insert_after+272>
sp             0x4083c5a0	0x4083c5a0
gp             0x408177c4	0x408177c4 <__compound_literal.6+208>
tp             0x4083c780	0x4083c780
t0             0x40030dca	1073941962
t1             0xf	15
t2             0xe7a8d1	15182033
fp             0x408363b4	0x408363b4
s1             0x40849064	1082429540
a0             0x0	0
a1             0x408498b4	1082431668
a2             0x40802c4a	1082141770
a3             0x0	0
a4             0x0	0
a5             0x40849c54	1082432596
a6             0x40849c54	1082432596
a7             0x20000000	536870912
s2             0x40849b34	1082432308
s3             0x40824000	1082277888
s4             0x9ff49068	-1611362200
s5             0x4083813c	1082360124
s6             0x0	0
s7             0x1000000	16777216
s8             0x4084a714	1082435348
s9             0x408363b4	1082352564
s10            0x0	0
s11            0x0	0
t3             0x0	0
t4             0x10000100	268435712
t5             0xe7ab60	15182688
t6             0xe7a661	15181409
pc             0x4080bcf0	0x4080bcf0 <r_ble_lll_conn_txbuf_insert_after+286>
```

### Steps to reproduce.

I attempted to reproduce this with an example program on ESP32-C6-DevKitM-1, but I could not.

1. Connect to device via nRF Connect.
3. Wait for GATT table to finish loading.
4. Disconnect from device.
5. Wait for Disconnect event to be displayed in serial debug output.
6. Go to step 1. Repeat until crash/breakpoint hit.

### Debug Logs.

```plain

```

### More Information.

Using CMake via VisualGDB in VisualStudio.

zhaoweiliang2021, 2025-07-30:
@kenh-sq 
Hi,
1.Unzip the provided libble_app.zip file. Will get a libble_app.a file(based on idf https://github.com/espressif/esp-idf/commit/f10ac3eec22efb1b892f0c8a9f5e871940185736 ).
2.Replace the existing libble_app.a file in the specified directory(esp-idf/components/bt/controller/lib_esp32c6/esp32c6-bt-lib/esp32c6).
3. Rebuild your project and perform the test again.

The issue cannot be reproduced locally, but a fix is available in the library that may have caused this issue.
We will further strengthen testing.

If you find a way to reproduce it, please share it with us.

If the issue persists, please provide further details.
Thanks.

[libble_app.zip](https://github.com/user-attachments/files/21509331/libble_app.zip)

zhaoweiliang2021, 2025-08-07:
@kenh-sq 
OK, if you can reproduce it later, please provide the relevant register information and the corresponding elf file at the time of the crash. 
Thanks.

zhaoweiliang2021, 2025-08-07:
@kenh-sq 
Hi,If you still encounter this issue, please help enable the following config and provide us with the last serial port log.
**CONFIG_BT_LE_CONTROLLER_LOG_ENABLED=y** 
 ((Top) -> Component config -> Bluetooth -> Controller Options -> Controller debug features)
**CONFIG_BT_LE_CONTROLLER_LOG_WRAP_PANIC_HANDLER_ENABLE=y** 
( (Top) -> Component config -> Bluetooth -> Controller Options -> Controller debug features -> Controller log enable)

Thanks

zhaoweiliang2021, 2025-08-08:
@kenh-sq 
yes, 
1. it's important to be able to trace why r_ble_lll_sched_get_realtime_info is returning a NULL pointer in this case.

2. Did you do anything extra to reproduce it? If you have a better way to reproduce, please provide it to us.

3. May need to provide the controller log first to better locate this issue.
Enable the following config and provide us with the last serial port log.
**CONFIG_BT_LE_CONTROLLER_LOG_ENABLED=y**
((Top) -> Component config -> Bluetooth -> Controller Options -> Controller debug features)
**CONFIG_BT_LE_CONTROLLER_LOG_WRAP_PANIC_HANDLER_ENABLE=y**
( (Top) -> Component config -> Bluetooth -> Controller Options -> Controller debug features -> Controller log enable)

kenh-sq, 2025-08-08:
@zhaoweiliang2021 

Please see attached two instances of crash:
[20250808_r_ble_lll_conn_txbuf_insert_after_crash_log.txt](https://github.com/user-attachments/files/21685020/20250808_r_ble_lll_conn_txbuf_insert_after_crash_log.txt)

[20250808_r_ble_lll_conn_txbuf_insert_after_crash_log_2.txt](https://github.com/user-attachments/files/21685615/20250808_r_ble_lll_conn_txbuf_insert_after_crash_log_2.txt)

[sdkconfig](https://github.com/user-attachments/files/21685777/sdkconfig.txt)

We have many things happening at once within the stack:
* Continuous passive scanning for CODED PHY devices.
   * When scanning is enabled, we request any connection (in-progress or current) to adjust the connection interval to 60ms so that we have a better chance of getting scan results.
* Updating scan response every 125ms.
   * Associated scannable, connectable, legacy advertisement is also getting set every 125ms.
* Sending large notifications to connected devices (I see some of them in the log output) every 125ms or more often.
* Possibly getting periodic sync messages from a CODED PHY device, but this is not required to get this issue to occur.

All application tasks are running at (configMAX_PRIORITIES - 6) or (configMAX_PRIORITIES - 5) to avoid conflicting with host task at (configMAX_PRIORITIES - 4).

BLE_ERR_DIFF_TRANS_COLL occur sometimes during connections, but I am not seeing those occurring specifically before this crash. Rarely, I will see BLE_ERR_LMP_COLLISION occur, but its been a while since I've seen one.

Thank you.

kenh-sq, 2025-08-11:
@zhaoweiliang2021

I have not been able to reproduce the issue using the libble_app.a provided in https://github.com/espressif/esp-idf/issues/16984#issuecomment-3174041874. I am not getting any assert during operation, so I have no new log data to share.

With respect to reproducing the issue, could you share an understanding of the sequence of controller events that lead to the problem? That might help me to define a reproduction.

Thank you.