espressif/esp-idf #16985: LEDC doesn't work in LightSleep since ESP-IDF 5.4

mtal, 2025-07-23:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.3 - v5.5

### Espressif SoC revision.

ESP32 (revision 3.1)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-WROOM32-D

### Power Supply used.

USB

### What is the expected behavior?

PWM generation during Light Sleep

### What is the actual behavior?

PWM generation during automatic Light Sleep is broken since v5.4. Led connected to LEDC output pin blinks randomly at visible intervals. 

If light sleep disabled there is no issue.

Before v5.4 it works perfectly with light sleep.

I updated code and configured ledc_channel_config_t::sleep_mode to LEDC_SLEEP_MODE_KEEP_ALIVE according to documentation. It doesn't work

### Steps to reproduce.

    bootloader_random_enable();
    esp_event_loop_create_default();
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);
    ESP_ERROR_CHECK(esp_netif_init());

    esp_netif_t *wifi_netif = esp_netif_create_default_wifi_sta();

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();

    esp_wifi_init(&cfg);
    esp_wifi_set_max_tx_power(8);

    char *ssid = "test";
    char *password = "test";

    static const esp_timer_create_args_t reconnect_timer_args = {
            .callback = &reconnect_timer_callback,
            /* argument specified here will be passed to timer callback function */
            .name = "wifi-reconnect",
            .dispatch_method = ESP_TIMER_TASK
    };

    esp_timer_create(&reconnect_timer_args, &reconnect_timer_handle);

    ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
                                                        ESP_EVENT_ANY_ID,
                                                        &event_handler,
                                                        wifi_netif,
                                                        NULL));

    wifi_config_t wifi_config = {
            .sta = {
                    .threshold.authmode = WIFI_AUTH_WPA2_PSK,
            },
    };

    memcpy(wifi_config.sta.ssid, ssid, strlen(ssid) + 1);
    memcpy(wifi_config.sta.password, password, strlen(password) + 1);

    esp_wifi_set_mode(WIFI_MODE_STA);
    esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config);

    esp_wifi_start();
    esp_wifi_set_ps(WIFI_PS_MAX_MODEM);

    esp_pm_config_t pm_config = {
            .max_freq_mhz = CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ,
            .min_freq_mhz = rtc_clk_xtal_freq_get(),
#if CONFIG_FREERTOS_USE_TICKLESS_IDLE
            .light_sleep_enable = true
#endif
    };
    ESP_ERROR_CHECK(esp_pm_configure(&pm_config)); 
   const ledc_timer_config_t ledc_timer = {
            .speed_mode       = LEDC_LOW_SPEED_MODE,
            .timer_num        = LEDC_TIMER_0,
            .duty_resolution  = LEDC_TIMER_10_BIT,
            .freq_hz          = 2000, 
            .clk_cfg          = LEDC_USE_RC_FAST_CLK
    };
    ESP_ERROR_CHECK(ledc_timer_config(&ledc_timer));
    const  ledc_channel_config_t ledc_channel = {
            .speed_mode     = LEDC_LOW_SPEED_MODE,
            .channel        = LEDC_CHANNEL_0,
            .timer_sel      = LEDC_TIMER_0,
            .intr_type      = LEDC_INTR_DISABLE,
            .gpio_num       = 18,
            .duty           = 0, // Set duty to 0%
            .hpoint         = 0
    };
    ESP_ERROR_CHECK(ledc_channel_config(&ledc_channel));
    ESP_ERROR_CHECK(ledc_set_duty(LEDC_LOW_SPEED_MODE, LEDC_CHANNEL_0, 255));
    // Update duty to apply the new value
    ESP_ERROR_CHECK(ledc_update_duty(LEDC_LOW_SPEED_MODE, LEDC_CHANNEL_0));

### Debug Logs.

```plain

```

mtal, 2025-07-23:
Edited: replaced real project code to example code easy to reproduce and attached full project with sdkconfig

ginkgm, 2025-07-23:
Hi @mtal ,

Thanks for your report. We appreciate your simplified example. It will help us to locate the issue earlier.

mtal, 2025-07-23:
The issue occurs only with auto sleep via `esp_pm_configure(&pm_config)` with `.light_sleep_enable = true`

Without `esp_pm_configure` there is no issue. If the line `ESP_ERROR_CHECK(esp_pm_configure(&pm_config));` is commented and the loop replaced to manual sleep there is no issue with LEDC

This works properly:
```c
void app_main(void) {
    const ledc_timer_config_t ledc_timer = {
            .speed_mode       = LEDC_LOW_SPEED_MODE,
            .timer_num        = LEDC_TIMER_0,
            .duty_resolution  = LEDC_TIMER_10_BIT,
            .freq_hz          = 2000,
            .clk_cfg          = LEDC_USE_RC_FAST_CLK
    };
    ESP_ERROR_CHECK(ledc_timer_config(&ledc_timer));
    const ledc_channel_config_t ledc_channel = {
            .speed_mode     = LEDC_LOW_SPEED_MODE,
            .channel        = LEDC_CHANNEL_0,
            .timer_sel      = LEDC_TIMER_0,
            .intr_type      = LEDC_INTR_DISABLE,
            .gpio_num       = GPIO_NUM_2,
            .duty           = 0, // Set duty to 0%
            .hpoint         = 0,
            .sleep_mode     = LEDC_SLEEP_MODE_KEEP_ALIVE
    };
    ESP_ERROR_CHECK(ledc_channel_config(&ledc_channel));
    ESP_ERROR_CHECK(esp_pm_configure(&pm_config));
    ESP_ERROR_CHECK(ledc_set_duty(LEDC_LOW_SPEED_MODE, LEDC_CHANNEL_0, 50));
    while (true) {
        ESP_ERROR_CHECK(esp_sleep_enable_timer_wakeup(1000000));
        esp_light_sleep_start();
        vTaskDelay(pdMS_TO_TICKS(2000));
    }

}
```

Also tried to investigate and replace `esp_driver_ledc` component with previous version from v5.3.3. Inside `ledc.c` renamed old name `gpio_hal_iomux_func_sel` to new name `gpio_func_sel`. The issue still persists.

So the issue is not related to `esp_driver_ledc` component.

songruo, 2025-07-23:
Hello @mtal, I ran your example on release/v5.5, I am not able to reproduce. Please provide a specific commit ID.

songruo, 2025-07-23:
Yes, I ran from your folder, changed commit to tag v5.5, still could not reproduce. Setting `.light_sleep_enable = true` or `false`, the LED behavior looks the same to me. Also I probed with logic analyzer, the PWM signal exists during auto light sleep period.

songruo, 2025-07-24:
It looks like you are not using our Espressif devkit board, but a third party development board. I feel the problem is with the board specifically. Please try to check the dev board schematics.

mtal, 2025-07-24:
To record a video a got a nearby board with soldered led. It is even not a ESP32-WROOM32-D. But it is a ESP32-D0WD-V3 chip.

```
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
```

Espressif devkit board doesn't have a led on it. It is hard to record such video with original board.

There is no particular with this board. I have esp32 devtik with wroom-32d too. The behavior of original devkit is the same.

> I feel the problem is with the board specifically.
How it can be? The board works perfectly with v5.3.3. The board works perfectly even with v5.5 without call esp_pm_configure with manual call esp_light_sleep_start.

songruo, 2025-07-24:
Is the LED an addressable programmable LED or normal RGB LED?

mtal, 2025-07-24:
Simple blue led.

[image]

songruo, 2025-07-24:
Oh, so in your video, the red LED is the power LED, which is always on? And the other blue LED is the one connects to IO2, and is blinking?

mtal, 2025-07-24:
>  And the other blue LED is the one connects to IO2, and is blinking?
Kind of blinking. It is not blinking at fixed frequency. It is sometimes on, sometimes off. And only when auto light sleep is enabled.

Without auto light sleep it dimmed, actually fast blinking with LEDC PWM frequency.

On v. 5.3.3 it is dimmed too, even with auto sleep.

songruo, 2025-07-24:
Another question I want to ask is: `ledc_channel_config_t::sleep_mode = LEDC_SLEEP_MODE_KEEP_ALIVE` is introduced in v5.4. For earlier IDF versions, the clock and IO needs to be handled by yourself, can you show me the code that you use to configure clock and IO on v5.3?

mtal, 2025-07-24:
> I would like to see whether the RC_FAST clock is considered to be on during sleep period.

Updated code

```c
    ...
    ESP_ERROR_CHECK(ledc_update_duty(LEDC_LOW_SPEED_MODE, LEDC_CHANNEL_0));
    static esp_sleep_context_t s_sleep_ctx;
    esp_sleep_set_sleep_context(&s_sleep_ctx);
    esp_log_level_set(TAG, ESP_LOG_DEBUG);

    while (true) {
        vTaskDelay(pdMS_TO_TICKS(2000));
        ESP_LOGI(TAG, "sleep_flags %lu", s_sleep_ctx.sleep_flags);
    }
    ...
```

v5.5:
```
I (320) pm: Frequency switching config: CPU_MAX: 80, APB_MAX: 80, APB_MIN: 40, Light sleep: ENABLED
I (2321) MAIN: sleep_flags 262528
I (4321) MAIN: sleep_flags 262528
I (6323) MAIN: sleep_flags 262528
I (8324) MAIN: sleep_flags 262528
I (10326) MAIN: sleep_flags 262528
```

v5.3.3
```
I (338) pm: Frequency switching config: CPU_MAX: 80, APB_MAX: 80, APB_MIN: 40, Light sleep: ENABLED
I (2340) MAIN: sleep_flags 327808
I (4342) MAIN: sleep_flags 327808
I (6345) MAIN: sleep_flags 327808
```

And here code for v5.5 without auto sleep and call ```esp_pm_configure``` which works properly
```c
...
    static esp_sleep_context_t s_sleep_ctx;
    esp_sleep_set_sleep_context(&s_sleep_ctx);
    esp_log_level_set(TAG, ESP_LOG_DEBUG);
    ESP_ERROR_CHECK(esp_sleep_enable_timer_wakeup(1000000));
    ESP_LOGI(TAG, "timer wakeup source is ready");
    while (true) {
        esp_light_sleep_start();
        ESP_LOGI(TAG, "sleep_flags %lu", s_sleep_ctx.sleep_flags);
        vTaskDelay(pdMS_TO_TICKS(2000));
    }
```
Flags:
```
I (320) MAIN: timer wakeup source is ready
I (321) MAIN: sleep_flags 327808
I (2321) MAIN: sleep_flags 327808
I (4321) MAIN: sleep_flags 327808
I (6321) MAIN: sleep_flags 327808
I (8321) MAIN: sleep_flags 327808
```