espressif/esp-idf #16992: Setting the state of the CS pin when using SPI

kdschlosser, 2025-07-23:
### Is your feature request related to a problem?

I am not sure if maybe I missed it somewhere but I have not been able to locate a place to set what the state of the CS line should be when it's considered to be active. The ability exists for the SPI LCD bus driver but not for normal SPI operation. Is there a way to instruct the SPI driver what the active state is for the CS line?

### Describe the solution you'd like.

I the ability doesn't exist perhaps it should be added??

kdschlosser, 2025-07-28:
kinda adds quite a bit of additional overhead for something that should really be done internally don't ya think? It is handled internally for the esp-lcd SPI IO... It would literally be a non add to memory use because the field could be added using a single bit and I believe there should be some free bits in one of the bytes that the structure is using no? Not a whole heck of a lot in terms of processor use when pulling the field from the structure to call the function to set the GPIO state. 

On the other hand having to make function calls that happen in an ISR to handle it is quite a bit more in terms of overhead.

IDK, I think it is something that should be added only for the reason of having the API consistent between components that offer an extension to the SPI component like the esp-lcd component. There might be others but I would have to check. IDK if setting the level for an active state CS line is available in the SDCard reader component or not. I have not dove into it to check and see where it is offered. It's not a bad thing to offer because if it is done in the spi component it is going to be a small binary, less memory use, and less processor time used then if a user has to register callback functions to handle it themselves.

wanckl, 2025-10-20:
@kdschlosser    
The CS line is active_low by default in normal SPI driver, If you want active_high, you can using flag `SPI_DEVICE_POSITIVE_CS` when adding the device