espressif/esp-idf #16998: HTTP server not processing socket after async request

tim-egnite, 2025-07-24:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v6.0-dev-1362-g346870a304 or v5.4.1

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.0)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Olimex ESP32-Gateway

### Power Supply used.

USB

### What is the expected behavior?

Async HTTP requests on the same socket should run after another.

### What is the actual behavior?

The first async request runs directly, but the second request on the same socket can wait a long time.

### Steps to reproduce.

Compile and run the example [async_handlers](https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/async_handlers). I modified the settings, so Ethernet is used. Testing is easier when the loop iterations in function `long_async` are reduced.

Open URL http://ip/long in the browser (tested with Firefox), where ip is replaced with the IP or hostname. The request should finish after some time, depending on modifications to function `long_async`.

Reload the page in the browser. The browser should make the request on the same socket. Don't make any other requests to the webserver. This request will then load forever.

If another request is made to the webserver with another browser or tab, then the long request will also be processed.

### Debug Logs.

```plain

```

### More Information.

Here is what I think happens internally:

The server thread in function [`httpd_server`](https://github.com/espressif/esp-idf/blob/346870a3044010f2018be0ef3b86ba650251c655/components/esp_http_server/src/httpd_main.c#L276) waits for new data using `select`. After the first async request has been moved to another thread, function `httpd_server` will again wait using `select`. This time the socket is not in the set for `select`, because it has `for_async_req` set. See [`case HTTPD_TASK_SET_DESCRIPTOR`](https://github.com/espressif/esp-idf/blob/346870a3044010f2018be0ef3b86ba650251c655/components/esp_http_server/src/httpd_sess.c#L90) in function `enum_function`.

After `long_async` sent the response, the browser keeps the socket for another request. Reloading the page then sents a new request to the server, but the server does not react, because the socket is still not in the set for `select`.

Possible solutions:
* Remove the check for `for_async_req` under `case HTTPD_TASK_SET_DESCRIPTOR` in function `enum_function`. The socket would then always be in the set for `select`. This seems to work, but could cause other problems, if data is received while the async handler is still running.
* Inform the thread of `httpd_server` that it should update the set for `select`. I'm now doing that as a workaround: After calling `httpd_req_async_handler_complete` I also call `httpd_queue_work` with an empty callback function. The `select` call will return and the next call will have the socket in the set again.

zinke-ct-video, 2025-07-24:
I can also confirm the bug in ESP-IDF v5.5.1, commit b66b54.
Our webserver becomes unresponsive after handling an async request.

BitsForPeople, 2025-07-24:
I think a safe and easy solution would be to just close the socket whenever it is returned to the httpd via `httpd_req_async_handler_complete()`.

zinke-ct-video, 2025-07-25:
Ah okay. You didn't mean this as a solution for the API-user of the HTTPD component but as a solution for espressif in order to fix it? Sorry, now I got your point.

BitsForPeople, 2025-07-25:
Might also be a work-around for the user ([`httpd_sess_trigger_close()`](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/protocols/esp_http_server.html#_CPPv424httpd_sess_trigger_close14httpd_handle_ti)).

hrushikesh430, 2025-08-14:
Hello @tim-egnite,
I was able to reproduce the issue you described: the server indeed fails to process new requests on the same socket after an asynchronous request. We’ve identified the root cause and are actively working on a fix. I will notify you as soon as the resolution is available.
Thank you for reporting this and for your patience.

DavyLandman, 2025-09-06:
> Which is what I'd expect when both are requesting the same URL at the same time...

Are you talking about REST semantics?

In this case it was 2 URIs that were dealt with by the same wild-card pattern. So they were 2 different URLs, the browser based fetch request never reached the http server, while the curl on woke the httpd handler, and then suddenly the browser one got the same result, even thought the httpd handler was only invoked once. So 2 connections that got the exact same content written to them.

hrushikesh430, 2025-09-08:
Hello @DavyLandman, Thanks for reporting this. We are working on this issue internally. We will fix this bug as soon as possible and let you know.

hrushikesh430, 2025-09-16:
@tim-egnite Oh yes, We will consider sending struct instead of string. Thanks.