espressif/esp-idf #17000: Reading from LittleFS (flash) and writing to CDC USB (ESP32S3) causes race condition, crash

aircable, 2025-07-24:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

6.0 dev

### Espressif SoC revision.

ESP32S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Seeed ESP32S3

### Power Supply used.

USB

### What is the expected behavior?

Expect that read flash interrupts, which disabled the cache, are shielded from USB CDC transmit callbacks, which needs the cache since it is in ROM.

### What is the actual behavior?

I found an interesting bug. It occurred while processing readdir() on a flash based LittleFS and printing the information about files on the console via USB CDC.
A standard "ls" implementation to list files, see below
The flash read disables the cache and gets interrupted by the USB tx callback and then crashes. 

This is the function that causes the problem, but the printf is handled by the console OTG CDC driver:

```
// Function to recursively list files in a directory
static void list_files(const char* path, int print_permissions, int recursive ) {
    DIR* dir = opendir(path);
    if (!dir) {
        perror("Cannot open directory");
        return;
    }

    struct dirent* entry;
    while ((entry = readdir(dir)) != NULL) {
        if (strcmp(entry->d_name, ".") == 0 || strcmp(entry->d_name, "..") == 0)
            continue;

        char filepath[PATH_MAX];
        const char *name = entry->d_name;
        // Some VFS implementations might return entry names with a leading slash.
        // We strip it here to ensure consistent path construction.
        if (name[0] == '/') {
            name++;
        }

        // When listing the root directory, `path` is "/". A simple concatenation
        // would create an invalid "//" path. We handle this special case.
        if (strcmp(path, "/") == 0) {
            snprintf(filepath, sizeof(filepath), "/%s", name);
        } else {
            snprintf(filepath, sizeof(filepath), "%s/%s", path, name);
        }

        struct stat file_stat;
        if (stat(filepath, &file_stat) == -1) {
            ESP_LOGI( TAG, "path %s, entry %s", path, entry->d_name );
            perror( "Cannot get file stat" );
            continue;
        }
        //ESP_LOGI( TAG, "permissions %d", print_permissions );
        if (print_permissions) {
            struct stat st;
            stat( entry->d_name, &st );
            permissions( st.st_mode ) ;
            printf("\t%s\t\t", entry->d_name);
            printf("%ld", file_stat.st_size);
            printf("\n");
        } else {
            if(( S_ISDIR(file_stat.st_mode)) && recursive )
                list_files( filepath, print_permissions, recursive );
            else
                printf("%s\n", entry->d_name);
        }
    }

    closedir(dir);
}

```

Guru Meditation Error: Core  0 panic'ed (Cache error). 
Cache disabled but cached memory region accessed

Core  0 register dump:
PC      : 0x4200816e  PS      : 0x00060034  A0      : 0x80050fa8  A1      : 0x3fca3e90  
--- 0x4200816e: cdcacm_tx_cb at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/esp-idf/components/esp_vfs_console/vfs_cdcacm.c:243

A2      : 0x3fca69bc  A3      : 0xfffffffd  A4      : 0x3fcd9df0  A5      : 0x3fcd9db0  
A6      : 0x00000080  A7      : 0x3fceb87c  A8      : 0x80385d94  A9      : 0x3fca69b0  
A10     : 0x3fcd9dbc  A11     : 0x3fcd9dbc  A12     : 0x3fcd9db0  A13     : 0x3fcd9d30  
A14     : 0x3fca47ac  A15     : 0x3fceb8bc  SAR     : 0x00000008  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  
--- 0x400570e8: memset in ROM
--- 0x400570f3: memset in ROM

Backtrace: 0x4200816b:0x3fca3e90 0x40050fa5:0x3fca3ec0 0x4005213a:0x3fca3ee0 0x40052cdd:0x3fca3f00 0x40385e1e:0x3fca3f20 0x403801c2:0x3fca3f50 0x40388fd7:0x3fcd9cd0 0x403889e9:0x3fcd9cf0 0x403912f1:0x3fcd9d30 0x4038f6ef:0x3fcd9db0 0x42048f01:0x3fcd9df0 0x420430b7:0x3fcd9e20 0x4204335a:0x3fcd9e50 0x42043f71:0x3fcd9e90 0x420440ed:0x3fcd9ef0 0x42047f9b:0x3fcd9f40 0x42041c6e:0x3fcd9f80 0x420424a9:0x3fcda0c0 0x4200939d:0x3fcda0f0 0x420165a1:0x3fcda120 0x42016653:0x3fcda310 0x420170d3:0x3fcda350 0x4201624d:0x3fcda370
--- 0x4200816b: cdcacm_rx_cb at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/esp-idf/components/esp_vfs_console/vfs_cdcacm.c:239
--- 0x40050fa5: cdc_acm_bulk_in in ROM
--- 0x4005213a: usb_dw_isr_handler in ROM
--- 0x40052cdd: usb_dc_check_poll_for_interrupts in ROM
--- 0x40385e1e: esp_usb_console_interrupt at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/esp-idf/components/esp_system/port/usb_console.c:188
--- 0x403801c2: _xt_lowint1 at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/esp-idf/components/xtensa/xtensa_vectors.S:1240
--- 0x40388fd7: spimem_flash_ll_cmd_is_done at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/esp-idf/components/hal/esp32s3/include/hal/spimem_flash_ll.h:63
---  (inlined by) spi_flash_hal_poll_cmd_done at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/esp-idf/components/hal/spi_flash_hal_common.inc:40
--- 0x403889e9: spi_flash_hal_read at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/esp-idf/components/hal/spi_flash_hal_common.inc:227
--- 0x403912f1: spi_flash_chip_winbond_read at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/esp-idf/components/spi_flash/spi_flash_chip_winbond.c:66
--- 0x4038f6ef: esp_flash_read at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/esp-idf/components/spi_flash/esp_flash_api.c:932
--- 0x42048f01: esp_partition_read at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/esp-idf/components/esp_partition/partition_target.c:38
--- 0x420430b7: littlefs_esp_part_read at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/WishMesh/managed_components/joltwallet__littlefs/src/littlefs_esp_part.c:34
--- 0x4204335a: lfs_bd_read at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/WishMesh/managed_components/joltwallet__littlefs/src/littlefs/lfs.c:116
--- 0x42043f71: lfs_dir_getslice at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/WishMesh/managed_components/joltwallet__littlefs/src/littlefs/lfs.c:739
--- 0x420440ed: lfs_dir_get at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/WishMesh/managed_components/joltwallet__littlefs/src/littlefs/lfs.c:785
---  (inlined by) lfs_dir_getinfo at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/WishMesh/managed_components/joltwallet__littlefs/src/littlefs/lfs.c:1418
--- 0x42047f9b: lfs_dir_read_ at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/WishMesh/managed_components/joltwallet__littlefs/src/littlefs/lfs.c:2798
---  (inlined by) lfs_dir_read at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/WishMesh/managed_components/joltwallet__littlefs/src/littlefs/lfs.c:6374
--- 0x42041c6e: vfs_littlefs_readdir_r at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/WishMesh/managed_components/joltwallet__littlefs/src/esp_littlefs.c:2150
--- 0x420424a9: vfs_littlefs_readdir at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/WishMesh/managed_components/joltwallet__littlefs/src/esp_littlefs.c:2135
--- 0x4200939d: esp_vfs_readdir at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/esp-idf/components/vfs/vfs.c:1250
--- 0x420165a1: list_files at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/WishMesh/components/settings/buildin_cmds.c:113
--- 0x42016653: ls_cmd at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/WishMesh/components/settings/buildin_cmds.c:179
--- 0x420170d3: esp_console_run at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/esp-idf/components/console/commands.c:257
--- 0x4201624d: console_task at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/WishMesh/components/settings/cmd_system.c:92

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

igrr, 2025-07-30:
Could you please attach your sdkconfig file?

igrr, 2025-07-30:
Thank you. Could you please try disabling CONFIG_ESP_CONSOLE_USB_CDC_SUPPORT_ETS_PRINTF option and see if that stops the cache error from happening?

aircable, 2025-07-30:
you mean "enable", so that we won't run into cache disable problems during
printf

aircable, 2025-07-31:
> No, I did mean to disable it. According to your sdkconfig, this option is
> currently enabled.
>
> However, when this option is enabled, two things happen:
>
>    - USB interrupt handler is configured to run even when the cache is
>    disabled
>    - Functions called from the interrupt handler — such as cdcacm_rx_cb —
>    are mistakenly *not* placed in IRAM
>
> We will fix the reason why cdcacm_rx_cb is not placed in IRAM, but I just
> wanted to confirm if the issue you observe is really related to this and
> not to something else, and disabling
> CONFIG_ESP_CONSOLE_USB_CDC_SUPPORT_ETS_PRINTF seemed like an effective way
> to test this.

That's weird, the one I checked on my system with menuconfig showed it
disabled.
But I doubt it's resolving the problem because the printf that causes the
error is in my code.
I could use the esp_rom_printf function instead and I'm sure it will
resolve it.
The way I implemented the solution was by printing into a buffer during the
diropen and then afterwards printing the buffer.
I'll test again today

igrr, 2025-07-31:
Thanks for the confirmation, this issue should get closed once the fix for CONFIG_ESP_CONSOLE_USB_CDC_SUPPORT_ETS_PRINTF is merged.

tore-espressif, 2025-08-25:
Fixed in https://github.com/espressif/esp-idf/commit/5ee52801956f2df2cab28e4fd41a6f47389237c0

closed 2025-08-25 by tore-espressif