espressif/esp-idf #17001: ESP32C2 log output baudrate is not setting into 115200 when 40MHz XTAL selected when running

Asanga-Viraj, 2025-07-25:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am working with ESP32C2 on idf 5.4.2. When I change XTAL frequency into 40MHz, the default UART0 log output baudrate is updated into 115200. I saw that on sdkconfig file. When the code is running, it is still printing debug outputs over 74880 baudrate.

L-KAYA, 2025-07-28:
Hi @Asanga-Viraj , thanks for the report~

For ESP32-C2, please double confirm the frequency of the crystal oscillator. The ROM will always configure the default clock division factor for the UART, therefore, no matter the XTAL is selected to `40M` or `26M`, as long as the actual frequency of crystal oscillator is `26M`, the actual UART baudrate will still be `74880` at the default division.

The `40M XTAL` option can only be selected when you're sure your XTAL is 40MHz.

Asanga-Viraj, 2025-07-28:
Thank @L-KAYA .

I attached the sdkconfig file for 40MHz and 26MHz. They have different baudrates. The baudrate 115200 in 40MHz is not applying when running. 

Yes, you are correct. When application running, the baudrate is 74880.

But , I like behave ESP32C2 as other ESPs which has 115200 baudrate for debug log output. Is there any way to set default baudrate into 115200?

[sdkconfig_40MHz.txt](https://github.com/user-attachments/files/21460513/sdkconfig_40MHz.txt)

[sdkconfig_26MHz.txt](https://github.com/user-attachments/files/21460514/sdkconfig_26MHz.txt)

Asanga-Viraj, 2025-07-28:
I tried the hello_world example. The `XTAL_FREQ` is `26MHz.` as default. No matter what you set the value for `ESP_CONSOLE_UART_BAUDRATE` in sdkconfig, it goes back to 74880. 

When XTAL_FREQ is selected as 40MHz, the default value of `ESP_CONSOLE_UART_BAUDRATE` goes to 115200. 

When code is running, the printing baudrate is 74880, no matter what the value in ESP_CONSOLE_UART_BAUDRATE which is not effecting in the code.

I think this is a bug.

Asanga-Viraj, 2025-08-08:
Thanks @songruo . But still it doesn't work. ESP outputs over 74880 bausrate. The sdkconfig changes are here.

`CONFIG_ESP_SYSTEM_EVENT_QUEUE_SIZE=32
CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=2304
CONFIG_ESP_MAIN_TASK_STACK_SIZE=3584
CONFIG_ESP_MAIN_TASK_AFFINITY_CPU0=y
# CONFIG_ESP_MAIN_TASK_AFFINITY_NO_AFFINITY is not set
CONFIG_ESP_MAIN_TASK_AFFINITY=0x0
CONFIG_ESP_MINIMAL_SHARED_STACK_SIZE=2048
# CONFIG_ESP_CONSOLE_UART_DEFAULT is not set
CONFIG_ESP_CONSOLE_UART_CUSTOM=y
# CONFIG_ESP_CONSOLE_NONE is not set
CONFIG_ESP_CONSOLE_UART=y
CONFIG_ESP_CONSOLE_UART_CUSTOM_NUM_0=y
# CONFIG_ESP_CONSOLE_UART_CUSTOM_NUM_1 is not set
CONFIG_ESP_CONSOLE_UART_NUM=0
CONFIG_ESP_CONSOLE_ROM_SERIAL_PORT_NUM=0
CONFIG_ESP_CONSOLE_UART_TX_GPIO=20
CONFIG_ESP_CONSOLE_UART_RX_GPIO=19
CONFIG_ESP_CONSOLE_UART_BAUDRATE=115200
CONFIG_ESP_INT_WDT=y
CONFIG_ESP_INT_WDT_TIMEOUT_MS=300
CONFIG_ESP_TASK_WDT_EN=y
CONFIG_ESP_TASK_WDT_USE_ESP_TIMER=y
CONFIG_ESP_TASK_WDT_INIT=y
# CONFIG_ESP_TASK_WDT_PANIC is not set
CONFIG_ESP_TASK_WDT_TIMEOUT_S=5
CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0=y
# CONFIG_ESP_PANIC_HANDLER_IRAM is not set
# CONFIG_ESP_DEBUG_STUBS_ENABLE is not set
CONFIG_ESP_DEBUG_OCDAWARE=y
CONFIG_ESP_SYSTEM_CHECK_INT_LEVEL_4=y
`

songruo, 2025-08-08:
@Asanga-Viraj Are you sure the baudrate is always 74880? The baudrate will be 74880 for a short period of time after boot, very soon the baudrate should be adjusted to 115200 afterwards.

songruo, 2025-08-08:
@Asanga-Viraj Yes, I have tried, my chip prints at 115200. Are you using IDF official monitor? Can you show me a screenshot of the "logs" since boot?

songruo, 2025-08-08:
What is CONFIG_XTAL_FREQ value in your sdkconfig? Can you switch to the other, and try again?

songruo, 2025-08-11:
> Still console custom baudrate is not applying with 40MHz XTAL which makes a bug in system.

This is not a bug. On C2, the chip cannot automatically detected the XTAL frequency, it has to be manually selected in menuconfig to match the actual XTAL frequency. Otherwise, the baudrate cannot be correct.

closed 2025-08-11 by songruo