espressif/esp-idf #17004: ESP-IDF v5.5版本，烧录xiaozhi-esp32，运行异常；v5.4.1版本，运行正常

yuwei-virtual, 2025-07-25:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.5

### Espressif SoC revision.

ESP32-D0WD

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROVER-E

### Power Supply used.

USB

### What is the expected behavior?

默认配置下，程序正常运行。

### What is the actual behavior?

打开PSRAM，编译烧录完成xiaozhi-esp32之后，程序不断复位。使用a2dp_source例程，同样打开PSRAM，烧录运行正常。

### Steps to reproduce.

1、获取xiaozhi-esp32，PSRAM关闭的情况下，运行正常；
2、不做任何修改，打开PSRAM，默认配置下，编译烧录运行异常。

### Debug Logs.

```plain
ELF file SHA256: 0e1559547

Rebooting...
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6380
ho 0 tail 12 room 4
load:0x40078000,len:15996
load:0x40080400,len:3860
--- 0x40080400: _invalid_pc_placeholder at /home/yuwei/Workspace/esp/esp-idf/components/xtensa/xtensa_vectors.S:2235
entry 0x40080638
--- Error: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
--- Waiting for the device to reconnect--- 0x40080638: call_start_cpu0 at /home/yuwei/Workspace/esp/esp-idf/components/bootloader/subproject/main/bootloader_start.c:25
I (28) boot: ESP-IDF v5.5 2nd stage bootloader
I (28) boot: compile time Jul 25 2025 11:28:29
I (28) boot: Multicore bootloader
I (30) boot: chip revision: v3.1
I (33) boot.esp32: SPI Speed      : 40MHz
I (36) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 4MB
I (44) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (64) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (70) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  3 model            Unknown data     01 82 00010000 000f0000
I (83) boot:  4 factory          factory app      00 00 00100000 00300000
I (90) boot: End of partition table
I (93) boot: Defaulting to factory image
I (97) esp_image: segment 0: paddr=00100020 vaddr=3f400020 size=c0960h (788832) map
I (374) esp_image: segment 1: paddr=001c0988 vaddr=3ff80000 size=00020h (    32) load
I (374) esp_image: segment 2: paddr=001c09b0 vaddr=3ffb0000 size=04008h ( 16392) load
I (383) esp_image: segment 3: paddr=001c49c0 vaddr=40080000 size=0b658h ( 46680) load
I (403) esp_image: segment 4: paddr=001d0020 vaddr=400d0020 size=13c4c8h (1295560) map
I (846) esp_image: segment 5: paddr=0030c4f0 vaddr=4008b658 size=0745ch ( 29788) load
I (867) boot: Loaded app from partition at offset 0x100000
I (867) boot: Disabling RNG early entropy source...
I (878) quad_psram: This chip is ESP32-D0WD
I (878) esp_psram: Found 8MB PSRAM device
I (878) esp_psram: Speed: 40MHz
I (878) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (885) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (893) cpu_start: Multicore app
I (1777) esp_psram: SPI SRAM memory test OK
I (1785) cpu_start: Pro cpu start user code
I (1785) cpu_start: cpu freq: 160000000 Hz
I (1785) app_init: Application information:
I (1786) app_init: Project name:     xiaozhi
I (1790) app_init: App version:      1.8.3
I (1794) app_init: Compile time:     Jul 25 2025 11:28:23
I (1799) app_init: ELF file SHA256:  0e1559547...
I (1803) app_init: ESP-IDF:          v5.5
I (1807) efuse_init: Min chip rev:     v0.0
I (1811) efuse_init: Max chip rev:     v3.99 
I (1815) efuse_init: Chip rev:         v3.1
I (1819) heap_init: Initializing. RAM available for dynamic allocation:
I (1825) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1830) heap_init: At 3FFBA900 len 00025700 (149 KiB): DRAM
I (1836) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1841) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1847) heap_init: At 40092AB4 len 0000D54C (53 KiB): IRAM
I (1852) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
Guru Meditation Error: Core  0 panic'ed (Cache error). 
Cache disabled but cached memory region accessed

Core  0 register dump:
PC      : 0x401fcc90  PS      : 0x00060934  A0      : 0x80084de0  A1      : 0x3ffe3ab0  
.--- 0x401fcc90: memset at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:113
A2      : 0x3ffb2924  A3      : 0x3ffe3b44  A4      : 0x00000000  A5      : 0x00060723  
A6      : 0x00000001  A7      : 0x3ffb2904  A8      : 0x80091550  A9      : 0x3ffe3ab0  
A10     : 0x3ffe3ab0  A11     : 0x00000000  A12     : 0x0000001c  A13     : 0x3f41dba4  
A14     : 0x00ff0000  A15     : 0x0000cdcd  SAR     : 0x00000020  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x401fccb6  LEND    : 0x401fccc1  LCOUNT  : 0x00000000  
.--- 0x401fccb6: memset at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:150
--- 0x401fccc1: memset at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:160

Backtrace: 0x401fcc8d:0x3ffe3ab0 0x40084ddd:0x3ffe3af0 0x4020b20d:0x3ffe3b20 0x40090df2:0x3ffe3b40 0x401485a5:0x3ffe3b70 0x400d2d70:0x3ffe3bd0 0x400d3507:0x3ffe3bf0 0x400d353a:0x3ffe3c10 0x4008170a:0x3ffe3c40 0x40079796:0x3ffe3c90 |<-CORRUPTED
.--- 0x401fcc8d: __memset_aux at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:106
--- 0x40084ddd: read_id_core at /home/yuwei/Workspace/esp/esp-idf/components/spi_flash/esp_flash_api.c:427
--- 0x4020b20d: esp_flash_read_chip_id at /home/yuwei/Workspace/esp/esp-idf/components/spi_flash/esp_flash_api.c:445
--- 0x40090df2: esp_flash_init_main at /home/yuwei/Workspace/esp/esp-idf/components/spi_flash/esp_flash_api.c:354
--- 0x401485a5: esp_flash_init_default_chip at /home/yuwei/Workspace/esp/esp-idf/components/spi_flash/esp_flash_spi_init.c:409
--- 0x400d2d70: __esp_system_init_fn_init_flash at /home/yuwei/Workspace/esp/esp-idf/components/esp_system/startup_funcs.c:117
--- 0x400d3507: do_system_init_fn at /home/yuwei/Workspace/esp/esp-idf/components/esp_system/startup.c:132
--- 0x400d353a: do_core_init at /home/yuwei/Workspace/esp/esp-idf/components/esp_system/startup.c:170
--- (inlined by) start_cpu0_default at /home/yuwei/Workspace/esp/esp-idf/components/esp_system/startup.c:203
--- 0x4008170a: call_start_cpu0 at /home/yuwei/Workspace/esp/esp-idf/components/esp_system/port/cpu_start.c:836
--- 0x40079796: set_cache_and_start_app at /home/yuwei/Workspace/esp/esp-idf/components/bootloader_support/src/bootloader_utility.c:1155
--- (inlined by) unpack_load_app at /home/yuwei/Workspace/esp/esp-idf/components/bootloader_support/src/bootloader_utility.c:890
--- (inlined by) load_image at /home/yuwei/Workspace/esp/esp-idf/components/bootloader_support/src/bootloader_utility.c:800
```

Solar914, 2025-07-25:
如果你的flash支持QIO mode的话，可以在serial flasher config->flash spi mode为QIO试试看

yuwei-virtual, 2025-07-26:
> 如果你的闪存支持QIO模式的话，可以在串行闪光器配置->闪存SPI MODE为QIO试试看

验证过了，还是会出现同样的问题。

[image]

stkw0, 2025-07-29:
I'm facing the same problem with v5.5. (but not v5.4). Changing the flash mode didn't help either.

ginkgm, 2025-08-08:
Hi @stkw0  @yuwei-virtual  @Solar914 ,

I think this issue is related to GCC optimization. Could you provide your sdkconfig file so that we can reproduce this issue easier?

igrr, 2025-08-14:
@stkw0 @Solar914 Could you please check, do you have CONFIG_NEWLIB_NANO_FORMAT option enabled in sdkconfig?

igrr, 2025-08-14:
And please try the workaround suggested in https://github.com/espressif/esp-idf/issues/17436#issuecomment-3188705254 while we are merging the actual fix.