espressif/esp-idf #17015: Seems something wrong on /peripherals/mcpwm/mcpwm_foc_svpwm_open_loop

MDLZCOOL, 2025-07-26:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When I use esp32s3 flash with the example, I got wrong frequency on modulation wave, which is expected to be 50hz.

Any replies are appreciated!

MDLZCOOL, 2025-07-26:
https://github.com/espressif/esp-idf/blob/346870a3044010f2018be0ef3b86ba650251c655/examples/peripherals/mcpwm/mcpwm_foc_svpwm_open_loop/main/app_main.c#L117

Is this correct? If so or not, the macro used is somewhat misleading and the comment is not clear. 

Thanks in advance. I need your help.

wanckl, 2025-07-28:
@MDLZCOOL It is incorrect,, should be:
```c
#define EXAMPLE_FOC_WAVE_FREQ    50
elec_theta_deg += (EXAMPLE_FOC_WAVE_FREQ * 360.f) / (EXAMPLE_FOC_MCPWM_TIMER_RESOLUTION_HZ / EXAMPLE_FOC_MCPWM_PERIOD);
```

Sorry for misleading you, will correct it

MDLZCOOL, 2025-07-28:
I have solved it yesterday. Thanks for replying!

I will close the issue as solved.

closed 2025-07-28 by MDLZCOOL