espressif/esp-idf #17029: ESP-IDF V5.4.2 W5500 SPI Ethernet Basic not working as stand-alone component.

andrew-sz, 2025-07-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.2

### Espressif SoC revision.

Chip is ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Elecrow 7" Advance ESP32S3 Development Kit

### Power Supply used.

USB

### What is the expected behavior?

I expected the component to initialize the W5500 Ethernet interface using either SPI2/SPI3 peripherals

### What is the actual behavior?

I reconfigured the basic ethernet example code to work as a component and it works.  However, when I use this component in any other code like Hello World example, W5500 fails to initialize for incorrect W5500 ID.   

### Steps to reproduce.

[esp-idf_v5.4.2_w5500_basic_example_as_component_in_hello_world_fail.zip](https://github.com/user-attachments/files/21456387/esp-idf_v5.4.2_w5500_basic_example_as_component_in_hello_world_fail.zip)

[esp-idf_v5.4.2_w5500_basic_example_as_component_working.zip](https://github.com/user-attachments/files/21456384/esp-idf_v5.4.2_w5500_basic_example_as_component_working.zip)

1. I'm using exact same hardware and hardware configuration for both sets of software.   So, I don't believe this issue is hardware related. 
2. Run the failing code.  
3. Debug log (posted below) will show that one version of the code initializes successfully the other does not.  I have compared the sdkconfigs and cannot find any substantial difference between the two sets of example code.
...

### Debug Logs.

```plain
/// Working example debug log
PIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x16d0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xec8
load:0x403cb700,len:0x3160
entry 0x403c8950
I (26) boot: ESP-IDF v5.4.2 2nd stage bootloader
I (27) boot: compile time Jul 27 2025 20:29:50
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (30) boot: efuse block revision: v1.3
I (33) qio_mode: Enabling default flash chip QIO
[... 149 lines cut ...]
I (789) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (796) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (0) main_task: Started on CPU1
I (10) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (10) main_task: Calling app_main()
Hello world!
This is esp32s3 chip with 2 CPU core(s), WiFi/BLE, silicon revision v0.2, 16MB external flash
Minimum free heap size: 8367316 bytes
E (130) w5500.mac: W5500 version mismatched, expected 0x04, got 0x00
E (130) w5500.mac: emac_w5500_init(826): verify chip ID failed
E (130) esp_eth: esp_eth_driver_install(228): init mac failed
E (140) ethernet_module: Failed to install Ethernet driver: ESP_ERR_INVALID_VERSION
I (150) gpio: GPIO[46]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (160) main_task: Returned from app_main()
```

### Diagnostic report archive.

[diag-d3fdd2bb-b600-4cb4-86fd-28a11e08394e_w5500_fail.zip](https://github.com/user-attachments/files/21456487/diag-d3fdd2bb-b600-4cb4-86fd-28a11e08394e_w5500_fail.zip)

[diag-c0c52b40-268a-400d-987f-39ad68007440.zip](https://github.com/user-attachments/files/21456489/diag-c0c52b40-268a-400d-987f-39ad68007440.zip)

### More Information.

I just want to emphasize that I'm using the exact same hardware.  I'm not even disconnected the usb cable between tests.   So, I'm really at a loss for the W5500 ID failure.

andrew-sz, 2025-07-28:
[image] What reference do you have the GPIO 46 is Input Only on esp32-s3?   According to the esp32-s3 Series Datasheet v1.9 Page 21 this pin is I/O/T.   Can you provide me with an updated reference?

nopnop2002, 2025-07-28:
@andrew-sz 

My mistake.
GPIO46 can be used for input and output.

andrew-sz, 2025-07-28:
> [@andrew-sz](https://github.com/andrew-sz)
> 
> > This GPIO is input only on the ESP32S3 and cannot be used for CS.
> 
> My mistake. ESP32S2's GPIO46 is input only. ESP32S3's GPIO46 can be used for input and output.
> 
> > #define ETH_SPI_PHY_RST_GPIO    (-1)
> 
> Is the RESET pin on the PHY pulled up?
No problem.  I've also been working with several Espressif chip's at the same time and have to constantly check pin limitations. 

Reset and Intx and CS are pulled up via 10k's.   

Here is relevant section of schematic. 
[image]

andrew-sz, 2025-07-29:
I used a Lilygo T-SIM7080 dev kit to test our custom board that has the w5500 and was able to bring up the W5500 reliably.   There is an SG Micro SGM3799 on the Elecrow board that is used to switch between SPI and I2S for the microphone.  I have tried toggling this analog dp/dt switch but so far still can't get the W5500 configured with the Elecrow.   

That said, with the Lilygo I can confirm the problem is not with SDK and we can close this ticket.   Thanks for your help!