espressif/esp-idf #17038: ESP-IDF ESP32S3 multiple I2C busses crashes

CMakeMain, 2025-07-28:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom PCB

### Power Supply used.

USB

### What is the expected behavior?

I have two I2C busses, one which is used to communicate with the BNO055 (and works fine) and another I2C bus which is supposed to communicate with the STUSB4710. When using both busses at the same time, neither of them work. The main (master) I2C bus talks to my sub mcu (the rp2040) and the STUSB4710. As my PCB has accidentally swapped the pins for SDA and SCL on the rp2040 side, I first attempt to use I2C to configure the STUSB4710, and then swap the I2C pins by deleting the master bus and reinstalling with the pins swapped to talk to the Pico. However I cannot write to either of them.

### What is the actual behavior?

comms_i2c.c
```
#include "comms_i2c.h"

static i2c_master_dev_handle_t BNO055_handle;
static i2c_master_dev_handle_t RP2040_handle;
static i2c_master_dev_handle_t STUSB_handle;
i2c_master_bus_handle_t master_bus_handle;
LineData line_sensor_data_raw;
SemaphoreHandle_t xLineMutex = NULL;

static void stusb_init(uint8_t pdo_reg_addr){
    uint8_t stusb_addr_w = STUSB_ADDR_1 << 1 | WRITE;
    i2c_operation_job_t stusb_config_pdos[] = {
        {.command = I2C_MASTER_CMD_START},
        {.command = I2C_MASTER_CMD_WRITE, .write = {.ack_check = true, .data = &stusb_addr_w, .total_bytes = 1}},
        {.command = I2C_MASTER_CMD_WRITE, .write = {.ack_check = true, .data = &pdo_reg_addr, .total_bytes = 1}},
        {.command = I2C_MASTER_CMD_WRITE, .write = {.ack_check = true, .data = &PDO_25W[0], .total_bytes = 3}},
        {.command = I2C_MASTER_CMD_WRITE, .write = {.ack_check = false, .data = &PDO_25W[3], .total_bytes = 1}},
        {.command = I2C_MASTER_CMD_STOP},
    };
    esp_err_t pdo_problem = i2c_master_execute_defined_operations(STUSB_handle, stusb_config_pdos, sizeof(stusb_config_pdos)/sizeof(i2c_operation_job_t), pdMS_TO_TICKS(150));
    if (pdo_problem != ESP_OK){
        ESP_LOGE("PDO INIT", "Error: %s", esp_err_to_name(pdo_problem));
    }
}

void bno_comms_init(i2c_port_t port){
    // specialised driver + device settings
    esp_err_t ret = 0;
[... 410 lines cut ...]

```plain
ESP32S3 main bus (Doesn't even contain any decodable data).

<img width="1855" height="451" alt="Image" src="https://github.com/user-attachments/assets/78086133-5761-45d3-b8f8-55b06f3febaf" />
```

### Diagnostic report archive.

### More Information.

After letting this issue run for a while, it will cause a stack over flow: 
```
***ERROR*** A stack overflow in task Tmr Svc has been detected.

Backtrace: 0x40375bdd:0x3fc9acf0 0x4037aead:0x3fc9ad10 0x4037bd26:0x3fc9ad30 0x4037d3cb:0x3fc9adb0 0x4037bdec:0x3fc9add0 0x4037bde2:0xa5a5a5a5 |<-CORRUPTED
--- 0x40375bdd: panic_abort at C:/Espressif/frameworks/esp-idf-v5.4.1/components/esp_system/panic.c:454
--- 0x4037aead: esp_system_abort at C:/Espressif/frameworks/esp-idf-v5.4.1/components/esp_system/port/esp_system_chip.c:87
--- 0x4037bd26: vApplicationStackOverflowHook at C:/Espressif/frameworks/esp-idf-v5.4.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:563
--- 0x4037d3cb: vTaskSwitchContext at C:/Espressif/frameworks/esp-idf-v5.4.1/components/freertos/FreeRTOS-Kernel/tasks.c:3701 (discriminator 7)
--- 0x4037bdec: _frxt_dispatch at C:/Espressif/frameworks/esp-idf-v5.4.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:451
--- 0x4037bde2: _frxt_int_exit at C:/Espressif/frameworks/esp-idf-v5.4.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:246
```

May have something to do with the excessive logging maybe?

mythbuster5, 2025-07-28:
Could you please verify each of them can work separately, but can't use at the same time?

CMakeMain, 2025-07-29:
@mythbuster5 If it helps, I get this error when writing to the STUSB4710 with the esp_err_to_name():

[image]

CMakeMain, 2025-07-31:
Hi, I just found another error with my read function. I intend to ACK, however it is detecting NAKs...

[image]

```
s8 pico_read(u8* reg_data, u8* checksum, size_t size){
    uint8_t addr_r = PICO_ADDR << 1 | READ;
    i2c_operation_job_t pico_read[] = {
        {.command = I2C_MASTER_CMD_START},
        {.command = I2C_MASTER_CMD_WRITE, .write = {.ack_check = true, .data = &addr_r, .total_bytes = 1}},
        {.command = I2C_MASTER_READ, .read = {.ack_value = I2C_ACK_VAL, .data = reg_data, .total_bytes = size}},
        {.command = I2C_MASTER_READ, .read = {.ack_value = I2C_NACK_VAL, .data = checksum, .total_bytes = 1}},
        {.command = I2C_MASTER_CMD_STOP},
    };
    esp_err_t ret = i2c_master_execute_defined_operations(RP2040_handle, pico_read, sizeof(pico_read)/sizeof(i2c_operation_job_t), pdMS_TO_TICKS(250));
    if(ret != ESP_OK){
        ESP_ERROR_CHECK(i2c_master_bus_reset(master_bus_handle));
        ESP_LOGE("Pico Read", "Read failed... Reseting bus");
    }
    return ret;
}
```

[image]