espressif/esp-idf #17040: ESP32-P4 Guru Meditation in jpeg_encoder_process without error return on failure

DT-art1, 2025-07-28:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5

### Espressif SoC revision.

Chip is ESP32-P4 (revision v1.0)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

esp32-p4-evboard

### Power Supply used.

USB

### What is the expected behavior?

The encoder should return an error code when the encoding fails instead of crashing.

### What is the actual behavior?

The encoder crashes without a proper error return, leading to a stack corruption or Guru Meditation errors. In some cases, the memory appears to be overwritten silently before the crash.

### Steps to reproduce.

```
#include <stdio.h>
#include "esp_log.h"
#include "driver/jpeg_encode.h"

jpeg_encoder_handle_t encoder_engine;
jpeg_encode_engine_cfg_t encode_eng_cfg = {
    .intr_priority = 0,
    .timeout_ms = 100,
};

void generate_image(uint8_t *rgb)
{
    static uint8_t frame_count = 0;

    int x = 0;
    int y = 0;
    int bytes_per_row = 256 * 3;
    int bytes_per_pixel = 3;
    while (x < 256) {
        while (y < 256) {
            int idx = y * bytes_per_row + x * bytes_per_pixel;
            rgb[idx + 0] = frame_count + x;
            rgb[idx + 1] = frame_count + y;
            rgb[idx + 2] = frame_count + x + y;
            ++y;
[... 188 lines cut ...]
4ff18540: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff18560: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff18580: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5
4ff185a0: 0xa5a5a5a5 0x4ff18440 0x00000600 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff185c0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff185e0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff18600: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff18620: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff18640: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff18660: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff18680: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff186a0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: 19f7fe7f0

Rebooting...
```

### Diagnostic report archive.

[idf-diag-62f26cb7-1a3b-467e-8406-00faac61f9a8.zip](https://github.com/user-attachments/files/21465044/idf-diag-62f26cb7-1a3b-467e-8406-00faac61f9a8.zip)

### More Information.

Issue shows up in **framework-espidf @ 3.50302.0 (5.3.2)** as well.

mythbuster5, 2025-07-29:
May you try a larger jpeg size ? : `size_t jpeg_size = 19660;       // 196608/10`

mythbuster5, 2025-07-30:
https://github.com/espressif/esp-idf/blob/v5.5/components/esp_driver_jpeg/jpeg_encode.c#L264 Wait, I remember I once added this thing on 5.5, but seems it didn't report this issue in your case? Could you add some log to verify?

DT-art1, 2025-07-30:
I increased the log level to DEBUG and modified one line in the test to:
```
    jpeg_encode_cfg_t enc_config = {
        .height = 256,
        .width = 256,
        //.src_type = JPEG_ENCODE_IN_FORMAT_RGB888,
        .src_type = 1234,
        .sub_sample = JPEG_DOWN_SAMPLING_YUV444,
        .image_quality = 95,
    };
```
With this change, I got the following output, confirming that the logging is active:
```
--- 0x400004fe: __esp_system_init_fn_init_disable_rtc_wdt at C:/Espressif/frameworks/esp-idf-v5.5/components/esp_system/startup_funcs.c:168
D (847) intr_alloc: Connected src 79 to int 1 (cpu 0)
D (851) app_start: Starting scheduler on CPU0
D (855) intr_alloc: Connected src 53 to int 2 (cpu 0)
D (855) intr_alloc: Connected src 80 to int 0 (cpu 1)
D (855) app_start: Starting scheduler on CPU1
D (865) intr_alloc: Connected src 54 to int 1 (cpu 1)
I (855) main_task: Started on CPU0
D (875) heap_init: New heap initialised at 0x4ff3afc0
D (875) heap_init: New heap initialised at 0x4ff40000
I (885) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
D (885) esp_psram: Allocating block of size 32768 bytes
D (895) intr_alloc: Connected src 48 to int 3 (cpu 0)
I (895) main_task: Calling app_main()
I (905) TEST: RGB Buffer Size: 196608, JPEG Buffer Size: 19712
D (905) jpeg.common: new jpeg module has been acquired at (0x480357c4)
D (915) intr_alloc: Connected src 95 to int 4 (cpu 0)
D (915) intr_alloc: Connected src 104 to int 4 (cpu 0)
D (925) intr_alloc: Connected src 105 to int 4 (cpu 0)
D (925) intr_alloc: Connected src 106 to int 4 (cpu 0)
D (935) intr_alloc: Connected src 107 to int 4 (cpu 0)
D (935) intr_alloc: Connected src 108 to int 4 (cpu 0)
E (955) jpeg.encoder: wrong, we don't support encode from such format.
E (955) TEST: ERROR: 262
I (955) main_task: Returned from app_main()

Done
```
However, when running the original (unaltered) test with log level set to DEBUG, no additional log messages from the encoder appear:
```
D (499) cpu_start: calling init function: 0x40000236 on core: 0
--- 0x40000236: __esp_system_init_fn_init_efuse_check at C:/Espressif/frameworks/esp-idf-v5.5/components/efuse/src/esp_efuse_startup.c:37
D (500) cpu_start: calling init function: 0x40000400 on core: 0
--- 0x40000400: __esp_system_init_fn_init_show_cpu_freq at C:/Espressif/frameworks/esp-idf-v5.5/components/esp_system/startup_funcs.c:64
I (500) cpu_start: Pro cpu start user code
I (503) cpu_start: cpu freq: 360000000 Hz
D (507) cpu_start: calling init function: 0x400000e0 on core: 0
--- 0x400000e0: __esp_system_init_fn_init_show_app_info at C:/Espressif/frameworks/esp-idf-v5.5/components/esp_app_format/esp_app_desc.c:123
I (513) app_init: Application information:
I (516) app_init: Project name:     p4_jpeg_encoder_test
I (521) app_init: App version:      1
I (525) app_init: Compile time:     Jul 30 2025 13:12:08
I (530) app_init: ELF file SHA256:  c1c20f850...
I (534) app_init: ESP-IDF:          v5.5
D (538) cpu_start: calling init function: 0x40000248 on core: 0
--- 0x40000248: __esp_system_init_fn_init_efuse_show_app_info at C:/Espressif/frameworks/esp-idf-v5.5/components/efuse/src/esp_efuse_startup.c:47
I (544) efuse_init: Min chip rev:     v0.1
I (547) efuse_init: Max chip rev:     v1.99
I (551) efuse_init: Chip rev:         v1.0
D (555) cpu_start: calling init function: 0x40002e4e on core: 0
--- 0x40002e4e: __esp_system_init_fn_init_heap at C:/Espressif/frameworks/esp-idf-v5.5/components/heap/heap_caps_init.c:26
D (561) memory_layout: Checking 8 reserved memory ranges:
D (566) memory_layout: Reserved memory range 0x30100000 - 0x30100068
--- 0x30100000: clk_ll_mpll_enable at C:/Espressif/frameworks/esp-idf-v5.5/components/hal/esp32p4/include/hal/clk_tree_ll.h:156
--- (inlined by) rtc_clk_mpll_enable at C:/Espressif/frameworks/esp-idf-v5.5/components/esp_hw_support/port/esp32p4/rtc_clk.c:592
D (572) memory_layout: Reserved memory range 0x48000000 - 0x4c000000
D (578) memory_layout: Reserved memory range 0x4ff00000 - 0x4ff11d80
[... 87 lines cut ...]
D (916) intr_alloc: Connected src 104 to int 4 (cpu 0)
D (926) intr_alloc: Connected src 105 to int 4 (cpu 0)
D (926) intr_alloc: Connected src 106 to int 4 (cpu 0)
D (936) intr_alloc: Connected src 107 to int 4 (cpu 0)
D (936) intr_alloc: Connected src 108 to int 4 (cpu 0)
I (956) TEST: Frame: 0, Bytes: 15425
I (966) TEST: Frame: 1, Bytes: 17691
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4ff084ae  RA      : 0x4000cebc  SP      : 0x4ff186a0  GP      : 0x4ff12580
--- Stack dump detected
--- 0x4ff084ae: xQueueGenericSend at C:/Espressif/frameworks/esp-idf-v5.5/components/freertos/FreeRTOS-Kernel/queue.c:937
--- 0x4000cebc: jpeg_encoder_process at C:/Espressif/frameworks/esp-idf-v5.5/components/esp_driver_jpeg/jpeg_encode.c:283
TP      : 0x4ff18800  T0      : 0x4fc10cc4  T1      : 0xfffffff0  T2      : 0xffff0879
S0/FP   : 0xd19fb6c3  S1      : 0x00004db6  A0      : 0xd19fb6c3  A1      : 0x00000000
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x480357c4
A6      : 0x08000000  A7      : 0x4fc10a5c  S2      : 0x00000000  S3      : 0x00000028
S4      : 0x00000000  S5      : 0x48000aa0  S6      : 0x4ff18768  S7      : 0x4ff3b440
S8      : 0x00030000  S9      : 0x02000000  S10     : 0x00000018  S11     : 0x00000000
T3      : 0x00000000  T4      : 0x00004ce1  T5      : 0xffffff78  T6      : 0xffffffc9
MSTATUS : 0x00011880  MTVEC   : 0x4ff00003  MCAUSE  : 0x00000005  MTVAL   : 0xd19fb703
--- 0x4ff00003: _vector_table at ??:?
MHARTID : 0x00000000
```