espressif/esp-idf #17041: ESP-IDF ESP32C5 flash inside docker running on raspberry py fails

mgandler-vahle, 2025-07-28:
Hi,

I am trying to run the esp-idf docker image on a raspberry pi 5 to build my application and flash it onto the device.

``

mgandler-vahle, 2025-07-29:
Update:

I tried the following combinations **without** success:

- Docker ESP-IDF & VM (VirtualBox) with Ubuntu **24.04** running on Windows 11 with USB passthrough
- VM (VirtualBox) with Ubuntu **24.04** running on Windows 11 with USB passthrough and esptool
- Docker ESP-IDF & VM (VirtualBox) with Ubuntu **22.04** running on Windows 11 with USB passthrough
- VM (VirtualBox) with Ubuntu **22.04** running on Windows 11 with USB passthrough and esptool

- Docker ESP-IDF & Raspberry PI 5 with Ubuntu **25.04** 
- Docker ESP-IDF & Raspberry PI 5 with Ubuntu **25.04** with telnet port passthrough
- Raspberry PI 5 with Ubuntu **25.04** and esptool
- Docker ESP-IDF & Raspberry PI 5 with Ubuntu **24.04** 
- Raspberry PI 5 with Ubuntu **24.04** and esptool

So far the only way it works is with WSL and usbipd on Windows.

dobairoland, 2025-08-04:
> So far the only way it works is with WSL and usbipd on Windows.

Then I'm afraid you have to stick to it. The issue is with the signals not properly transferred from the docker to the serial port. This is a Windows issue. Some is able to solve such issues with trying different drivers. But since the WSL solution is working for you, I'm guessing the issue is with the docker. 

Holding the BOOT button while starting esptool might also help to work around the issue.

I'm sorry but on our end I don't think we can do anything about this.

dobairoland, 2025-08-04:
Oh, sorry. I've got a little confused.

> espressif/idf:release-v5.4

ESP32-C5 is supported in the v5.5 release (preview support only). But you have the v0.1 version which is not the mass production version and esp-idf as well as esptool is shifting for supporting v1.0 - this one will be fully supported in ESP-IDF v5.5.1.

Flashing from docker running on RPI is tested well in our infrastructure and shouldn't be any issues with it.

Which version of esptool is working for you on Windows WSL? Could you try the same one on RPI? First outside of docker and then in docker as well. You don't have to experiment with Ubuntu or ESP-IDF versions for now.

mgandler-vahle, 2025-08-04:
I am using docker image v5.4 which has esptool version 4.8.1 on desktop (maybe a difference between x64 and arm64) and 4.9.0 in the docker image for the raspberry. 

I manually downgraded the esptool inside the docker container on the raspberry pi with `pip3 install "esptool==4.8.1"` and afterwards the flashing process worked.

It seems like esptool versions > 4.8.1 are unable to set the internal RTS pin or other signals through the ACM interface which are necessary.