espressif/esp-idf #17046: How to change spi flash clk in app partition code?

wishys, 2025-07-29:
### Is your feature request related to a problem?

In early products, the SPI flash was configured with a clock speed of 40 MHz. Now, there is a need to configure the SPI flash to 80 MHz. Is such an operation supported? 
I noticed in `spi_falsh.h` has such `esp_rom_spiflash_config_clk() `, but the IDF states that it should not be called within the SDK. However, after calling it in the app_main entry point, like 
```
void app_main(void)
{
    esp_rom_spiflash_result_t ret = esp_rom_spiflash_config_clk(1,0);
    ESP_LOGI(TAG,"set spiflash clk return %d ",ret );
....
}
```
it indeed achieved the desired functionality. What issues might arise from doing this?

mythbuster5, 2025-07-29:
Which esp chip are you using?

wishys, 2025-07-30:
esp32c3 @mythbuster5