espressif/esp-idf #17054: Compile error for ULP LP Core

simoberny, 2025-07-30:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Trying a simple example for ULP on a ESP32P4 in a C++ application. 

### What is the actual behavior?

I got a build error both on v5.4.1 and v5.5 on idf lp_core.c file, due to compiler considering the `_rtc_ulp_memory_start` as an array. 
The error is therefore caused by the c++ compiler. 
If I directly compile the example from idf folder I got no compilation error. 

### Steps to reproduce.

Compile a ULP example with C++ compiler

### Build or installation Logs.

```plain
C:/Users/--/esp/v5.5/esp-idf/components/ulp/lp_core/lp_core.c: In function 'ulp_lp_core_run':
C:/Users/--/esp/v5.5/esp-idf/components/ulp/lp_core/lp_core.c:40:62: error: array subscript 32 is outside array bounds of 'uint32_t[1]' {aka 'long unsigned int[1]'} [-Werror=array-bounds=]
   40 | #define RESET_HANDLER_ADDR (intptr_t)(&_rtc_ulp_memory_start + 0x80 / 4) // Placed after the 0x80 byte long vector table
      |                                      ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
C:/Users/--/esp/v5.5/esp-idf/components/ulp/lp_core/lp_core.c:77:21: note: in expansion of macro 'RESET_HANDLER_ADDR'
   77 |         boot_addr = RESET_HANDLER_ADDR;
      |                     ^~~~~~~~~~~~~~~~~~
C:/Users/--/esp/v5.5/esp-idf/components/ulp/lp_core/lp_core.c:33:17: note: at offset 128 into object '_rtc_ulp_memory_start' of size 4
   33 | extern uint32_t _rtc_ulp_memory_start;
      |                 ^~~~~~~~~~~~~~~~~~~~~
C:/Users/--/esp/v5.5/esp-idf/components/ulp/lp_core/lp_core.c:40:62: error: array subscript 32 is outside array bounds of 'uint32_t[1]' {aka 'long unsigned int[1]'} [-Werror=array-bounds=]
   40 | #define RESET_HANDLER_ADDR (intptr_t)(&_rtc_ulp_memory_start + 0x80 / 4) // Placed after the 0x80 byte long vector table
      |                                      ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
C:/Users/--/esp/v5.5/esp-idf/components/ulp/lp_core/lp_core.c:89:37: note: in expansion of macro 'RESET_HANDLER_ADDR'
   89 |     lp_core_ll_set_app_boot_address(RESET_HANDLER_ADDR);
      |                                     ^~~~~~~~~~~~~~~~~~
C:/Users/--/esp/v5.5/esp-idf/components/ulp/lp_core/lp_core.c:33:17: note: at offset 128 into object '_rtc_ulp_memory_start' of size 4
   33 | extern uint32_t _rtc_ulp_memory_start;
      |                 ^~~~~~~~~~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
```

### More Information.

Right now I resolved changing the declaration of RESET_HANDLER_ADDR with: 

`uintptr_t addr = (uintptr_t)(&_rtc_ulp_memory_start) + 0x80;`

directly inside the `ulp_lp_core_run` function.
Now the code compile and run fine.

o-marshmallow, 2025-07-31:
Hello @simoberny ,

I was not able to reproduce your issue on my end with IDF v5.4.1. CMake is able to differenciate between C files and C++ files and compile them using `gcc` and `g++` respetively, so I am a bit surprised you get this error when compiling this C file.

 Did you add any particular compile option in your `CMakeLists.txt`?

simoberny, 2025-07-31:
Thank you @o-marshmallow. 
No I think I don't have any particular compile option that can cause that. 

Top dir CMakeLists
```cmake
cmake_minimum_required(VERSION 3.16)
set(CMAKE_OBJECT_PATH_MAX 512)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
set(CMAKE_CXX_STANDARD 17)

set(EXCLUDE_COMPONENTS esp_lcd)
set(EXTRA_COMPONENT_DIRS "./components/esp-dl")
add_definitions(-DSPDLOG_NO_EXCEPTIONS)

project(Name VERSION 0.8.0)
```

Main CMakeLists
```cmake
cmake_minimum_required(VERSION 3.16)

idf_build_get_property(component_targets __COMPONENT_TARGETS)
if ("___idf_espressif__esp-dl" IN_LIST component_targets)
    idf_component_get_property(espdl_dir espressif__esp-dl COMPONENT_DIR)
elseif("___idf_esp-dl" IN_LIST component_targets)
    idf_component_get_property(espdl_dir esp-dl COMPONENT_DIR)
endif()

set(cmake_dir ${espdl_dir}/fbs_loader/cmake)
include(${cmake_dir}/utilities.cmake)

set(src_dirs        .
                    src/)

set(include_dirs    ./include/
                    .)

set(requires    esp_wifi
                esp_video
                esp-dl
                esp_http_server
                spiffs
                spdlog
                ulp)

set(packed_model ${BUILD_DIR}/espdl_models/pedestrian_detect.espdl)

idf_component_register(SRC_DIRS ${src_dirs}
                    REQUIRES ${requires} 
                    PRIV_REQUIRES nvs_flash esp_hosted esp_driver_sdmmc esp_driver_jpeg esp_mm esp32_p4_function_ev_board esp_video bsp_extra esp_eth
                    INCLUDE_DIRS ${include_dirs}
                    )
                    
# Webpage static file
set(STATIC_FILES_DIR "${CMAKE_SOURCE_DIR}/static")
spiffs_create_partition_image(storage "${STATIC_FILES_DIR}" FLASH_IN_PROJECT)

# YOLOV11 model
set(models_dir ${COMPONENT_DIR}/models)

file(MAKE_DIRECTORY ${BUILD_DIR}/espdl_models)
set(models)
list(APPEND models ${models_dir}/y11_384_p4_v9.espdl)

set(pack_model_exe ${espdl_dir}/fbs_loader/pack_espdl_models.py)

add_custom_command(
    OUTPUT ${packed_model}
    COMMENT "Move and Pack models..."
    COMMAND python ${pack_model_exe} --model_path ${models} --out_file ${packed_model}
    DEPENDS ${models}
    VERBATIM)

add_custom_target(COCO_DETECT_MODEL ALL DEPENDS ${packed_model})
add_dependencies(flash COCO_DETECT_MODEL)
esptool_py_flash_to_partition(flash "coco_det" ${packed_model})

# ULP
set(ulp_app_name lp_core_${COMPONENT_NAME})
set(ulp_lp_core_sources "lp_core/main.c")
set(ulp_exp_dep_srcs "main.cpp")
ulp_embed_binary(${ulp_app_name} "${ulp_lp_core_sources}" "${ulp_exp_dep_srcs}")
```

simoberny, 2025-08-01:
I'm using the same version as you
```bash
riscv32-esp-elf-gcc (crosstool-NG esp-14.2.0_20241119) 14.2.0
```

I don't get any compilation error by compiling the examples from idf folder. I don't undestand what in my application is causing this strange behaviour. The only thing that I thought could cause the problem is that my main code, where I actually include the lp_core resources, is C++ code.

closed 2025-08-01 by simoberny

ESP-Marius, 2025-08-26:
@vortigont are you able to provide a way we can reproduce this?

Or the full output from the compilation process

vortigont, 2025-08-26:
@ESP-Marius,
OK, I nailed it to the steps where it reproduces on Ubuntu x64 and arm64

Initial conditions - clean user env, only prereqs are installed as in the [docs](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/linux-macos-setup.html#step-1-install-prerequisites)

**Step 1.**
```sh
mkdir -p /opt/build/espressif
cd /opt/build/espressif
git clone --depth 1 -b v5.5 --recursive https://github.com/espressif/esp-idf.git
```

**Step 2.**
```sh
cd esp-idf
./install.sh
. export.sh
```

**Step 3.**
```sh
cd /opt/build/
git clone --depth 1 -b v5.5_Arduino_v https://github.com/vortigont/esp32-arduino-lib-builder.git
cd esp32-arduino-lib-builder/
./build.sh -t esp32p4
```

it will eventually fail with
```
/opt/build/espressif/esp-idf/components/ulp/lp_core/lp_core.c: In function 'ulp_lp_core_run':
/opt/build/espressif/esp-idf/components/ulp/lp_core/lp_core.c:40:62: error: array subscript 32 is outside array bounds of 'uint32_t[1]' {aka 'long unsigned int[1]'} [-Werror=array-bounds=]
   40 | #define RESET_HANDLER_ADDR (intptr_t)(&_rtc_ulp_memory_start + 0x80 / 4) // Placed after the 0x80 byte long vector table
      |                                      ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
```

The repo from step 3 is a bit customized build env from [Tasmota](https://github.com/Jason2866/esp32-arduino-lib-builder).
It installs and runs it's own copy of IDF to build the the Arduino libs.

Now the most interesting part - if I skip *Step 2* and won't do `. export.sh` from vanilla IDF but just go to *Step 3* it will complete without the issues and builds the libs as expected.