espressif/esp-idf #17061: SLEEP current: ESP32-C5-DeVKitC-1

loraadh, 2025-07-31:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I measure current consumption of the board (ESP32-C5-DeVKitC-1) in power save mode (DTIM).
I use the project powersave for that purpose. I've installed it from the "examples" folder of ESP-IDF V5.5 using VSCode.
I am measuring 150uA during the SLEEP interval between DTIMs. 
Is there a way to configure the chip to lower the current consumption in SLEEP mode between DTIMs? sdkconfig file contains lots of paramters that I can config to lower the current but I don't know how to do it.
thanks 
BR

esp-lis, 2025-08-01:
@loraadh The ECO2 version of the ESP32-C5 does not support powering down digital peripherals, but this issue will be fixed in a future version.

loraadh, 2025-08-01:
thanks @esp-lis 
do you mean that with ECO2 none of the peripherals is switched off during the SLEEP between DTIMs? if not, do you know the list of peripherals that are switched off? 
BR