espressif/esp-idf #17063: Questions about using wifi_prov_mgr_start_provisioning

shuofeng1, 2025-08-01:
The wifi_prov_mgr_start_provisioning api will automatically connect to wifi, etc. How do I configure parameters such as wifi_sta_config_t.threshold.authmode during use? I need to send ssid and password through the client

mahavirj, 2025-08-01:
@shuofeng1 

For advanced STA configuration settings you may use following event in the application:
https://github.com/espressif/esp-idf/blob/4e036983a751e4667ade94c8f6f6bf1e7f78eff0/examples/provisioning/wifi_prov_mgr/main/app_main.c#L290-L297

Please refer to [Wi-Fi provisioning documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/provisioning/index.html) for details on how the device and provisioning entity communicate for sharing the credentials. 

Hope this helps!

shuofeng1, 2025-08-01:
> [@shuofeng1](https://github.com/shuofeng1)
> 
> For advanced STA configuration settings you may use following event in the application:对于高级 STA 配置设置，您可以在应用程序中使用以下事件：
> 
> [esp-idf/examples/provisioning/wifi_prov_mgr/main/app_main.c](https://github.com/espressif/esp-idf/blob/4e036983a751e4667ade94c8f6f6bf1e7f78eff0/examples/provisioning/wifi_prov_mgr/main/app_main.c#L290-L297)
> 
> Lines 290 to 297 in [4e03698](/espressif/esp-idf/commit/4e036983a751e4667ade94c8f6f6bf1e7f78eff0)
> 
>          case WIFI_PROV_SET_STA_CONFIG: { 
>              /** 
>               * Wi-Fi configurations can be set here before the Wi-Fi is enabled in 
>               * STA mode. 
>              */ 
>              wifi_config_t *wifi_config = (wifi_config_t*)event_data; 
>              (void) wifi_config; 
>              break; 
> Please refer to [Wi-Fi provisioning documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/provisioning/index.html) for details on how the device and provisioning entity communicate for sharing the credentials.有关设备和配置实体如何通信以共享凭据的详细信息，请参阅 [Wi-Fi 配置文档 ](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/provisioning/index.html)。
> 
> Hope this helps!  希望这有帮助！

@mahavirj I have tried:

[image]
But it seems that the default configuration will be used regardless of whether I have any configuration:

[image]

hrushikesh430, 2025-08-14:
Hello @shuofeng1, I tried to reproduce your issue locally, could you please check if you have enabled `EXAMPLE_PROV_ENABLE_APP_CALLBACK` config from menuconfig. 

If you are still facing issue, even after proivisioning, please let me know.

shuofeng1, 2025-08-20:
> Hello [@shuofeng1](https://github.com/shuofeng1), I can see in your sdkconfig.txt `CONFIG_EXAMPLE_PROV_ENABLE_APP_CALLBACK` is not enabled. This config enables the application callback feature. Please enable and check if it works.您好，我发现您的 sdkconfig.txt 文件里 `CONFIG_EXAMPLE_PROV_ENABLE_APP_CALLBACK` 未启用。此配置启用了应用程序回调功能。请启用它并检查是否正常工作。

@hrushikesh430 Sorry, it's my problem. The case I provided before was to receive the WIFI_PROV_SET_STA_CONFIG event in esp_event_handler for configuration. Now everything is normal.

hrushikesh430, 2025-08-20:
@shuofeng1 Closing this issue, if you face any issue, please feel free to reopen this issue. Thanks.

closed 2025-08-20 by hrushikesh430