espressif/esp-idf #17066: Add ESP_RST_ROLLBACK enumerator to esp_reset_reason_t

ddomnik, 2025-08-01:
### Is your feature request related to a problem?

If a rollback has happened, I may want to send logs to a backend. How can this be easily detected?
Currently, it seems a reset performed via esp_ota_mark_app_invalid_rollback_and_reboot() can not be detected based on built in functions.

is detected as ESP_RST_SW (3) from esp_reset_reason_t
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html#_CPPv418esp_reset_reason_t

### Describe the solution you'd like.

Add **ESP_RST_ROLLBACK** to **esp_reset_reason_t**

### Describe alternatives you've considered.

For now it seems the only alternative would be to use flash or RTC memory and build a custom logic.

o-marshmallow, 2025-08-04:
Hello @ddomnik ,

One solution I can think of in your case would be to override the function `esp_ota_mark_app_invalid_rollback_and_reboot` in your project by wrapping it (https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html#wrappers-to-redefine-or-extend-existing-functions) Then in your new implementation, you can use flash or RTC memory to store your rollback state.

I checked if it would be possible to define a new custom reset reason without modifying IDF, it would be possible but would be a bit complex since you'd have to implement a workaround to prevent `esp_reset_reason_init` from erasing the custom hint... On boot, it erases any hint previously set using ` esp_reset_reason_set_hint`. The latter, always clears the hint expcet if it is `ESP_RST_UNKNOWN`. So one way to prevent it from clearing the hint, would be to also override `esp_reset_reason_get_hint` function and make sure it returns `ESP_RST_UNKNOWN` if your custom hint is detected.

That's a dirtier and harder to maintain solution than using flash or RTC in my opinion.

ddomnik, 2025-08-04:
@o-marshmallow thanks. Currently, I go with the RTC memory. However, adding ESP_RST_ROLLBACK  would be a nice feature.