espressif/esp-idf #17070: esp_wifi_80211_tx en_sys_seq error

sysytwl, 2025-08-02:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.5.0

### Espressif SoC revision.

Chip is ESP32-D0WDQ6 (revision v1.0)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32 cam AI thinker

### Power Supply used.

USB

### What is the expected behavior?

working

### What is the actual behavior?

ESP_ERR_INVALID_ARG: en_sys_seq must be true balabala
ap mode, no connection,
problem: Source address(mac) must not equal to wifi ap mac.
it will cause this error, if u use it, even without connections.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

zhangyanjiaoesp, 2025-08-04:
@sysytwl 
Are you sending a broadcast frame using 802.11 TX? If it’s a broadcast frame and the source address is equal to the AP’s MAC address, it will return: `ESP_ERR_INVALID_ARG: en_sys_seq should be true to avoid side-effect...`

sysytwl, 2025-08-04:
Hi, yes. I changed the MAC address. I think this should be mentioned in the documentation or header file. It took me three days to figure it out.

zhangyanjiaoesp, 2025-08-05:
> Hi, yes. I changed the MAC address. I think this should be mentioned in the documentation or header file. It took me three days to figure it out.
 we will update the doc, thanks for reporting.

sysytwl, 2025-12-29:
bandwidth is not working for esp_wifi_80211_tx 
V5.5.1
Hi, I find that bandwidth is not working for esp_wifi_80211_tx
esp_wifi_set_bandwidth(WIFI_IF_STA, WIFI_BW_HT40);
ESP_ERROR_CHECK(esp_wifi_config_80211_tx_rate(WIFI_IF_STA, WIFI_PHY_RATE_MCS7_SGI));

but in wireshark: bandwidth is still 20MHz, data rate is 72.2