espressif/esp-idf #17236: nimble_port_deinit fails if CONFIG_BT_NIMBLE_GATT_SERVER is not set

mkellner, 2025-08-04:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

If `CONFIG_BT_NIMBLE_GATT_SERVER` is not set, the call to `nimble_port_deinit` will fail.

IDF v5.4 (and prior) did not exhibit the issue.

### What is the actual behavior?

The final link fails:

```
/Users/mkellner/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/bt/libbt.a(ble_hs.c.obj):(.literal.ble_hs_deinit+0x8): undefined reference to `ble_gatts_stop'
/Users/mkellner/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/bt/libbt.a(ble_hs.c.obj): in function `ble_hs_deinit':
/Users/mkellner/esp32/esp-idf-v5.5/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:933:(.text.ble_hs_deinit+0x50): undefined reference to `ble_gatts_stop'
collect2: error: ld returned 1 exit status
```

### Steps to reproduce.

1. Using the [Moddable SDK](https://github.com/Moddable-OpenSource/moddable?tab=readme-ov-file#esp32-by-espressif) install the Moddable SDK and Espressif tools as directed.
2. Build the BLE scanner example:
```
cd $MODDABLE/examples/network/ble/scanner
mcconfig -d -m -p esp32/esp32s3
```
3. The final link fails:

```
/Users/mkellner/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/bt/libbt.a(ble_hs.c.obj):(.literal.ble_hs_deinit+0x8): undefined reference to `ble_gatts_stop'
/Users/mkellner/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/bt/libbt.a(ble_hs.c.obj): in function `ble_hs_deinit':
/Users/mkellner/esp32/esp-idf-v5.5/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:933:(.text.ble_hs_deinit+0x50): undefined reference to `ble_gatts_stop'
collect2: error: ld returned 1 exit status
```

### Build or installation Logs.

```plain

```

mkellner, 2025-08-04:
`ble_gatts_stop` is defined in components/bt/host/nimble/nimble/nimble/host/src/ble_gatts.c

It is in an #if block for MYNEWT_VAL(BLE_GATTS) which gets assigned CONFIG_BT_NIMBLE_GATT_SERVER in components/bt/host/nimble/port/include/esp_nimble_cfg.h

For the Moddable scanner example, CONFIG_BT_NIMBLE_GATT_SERVER is not set, so the ble_gatts_stop is never defined and the build fails.

rahult-github, 2025-08-05:
Hi @mkellner , 

Thanks for reporting. This issue has been fixed currently. You can cherry-pick the [commit](https://github.com/espressif/esp-nimble/commit/9d164c2b1b3ba27d5e72a2ce9388fa0ad77b0ee9) and try.