espressif/esp-idf #17247: ble50_security_server示例如何兼容 被BLE 4.2 和 BLE 5.0 客户端发现，连接上

Roy-ldh, 2025-08-06:
### Is your feature request related to a problem?

请问如何基于gatt security server 或者ble50_security_server示例例启用 BLE 5.0 独有的性能特性，兼容 BLE 4.2 和 BLE 5.0 客户端发现？
1. BLE 4.2 客户端依然可以发现、连接和加密，只是无法使用 BLE 5.0 独有的新特性。
2. BLE 5.0 客户端可以发现、连接和加密，能使用 BLE 5.0 独有的新特性。

esp-zhp, 2025-08-07:
您好，感谢您的反馈！

关于您提到的 **`ble50_security_server` 示例如何兼容 BLE 4.2 和 BLE 5.0 客户端** 的问题，这边说明如下：

* 配置项 `BT_BLE_50_FEATURES_SUPPORTED` 和 `BT_BLE_42_FEATURES_SUPPORTED` **不能同时打开**，因为这两个配置项之间存在不兼容性。如果希望启用 BLE 5.0 的特性，请仅开启 `BT_BLE_50_FEATURES_SUPPORTED`。

* 但请放心，即使只启用了 BLE 5.0（即开启 `BT_BLE_50_FEATURES_SUPPORTED`），**仍然可以兼容 BLE 4.2 客户端**，包括：

  * 被 BLE 4.2 客户端发现
  * 建立连接
  * 完成加密配对

* 举个例子：在使用 BLE 5.0 的 Extended Advertising 功能时，您可以通过设置 legacy 类型的广播参数来兼容 BLE 4.2 客户端：

```c
esp_ble_gap_ext_adv_set_params(...);
```

设置参数时，可选择以下广播类型（这些广播类型兼容 BLE 4.2 客户端）：

```c
/*!< ADV_IND */
#define ESP_BLE_GAP_SET_EXT_ADV_PROP_LEGACY_IND

/*!< ADV_DIRECT_IND (low duty cycle) */
#define ESP_BLE_GAP_SET_EXT_ADV_PROP_LEGACY_LD_DIR

/*!< ADV_DIRECT_IND (high duty cycle) */
#define ESP_BLE_GAP_SET_EXT_ADV_PROP_LEGACY_HD_DIR

/*!< ADV_SCAN_IND */
#define ESP_BLE_GAP_SET_EXT_ADV_PROP_LEGACY_SCAN

/*!< ADV_NONCONN_IND */
#define ESP_BLE_GAP_SET_EXT_ADV_PROP_LEGACY_NONCONN
```

使用上述 legacy 类型广播参数，即可让 BLE 5.0 设备发出 **BLE 4.2 兼容格式的广播包**，从而支持 BLE 4.2 客户端发现。

### 总结：

* 您可以使用 `ble50_security_server` 示例并启用 BLE 5.0 特性；
* 同时通过设置兼容的广播参数，**支持 BLE 4.2 客户端发现、连接、加密**；
* BLE 5.0 客户端还能使用新特性（如 2M PHY、Extended Adv、LE Secure Connections 等）。

如您在实际操作中遇到具体配置问题，欢迎继续反馈，我们会协助您进一步调试。

esp-zhp, 2025-08-07:
BLE 5.0 接口 = 支持 BLE 5.0 新特性 + 兼容 BLE 4.2 功能
如果使用了 BLE5.0 ，无需再使用 BLE 4.2 接口,如  esp_ble_gap_start_advertising()