espressif/esp-idf #17248: PMS configuration doesn't work as expected

HDLA-BG, 2025-08-06:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
We are using ESP32S3 with ESP-IDF 5.3.3

We are trying to activate Permission Control for protecting specific areas of the External Flash for unexpected writes.
According esp32-s3_technical_reference_manual v1.2 
PMS supports 4 region descriptors for Flash protection and different permissions can be configured.
Our idea is to have almost whole flash protected and dynamically unprotect different areas 
only when it is needed(normally during configuration or saving internal logs). 
We don't need to use World controller because it'll add too much complexity, which we would like to avoid.

I did several tests configuring access writes for one of the areas. 
Here is how the registers are configured:
```
CONFIG 0
read APB_CTRL_FLASH_ACE0_ATTR_REG(0x60026028): 0x000000db
read APB_CTRL_FLASH_ACE0_ADDR_REG(0x60026038): 0x00000000
read APB_CTRL_FLASH_ACE0_SIZE_REG(0x60026048): 0x00000001

CONFIG 1
read APB_CTRL_FLASH_ACE1_ATTR_REG(0x6002602C): 0x000000db
read APB_CTRL_FLASH_ACE1_ADDR_REG(0x6002603C): 0x00010000
read APB_CTRL_FLASH_ACE1_SIZE_REG(0x6002604C): 0x00000030

CONFIG 2
read APB_CTRL_FLASH_ACE2_ATTR_REG(0x60026030): 0x000000db
read APB_CTRL_FLASH_ACE2_ADDR_REG(0x60026040): 0x00310000
read APB_CTRL_FLASH_ACE2_SIZE_REG(0x60026050): 0x00000040

CONFIG 3
read APB_CTRL_FLASH_ACE3_ATTR_REG(0x60026034): 0x000000XX
read APB_CTRL_FLASH_ACE3_ADDR_REG(0x60026044): 0x00710000
read APB_CTRL_FLASH_ACE3_SIZE_REG(0x60026054): 0x0000008e
```

Regions from 0 to 2 are not touched.
I change only Attribute register APB_CTRL_FLASH_ACE3_ATTR_REG.
After changing the register I get these results for the different APB_CTRL_FLASH_ACE3_ATTR_REG values:
```
        Read Allowed      Write Allowed
0x0         NO                NO
0x40        YES               YES
0xC0        YES               YES
0x80        NO                NO
```

So I can enable/disable write and read at the same time but can't disable only write and leave read allowed.
Is it an normal behavior and is my UseCase supported?

Thank you for your support in advance!

georgi-pachov, 2025-08-12:
I am facing the same issue.

diplfranzhoepfinger, 2025-09-03:
@igrr can these guys be helped?

pacucha42, 2025-09-07:
I'll do, sorry for the delay. This issue unintentionally stayed buried under other tasks. Need one day or such, as the Flash is not directly supported by the IDF Memory Protection sw module

laukik-hase, 2025-09-09:
Hello, @HDLA-BG!

Sorry for the delay and thank you for the issue report. I was investigating this issue but unfortunately could not reproduce it. Here are my [test application](https://github.com/user-attachments/files/22234006/pms_flash_debug.zip) and logs for reference, please take a look.

[image]

laukik-hase, 2025-10-16:
Hello, @HDLA-BG!

As per the TRM, the starting address as well as the size of a configured region should be a multiple of 64KB. Could you try updating your partition table and verifying the behaviour? Thanks!

[image]

ivpeykov, 2025-12-05:
Hi, I have some updates on this matter.

The first obvious fix was the change in the address of the protected partition - it indeed was not aligned properly.

Second - We were using AREA_0’s address not set at 0x0 hence got the whole PMS confused where it is it’s base (as of my understanding) and this was the reason of the WDT reset. When we used Area_1 and split line 1 it worked without crashing and allowed me to write to flash when protection was disabled and called the callback and then abort when the protection was enabled, which is great news.

Third - for dynamic enabling and disabling of the protection, I found out that it is necessary to set SYSCON_SPI_MEM_REJECT_CLR bit in the SYSCON_SPI_MEM_PMS_CTRL_REG register to clear interrupts before re-enabling. Otherwise when the protection is re-enabled it immediately calls the interrupt handler for access violation on spi1 :

static void hal_pms_flash_reset_intr_cfg(void)
{
  // Clear any interupts that may be leftover by setting SPI_MEM_PMS_REJECT_INT_CLR
  REG_SET_FIELD(SYSCON_SPI_MEM_PMS_CTRL_REG, SYSCON_SPI_MEM_REJECT_CLR, 1);

  permc_core_t core = (permc_core_t)esp_cpu_get_core_id();

  permc_ll_flash_icache_clear_int(core);
  permc_ll_flash_dcache_clear_int(core);

  esp_intr_enable(pms_flash_isr_cache_handle);
  esp_intr_enable(pms_flash_isr_spi1_handle);

  permc_ll_flash_icache_enable_int(core);
  permc_ll_flash_dcache_enable_int(core);
} Can you varify if this is the correct way to handle this?

Now here is the interesting part. This example that @laukik-hase provided, I was not able to run without disabling the Memory protection option in the menuconfig. Can you provide more information what exactly it does as I was unable to find detailed information about it, except it configures the IRAM and DRAM. I also tried it with the option for locking the configuration on and off, but the result was the same. Here is what happens:

Std_ReturnType IRAM_ATTR HAL_PMS_ext_flash_protect_enable(void)
{
    Std_ReturnType result = E_OK;

    uint32 line0_addr = 0x0;
    uint32 line1_addr = 0x10000;

    uint32 regn0_size = line1_addr / SOC_MMU_PAGE_SIZE;

    cache_hal_suspend(CACHE_LL_LEVEL_EXT_MEM, CACHE_TYPE_ALL);
    permc_ll_flash_cache_set_split_line(PERMC_SPLIT_LINE_0, line0_addr, regn0_size);
    permc_ll_flash_cache_set_perm(PERMC_AREA_0, PERMC_WORLD_0, PERMC_ACCESS_R);
    permc_ll_flash_cache_set_perm(PERMC_AREA_0, PERMC_WORLD_1, PERMC_ACCESS_R);
    permc_ll_flash_spi1_set_perm(PERMC_AREA_0, PERMC_ACCESS_R);

    cache_hal_resume(CACHE_LL_LEVEL_EXT_MEM, CACHE_TYPE_ALL);

    hal_pms_flash_reset_intr_cfg();

    return result;
}

This is with 
CONFIG_ESP_SYSTEM_MEMPROT_FEATURE=y
# CONFIG_ESP_SYSTEM_MEMPROT_FEATURE_LOCK is not set

when I call the function for enable, I immidately crash with:

Guru Meditation Error: Core  / panic'ed (Cache disabled but cached memory region accessed).

Core  0 register dump:
PC      : 0x42006f7f  PS      : 0x00060a34  A0      : 0x82006fe8  A1      : 0x3fcc1fb0
--- 0x42006f7f: esp_intr_alloc_intrstatus at C:/Espressif/frameworks/esp-idf-v5.3.3/components/esp_hw_support/intr_alloc.c:667

A2      : 0x00000014  A3      : 0x0000040e  A4      : 0x00000000  A5      : 0x3fcdfe54
A6      : 0x00000008  A7      : 0x0000005e  A8      : 0x3fcdfe44  A9      : 0x40380a28
--- 0x40380a28: esp_coex_common_env_is_chip_wrapper at ??:?

A10     : 0x3fca1538  A11     : 0x600c2178  A12     : 0x00000014  A13     : 0x000c1320
A14     : 0x000c1320  A15     : 0x00000000  SAR     : 0x0000000c  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff
--- 0x400570e8: memset in ROM
--- 0x400570f3: memset in ROM
[... 26 lines cut ...]
I tried not calling HAL_PMS_init_intr_cfg at all, as my guess is that when the memory protection is enabled in the menuconfig, it already has allocated the interrupt and when I try to do it again I crash it. 

Well the result was the following:

Enabling FLASH Protection
Disabling FLASH Protection
Executing FLASH read and write test with protection disabled
before erase: 0xFFFFFFFF
after erase : 0xFFFFFFFF
after write: 0xBABACECA
Enabling FLASH Protection
Executing FLASH read and write test with protection enabled
before erase: 0xBABACECA
after erase : 0xFFFFFFFF
after write: 0xFFFFFFF

and I don’t get abort, well of course I haven’t configured the intr handler. So the question is, is there a way to access this handler, or should I just turn off the option in the menuconfig and configure the permissions for the flash only. Do I lose some other configurations this way?

Next important point, in all cases so far, esp_partition_erase_range does not trigger my callback handler to abort. It does not trigger the PMS monitor. Which is not ideal for us because we want to protect our flash from erasing. Writing to the range does trigger it on the other hand (esp_partition_write). But for me this is not enough because once the “protected” flash is erased, I lose my data and it’s as if it has not been protected at all.

- Also, is it needed to set all 3 split lines and explicitly set permissions for the whole flash or is it fine to set only for example AREA_1 from 0x10000 until 0x20000? How will this affect the other partitions?

- And lastly, in my experience calling cache_hal_suspend before configuring will disable the resource for the other core and if it tries to read/write in this time frame it will crash. When I skipped this cache_hal_suspend and cache_hal_resume the problem went away. Can you confirm that it is not a problem to handle this case?

Thank you in advance, I hope my findings/questions help anyone else coming along this.

ivpeykov, 2026-01-08:
Hi @laukik-hase,

yes that is indeed strange. I am using ESP-IDF 5.3.3. 

I am also using sdkconfig generated from the sdkconfig.defaults that you sent me, adding set-target esp32s3. 
(Effectively I'm using idf.py set-target esp32s3 build flash monitor) and configuring the serial flasher config according to the flash size. These are all the changes I did to the sdkconfig.

Attaching the whole sdkconfig here 

[sdkconfig.txt](https://github.com/user-attachments/files/24492290/sdkconfig.txt)

I tried the example on different devkits and memory configurations - 

all were ESP32S3-WROOM-1
- 16R2
- 8R8

I made sure the serial flasher config was setup correctly for the boards I tested.
Results were the same as before, erase flash didn't fire a violation.

Here are the logs upon boot:

[image]

closed 2026-09-16 by mahavirj