espressif/esp-idf #17251: Method to run post-build commands on the `.elf`, before `.bin` generation (regression)

noahp, 2025-08-06:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v6.0-dev-1489-g4e036983a7

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The change in this commit breaks some of our build tooling:
https://github.com/espressif/esp-idf/commit/7c75795a0be199e69d1d9c780077c64343be0cb4

Specifically, we were relying on the `gen_project_binary`/`gen_signed_project_binary` targets being exposed to our component, so we could insert a custom command (via a custom target) in between the elf and image generation.

I.e. we want to run a script that processes the elf _before_ the image generation is performed. You can find the snippet here:
https://github.com/memfault/memfault-firmware-sdk/blob/60e9ab93eb70d237a61b7f5a1d1572118f7342d7/ports/esp_idf/memfault/CMakeLists.txt#L377-L401

### What is the actual behavior?

We'd like to have a way to add commands that modify the elf, that always run before the image gen steps

### Steps to reproduce.

You can reproduce this by trying to build this sample app, while on esp-idf master:

https://github.com/memfault/memfault-firmware-sdk/tree/60e9ab93eb70d237a61b7f5a1d1572118f7342d7/examples/esp32/apps/memfault_demo_app

1. `git clone https://github.com/memfault/memfault-firmware-sdk.git`
2. `cd memfault-firmware-sdk/examples/esp32/apps/memfault_demo_app`
3. set a dummy project key: `echo "CONFIG_MEMFAULT_PROJECT_KEY=\"dummy\"\n" >> sdkconfig.defaults`
4. `idf.py build`

### Build or installation Logs.

```plain

```

sudeep-mohanty, 2025-08-08:
Hi @noahp,
While investigating this problem, I realized that perhaps the `memfault_build_id` may not be working as expected, and it would be great if you could clarify the below assumption -

Even prior to the refactoring, the [custom command](https://github.com/espressif/esp-idf/blob/b66b5448e0dfac6be135e6f0ac21010aba231180/components/esptool_py/project_include.cmake#L132) which invokes the `elf2image` tool is invoked when the `elf` is created. The same is true for the `memfault_build_id` target. Hence, IOW, both actions could occur concurrently and there could be a chance that the elf2image kicks in before memfault has had a chance to update the elf. Is that understanding correct? If so, then even if ESP-IDF creates the `gen_project_binary` target early for `memfault` to add a dependency to it, it may fix the `memfault` CMake build but the functionality may not work reliably.

```mermaid
%%{init: {'theme':'base','flowchart':{'diagramPadding':4,'nodeSpacing':18,'rankSpacing':18}, 'themeVariables':{'primaryColor':'#fff','primaryTextColor':'#000','primaryBorderColor':'#000','lineColor':'#000','secondaryColor':'#fff','tertiaryColor':'#fff'}}}%%
flowchart LR
  ELF[ELF file exists]
  MFLT[memfault_build_id]
  E2I[elf2image]
  UMB[gen_project_binary]

  ELF --> MFLT
  ELF --> E2I
  MFLT --> UMB
  E2I  -->UMB
```

That said, I think what we may need is a way for `memfault` to add a pre-binary generation dependency.

noahp, 2025-09-02:
Hi @sudeep-mohanty , did you have an idea of how to add the pre-binary generation dependency?

sudeep-mohanty, 2025-09-03:
Hi @noahp,
Apologies for the delayed update. Yes, we will be adding a new build system CMake API for users to declare dependencies which should run post ELF generation and before the binary generation happens. The semantics would be like so -

```cmake
idf_build_add_post_elf_dependency("${CMAKE_PROJECT_NAME}.elf" my_postelf)

# my_post elf should be the custom target to run after the ELF is generated
```

Attached is the complete git diff of the new feature with its documentation and some test examples. The changes are in our internal repository but will appear in the public master branch once it passes internal CI. Please have a try and let us know if it helps to solve your issue. Thank you.

[post-elf-hook.patch](https://github.com/user-attachments/files/22110306/post-elf-hook.patch)

noahp, 2025-09-03:
@sudeep-mohanty thank you so much, this works perfectly! I added a target following the instructions, and everything looks good 👍 

Much appreciated!

closed 2025-09-17 by espressif-bot