espressif/esp-idf #17252: idf_component_optional_requires incompatible with idf_build_set_property(MINIMAL_BUILD ON)

greenaddress, 2025-08-06:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expected idf_component_optional_requires to work together with idf_build_set_property(MINIMAL_BUILD ON)

### What is the actual behavior?

Instead they work independently but not together

### Steps to reproduce.

Make a project that conditionally depends on a component via Kconfig option, builds fine, but if you add idf_build_set_property(MINIMAL_BUILD ON) to the root CMakeLists.txt then it breaks.

### Build or installation Logs.

```plain

```

ESP-Marius, 2025-08-07:
I think it works as expected. `idf_component_optional_requires` adds an requirement if the component is already in the build. It cannot be used to pull a component optionally into the build.

With the current build-system it is not possible to let a kconfig option decide if a component is included in the build or not, as which components are in the build is decided in the initial parse, at which point kconfig optoins are not known.

This will be fixed/improved with the current upgrade/makeover of the build-system we are working on for IDF 6.0.  

CC @fhrbata

fhrbata, 2025-08-07:
Hello @greenaddress ,

> idf_component_optional_requires works conditionally with Kconfig as far as I can tell as long as you don't set idf_build_set_property(MINIMAL_BUILD ON) in the root CMakeLists.txt file but I reckon it should work with MINIMAL_BUILD too.

Yes, but the optional component has to be added to the build.

Here's some background on why this is happening.

The build system operates as follows:

1. It searches for available components in specified [directories](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html#optional-project-variables).
2. It creates a list of components to be added to the build, stored internally in the `BUILD_COMPONENTS` variable. This list includes components specified in the `COMPONENTS` variable and their transitive dependencies, forming a subset of the components discovered in step 1. If the `COMPONENTS` variable is not set, all components from step 1 are added to `BUILD_COMPONENTS`. The minimal build feature is essentially a shorthand for `set(COMPONENTS main)`, which means only the `main` component and its transitive dependencies are added to the build (`BUILD_COMPONENTS`). Component dependencies are also resolved at this stage using the cmake script mode.
3. It generates the sdkconfig for components in `BUILD_COMPONENTS`. The component dependencies are already resolved in step 2, where the sdkconfig is not available, so the component dependencies cannot rely on the sdkconfig when the `idf_component_register()` is called.
4. It calls `add_subdirectory` to evaluate the components from `BUILD_COMPONENTS`.

The [idf_component_optional_requires](https://github.com/espressif/esp-idf/blob/master/tools/cmake/component.cmake#L602) function checks the `BUILD_COMPONENTS` list and adds the specified component as a dependency only if it is present. With minimal build enabled the list of components in `BUILD_COMPONENTS` is reduced.

From the [idf_component_optional_requires description](https://github.com/espressif/esp-idf/blob/master/tools/cmake/component.cmake#L602)

> Calling idf_component_optional_requires(PRIVATE dependency_name) has the similar effect to target_link_libraries(${COMPONENT_LIB} PRIVATE idf::dependency_name), only if 'dependency_name'  component is part of the build.

The typical ` idf_component_optional_requires` usage is

```cmake
idf_component_register(...)

if(CONFIG_XXX)
     idf_component_optional_requires(PRIVATE OPTIONAL_COMPONENT)
endif()
```

This is evaluated in step 4, the sdkconfig is available, but the `BUILD_COMPONENTS` list is already set and cannot be modified. This means that if `OPTIONAL_COMPONENT` from the example above is not included in `BUILD_COMPONENTS`, it will not be added. Additionally, configuration options from `OPTIONAL_COMPONENT` will not be accessible. When minimal build is enabled, `OPTIONAL_COMPONENT` must be included in the build as an explicit dependency of another component otherwise is't not visible to `idf_component_optional_requires`. 

You highlighted, "Build & Config: Added support for conditional dependencies in idf_component.yml based on Kconfig option values (b43b200)", which is an excellent observation. This change essentially restarts almost the entire CMake configuration if the component manager lacks a configuration option used in the component manifest file. When the configuration is restarted, one would expect the sdkconfig to be available in step 2. Unfortunately, it isn't, because component dependencies are evaluated in [CMake's script mode](https://github.com/espressif/esp-idf/blob/master/tools/cmake/scripts/component_get_requirements.cmake)), and sdkconfig is not passed there. Moreover, relying on the component manager to trigger a re-run of the build process is error-prone and, in most cases, will not occur. Even if it did work, the components in the `BUILD_COMPONENTS` list would still be limited by the `COMPONENTS` variable set by the minimal build.

As @ESP-Marius  has already mentioned, with the current build system, it is not possible for a Kconfig option to determine whether a component is included in the build. We are aware of the current limitations of the build system, and work is in progress to make improvements.

Thank you

greenaddress, 2025-08-07:
thanks @fhrbata 

To understand better, does it ever make sense to use a Kconfig idf_component_optional_requires and MINIMAL_BUILD at the same time? in other words, does it skip compiling them if idf_component_optional_requires even if they are explicitly listed under PRIV_REQUIRES?

thanks!

closed 2025-08-22 by dobairoland