espressif/esp-idf #17256: NVS Encryption not properly including nvs_sec_provider component

sidwarkd, 2025-08-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expect the build system to automatically include the nvs_sec_provider component when NVS Encryption is enabled in menuconfig when I am using the minimal build configuration or `set(COMPONENTS main)`.

### What is the actual behavior?

The proper Kconfig settings for `nvs_sec_provider` are not included and therefore nvs encryption is not properly configured at runtime so calls to `nvs_flash_read_security_cfg_v2` and `nvs_flash_generate_keys_v2` fail when the application calls `nvs_flash_init`.

This is because the functions on `nvs_sec_default_scheme_cfg` are not properly initialized. 

### Steps to reproduce.

1. Start with a standard, simple project that calls `nvs_flash_init`
2. Modify the root CMakeLists.txt to have `set(COMPONENTS main)`
3. Enable Flash Encryption and NVS encryption in menuconfig.
4. Build with `idf.py build`

The project will build just fine. However, when you run the application you will get an error "Failed to generate NVS encr-keys"

Modify the `set` line in CMakeLists.txt to be `set(COMPONENTS main nvs_sec_provider)` and rebuild/reflash. The application will run fine. 

I would expect, if flash and nvs encryption are enabled in menuconfig, the `nvs_sec_provider` module would be included automatically by the build system.

### Build or installation Logs.

```plain
W (449) flash_encrypt: Flash encryption mode is DEVELOPMENT (not secure)
I (458) main_task: Started on CPU0
I (468) main_task: Calling app_main()
W (468) nvs: Failed to read NVS security cfg: [0x102] (ESP_ERR_INVALID_ARG)
I (468) nvs: Generating NVS encr-keys...
E (468) nvs: Failed to generate NVS encr-keys: [0x102] (ESP_ERR_INVALID_ARG)
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x400d7842
```

sidwarkd, 2025-08-07:
@RathiSonika Thanks for the suggestion. Yes, could use REQUIRES as well but I definitely think that shouldn't be necessary from a user perspective. But if it does end up needing to be that way it should be documented in the NVS Encryption docs. The additional issue is that CONFIG_NVS_ENCRYPTION is automatically set when you enable CONFIG_SECURE_FLASH_ENC_ENABLED so it would be non-obvious, as a user, that enabling flash encryption would, in turn, require me to add an additional REQUIRES entry for everything to work.

Thanks again for the quick response and I look forward to hearing what you find.

Lucasmm016, 2025-11-01:
I just ran into this problem: when I enabled flash encryption, nvs encryption was automatically activated, which I expected. However, when I tried to run the project normally with some components installed (including Arduino), I started getting the error mentioned where the key generation failed. I tried forcing a new key generation, but the error persisted. After tireless searching, I found this post, which saved me. Basically, I needed to add nvs_sec_provider to the REQUIRES of the CMakeLists file, but this was literally not obvious at all. I'm still wondering if it's correct or not. I also didn't find anything in the official documentation about it.

KaeLL, 2025-11-01:
https://www.youtube.com/watch?v=7utLBxSOXlQ&t=2021s