espressif/esp-idf #17258: test qemu with flash encryption&secure boot fail

Dandjinh, 2025-08-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have wrote a bash script to test flash encryption & secure boot v2 with qemu-esp32s3.

```bash
#!/bin/bash

set -e

SCRIPT_DIR=$(dirname "$(readlink -f "$0")")

EFUSE_FILE_PATH="$SCRIPT_DIR/efuse.bin"
AES_KEY_PATH="$SCRIPT_DIR/aes-key.bin"
RSA_KEY_PATH="$SCRIPT_DIR/rsa-key.pem"
DIGEST_PATH="$SCRIPT_DIR/digest.bin"
FLASH_PATH="$SCRIPT_DIR/flash.bin"

FW_PACK_PATH=build/pack
FW_FILES=(
    "0x0:bootloader.bin"
    "0x10000:partition-table.bin"
    "0x13000:ota_data_initial.bin"
    "0x20000:secure_test.bin"
)

enable_flash_encryption()
{
    # generate an aes key if not exist
    if [[ ! -f $AES_KEY_PATH ]]; then
        espsecure.py generate_flash_encryption_key \
            --keylen 512 $AES_KEY_PATH
    fi

    # burn the aes key into efuse
    idf.py qemu --efuse-file $EFUSE_FILE_PATH \
        efuse-burn-key BLOCK_KEY1 $AES_KEY_PATH XTS_AES_256_KEY
    
    # enable flash encryption
    idf.py qemu --efuse-file $EFUSE_FILE_PATH \
        efuse-burn SPI_BOOT_CRYPT_CNT 1
}

encrypt_files()
{
    # create an empty file and fill it with 0xFF
    dd if=/dev/zero bs=1M count=4 | tr '\000' '\377' > "$FLASH_PATH"

    for item in "${FW_FILES[@]}"; do
        echo "encrypt $item"
        IFS=':' read -r addr file <<< "$item"

        local src_file="$FW_PACK_PATH/$file"
        local encrypted_file="$SCRIPT_DIR/tmp/$file"

        echo "Encrypting $src_file (addr: $addr)"
        
        # encrypt file
        espsecure.py encrypt_flash_data \
            --aes_xts --keyfile "$AES_KEY_PATH" \
            --address "$addr" --output "$encrypted_file" \
            "$src_file"
        
        # copy encrypted file into flash file
        dd if="$encrypted_file" of="$FLASH_PATH" \
           bs=1 seek=$((addr)) conv=notrunc
    done
}

enable_secure_boot_v2()
{
    # create rsa key if not exist
    if [[ ! -f "$DIGEST_PATH" ]]; then
        espsecure.py generate_signing_key --version 2 --scheme rsa3072 $RSA_KEY_PATH
        espsecure.py digest_sbv2_public_key --keyfile $RSA_KEY_PATH --output $DIGEST_PATH
    fi

    # burn the digest data into efuse
    idf.py qemu --efuse-file $EFUSE_FILE_PATH \
        efuse-burn-key BLOCK_KEY3 $DIGEST_PATH SECURE_BOOT_DIGEST0

    # enable secure boot
    idf.py qemu --efuse-file $EFUSE_FILE_PATH \
        efuse-burn SECURE_BOOT_EN
}

burn_efuse()
{
    local values=(
        DIS_DOWNLOAD_ICACHE
        DIS_DOWNLOAD_DCACHE
        HARD_DIS_JTAG
        SOFT_DIS_JTAG
        DIS_DIRECT_BOOT
        DIS_USB_JTAG
        DIS_DOWNLOAD_MANUAL_ENCRYPT
    )

    for val in "${values[@]}"; do
        idf.py qemu --efuse-file $EFUSE_FILE_PATH \
            efuse-burn $val 0x1
    done
}

start()
{
    # start qemu with specific efuse and flash file
    idf.py qemu \
        --efuse-file $EFUSE_FILE_PATH \
        --flash-file $FLASH_PATH \
        monitor
}

# clear efuse file
if [[ -f $EFUSE_FILE_PATH ]]; then
    rm $EFUSE_FILE_PATH
fi

enable_flash_encryption
encrypt_files
enable_secure_boot_v2
burn_efuse
start
```

But it ends up with `invalid header`.

the whole project is attached below.
[secure_test.zip](https://github.com/user-attachments/files/21664371/secure_test.zip)

Just runs
```bash
idf.py build
./qemu/run.sh
```

> sdk version is release/v5.4

I have tried few days and don't have any ideas now.
Could someone give me some advices. I really appreciate that.

Dandjinh, 2025-08-08:
Interesting! I changed to AES128 and it boots successfully!

New script below:
```bash
#!/bin/bash

set -e

SCRIPT_DIR=$(dirname "$(readlink -f "$0")")

EFUSE_FILE_PATH="$SCRIPT_DIR/efuse.bin"
AES_KEY_PATH="$SCRIPT_DIR/aes-key.bin"
RSA_KEY_PATH="$SCRIPT_DIR/rsa-key.pem"
DIGEST_PATH="$SCRIPT_DIR/digest.bin"
FLASH_PATH="$SCRIPT_DIR/flash.bin"

FW_PACK_PATH=build/pack
FW_FILES=(
    "0x0:bootloader.bin"
    "0x10000:partition-table.bin"
    "0x13000:ota_data_initial.bin"
    "0x20000:secure_test.bin"
)

enable_flash_encryption()
{
    local use_large_key=0
    local key_size=0

    # rm key if size not match
    if [[ -f $AES_KEY_PATH ]]; then
        key_size=$(ls -all $AES_KEY_PATH | awk '{print $5;}')
        if [[ ($use_large_key -eq 1) && ($key_size -ne 64) ]]; then
            rm $AES_KEY_PATH
        elif [[ ($use_large_key -eq 0) && ($key_size -ne 32) ]]; then
            rm $AES_KEY_PATH
        fi
    fi

    # generate an aes key if not exist
    if [[ ! -f $AES_KEY_PATH ]]; then
        if [[ $use_large_key -eq 1 ]]; then
            espsecure.py generate_flash_encryption_key \
                --keylen 512 $AES_KEY_PATH
        else
            espsecure.py generate_flash_encryption_key \
                $AES_KEY_PATH
        fi
    fi

    # burn the aes key into efuse
    if [[ $use_large_key -eq 1 ]]; then
        idf.py qemu --efuse-file $EFUSE_FILE_PATH \
            efuse-burn-key BLOCK_KEY1 $AES_KEY_PATH XTS_AES_256_KEY
    else
        idf.py qemu --efuse-file $EFUSE_FILE_PATH \
            efuse-burn-key BLOCK_KEY1 $AES_KEY_PATH XTS_AES_128_KEY
    fi

    # enable flash encryption
    idf.py qemu --efuse-file $EFUSE_FILE_PATH \
        efuse-burn SPI_BOOT_CRYPT_CNT 1
}

encrypt_files()
{
    # create an empty file and fill it with 0xFF
    dd if=/dev/zero bs=1M count=4 | tr '\000' '\377' > "$FLASH_PATH"

    for item in "${FW_FILES[@]}"; do
        echo "encrypt $item"
        IFS=':' read -r addr file <<< "$item"

        local src_file="$FW_PACK_PATH/$file"
        local encrypted_file="$SCRIPT_DIR/tmp/$file"

        echo "Encrypting $src_file (addr: $addr)"

        # encrypt file
        espsecure.py encrypt_flash_data \
            --aes_xts --keyfile "$AES_KEY_PATH" \
            --address "$addr" --output "$encrypted_file" \
            "$src_file"

        # copy encrypted file into flash file
        dd if="$encrypted_file" of="$FLASH_PATH" \
           bs=1 seek=$((addr)) conv=notrunc
    done
}

enable_secure_boot_v2()
{
    # create rsa key if not exist
    if [[ ! -f "$DIGEST_PATH" ]]; then
        espsecure.py generate_signing_key --version 2 --scheme rsa3072 $RSA_KEY_PATH
        espsecure.py digest_sbv2_public_key --keyfile $RSA_KEY_PATH --output $DIGEST_PATH
    fi

    # burn the digest data into efuse
    idf.py qemu --efuse-file $EFUSE_FILE_PATH \
        efuse-burn-key BLOCK_KEY3 $DIGEST_PATH SECURE_BOOT_DIGEST0

    # enable secure boot
    idf.py qemu --efuse-file $EFUSE_FILE_PATH \
        efuse-burn SECURE_BOOT_EN
}

burn_efuse()
{
    local values=(
        DIS_DOWNLOAD_ICACHE
        DIS_DOWNLOAD_DCACHE
        HARD_DIS_JTAG
        SOFT_DIS_JTAG
        DIS_DIRECT_BOOT
        DIS_USB_JTAG
        DIS_DOWNLOAD_MANUAL_ENCRYPT
    )

    for val in "${values[@]}"; do
        idf.py qemu --efuse-file $EFUSE_FILE_PATH \
            efuse-burn $val 0x1
    done
}

start()
{
    # start qemu with specific efuse and flash file
    idf.py qemu \
        --efuse-file $EFUSE_FILE_PATH \
        --flash-file $FLASH_PATH \
        monitor
}

show_efuse()
{
    idf.py qemu \
        --efuse-file $EFUSE_FILE_PATH \
        --flash-file $FLASH_PATH \
        efuse-summary
}

if [[ -f $EFUSE_FILE_PATH ]]; then
    rm $EFUSE_FILE_PATH
fi

enable_flash_encryption
encrypt_files
enable_secure_boot_v2
burn_efuse
start
```

Although, it keeps rebooting
```
[0;32mI (343) cpu_start: Multicore app[0m
[0;32mI (357) cpu_start: Pro cpu start user code[0m
[0;32mI (357) cpu_start: cpu freq: 160000000 Hz[0m
[0;32mI (357) app_init: Application information:[0m
[0;32mI (357) app_init: Project name:     secure_test[0m
[0;32mI (357) app_init: App version:      be0503f-dirty[0m
[0;32mI (357) app_init: Compile time:     Aug  8 2025 11:50:47[0m
[0;32mI (357) app_init: ELF file SHA256:  f73d31e60...[0m
[0;32mI (357) app_init: ESP-IDF:          v5.4.1-855-g46909c5048-dirty[0m
[0;32mI (357) efuse_init: Min chip rev:     v0.0[0m
[0;32mI (357) efuse_init: Max chip rev:     v0.99 [0m
[0;32mI (357) efuse_init: Chip rev:         v0.3[0m
[0;32mI (358) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (358) heap_init: At 3FC959B8 len 00053D58 (335 KiB): RAM[0m
[0;32mI (358) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM[0m
[0;32mI (358) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM[0m
[0;32mI (358) heap_init: At 600FE01C len 00001FCC (7 KiB): RTCRAM[0m
[0;32mI (363) spi_flash: detected chip: gd[0m
[0;32mI (363) spi_flash: flash io: qio[0m

assert failed: 0x40380341 <cached disabled>:95 (<cached disabled>)

Backtrace: 0x40375a6d:0x3fceb0e0 0x40379cd9:0x3fceb100 0x403803d5:0x3fceb120 0x420026a2:0x3fceb240 0x42003212:0x3fceb260 0x4200325d:0x3fceb280 0x420032c7:0x3fceb2a0 0x4037560e:0x3fceb2c0 0x403cd5fd:0x3fceb330 0x403cd783:0x3fceb380 0x403c8a1d:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
```
Looks like change flash to qio triggers assertion.

[details.log](https://github.com/user-attachments/files/21676639/run.log)

o-marshmallow, 2025-08-08:
Hello @Dandjinh , thanks for providing a complete archive to reproduce the issue.

The current version of QEMU provided with ESP-IDF is not up to date, I highly advise you to compile Espressif QEMU from source and have a try with it: https://github.com/espressif/qemu

Dandjinh, 2025-08-08:
Hello @o-marshmallow,
I tested with the latest release version of [qemu](https://github.com/espressif/qemu) and it all failed.

Test Result:
| qemu version | build/128 & efuse/128 | build/128 & efuse/256 | build/256 & efuse/256 | build/256 & efuse/128 |
| :--: | :--: | :--: | :--: | :--: |
| esp_develop_9.0.0_20240606 | ok | fail | fail | ok |
| esp_develop_9.2.2_20250228 | fail | fail | fail | fail |

> build switch at
[image]

> efuse switch at
[image]

> whole project files
[secure_test2.zip](https://github.com/user-attachments/files/21677996/secure_test2.zip)

Harshal5, 2025-08-14:
Hello @Dandjinh,

Thanks for trying out the QEMU emulation!

I think you are mentioning two issues here:
1. The incorrect header logs on boot
2. And the assertion when flash QIO mode is enabled

The first issue can be resolved by using the latest QEMU release (v9.2.2). The older QEMU release v9.0.0 did not contain support for Flash Encryption for ESP32-S3.

Note: Your script seems to use `idf.py qemu` command to invoke a QEMU instance. As ESP-IDF currently supports only v9.0.0 (the older version), running idf.py qemu would result in creation of v9.0.0 QEMU instance. To use the latest v9.2.2 QEMU instance, you should run the following commands ([reference](https://github.com/espressif/esp-toolchain-docs/tree/main/qemu/esp32s3)) with the latest QEMU release artifacts:

1. Download mode command (needed to burn the efuses)
```
./qemu-system-xtensa -nographic \
                        -M esp32s3 \
                        -drive file= ~ /Downloads/secure_test2/qemu/flash.bin,if=mtd,format=raw \                
                        -global driver=esp32s3.gpio,property=strap_mode,value=0x0 \                                   
                        -drive file=~ /Downloads/secure_test2/qemu/efuse.bin,if=none,format=raw,id=efuse \
                        -global driver=nvram.esp32s3.efuse,property=drive,value=efuse \    
                        -serial tcp::5555,server,nowait
```
2. Boot mode command
```
 ./qemu-system-xtensa -nographic \                                                                                                             
                        -M esp32s3 \
                        -drive file=~ /Downloads/secure_test2/build/encrypted_qemu_image.bin,if=mtd,format=raw \
                        -drive file=~ /Downloads/secure_test2/qemu/efuse.bin,if=none,format=raw,id=efuse \
                        -global driver=nvram.esp32s3.efuse,property=drive,value=efuse \
                        -global driver=timer.esp32s3.timg,property=wdt_disable,value=true \
                        -nic user,model=open_eth -serial mon:stdio
```

<hr />

Regarding the 2nd issue of hitting an assertion when flash QIO is enabled, it is because as of now we do not support the QIO mode in QEMU.

I tried running your project (secure_test2.zip) by changing the QIO config to DIO using the latest v9.2.2 QEMU artifacts, and then I am able to run the application successfully no matter what flash encryption key size is used.

Thanks!

o-marshmallow, 2025-08-14:
As @Harshal5 said, the issue you see in IDF is because we only support DIO for the SPI flash. The Quad and Octal SPI are only supported for the PSRAM in the ESP32-S3 QEMU target.

Dandjinh, 2025-08-14:
Hello @Harshal5 ,

I changed the flash configuration to DIO and it did work using AES128.

Although I just changed the script `run.sh` and it now will auto create a soft link to target qemu under idf tools path(also what I did last time), the result is still failure.

Here is the log: [detail.log](https://github.com/user-attachments/files/21767848/detail.log), which shows the qemu-system-xtensa has been changed to target version.

```bash
./qemu/run.sh
@@@ qemu version
QEMU emulator version 9.2.2 (esp_develop_9.2.2_20250228)
Copyright (c) 2003-2024 Fabrice Bellard and the QEMU Project developers
@@@ burn aes-key, 64
```

Here is the updated test project: [qemu-esp32s3-secure-test](https://github.com/Dandjinh/qemu-esp32s3-secure-test).

just change `target_version` in `run.sh` to select target qemu executable.

```bash
select_qemu()
{
    # local target_version="9.0.0"
    local target_version="9.2.2"

    # check qemu version
    local qemu_version=$(qemu-system-xtensa --version | head -n 1 | awk '{print $4;}')
    ...
}
```

Dandjinh, 2025-08-21:
It does work after changing efuse type, thanks @Harshal5 .

closed 2025-08-21 by Dandjinh