espressif/esp-idf #17262: ESP32-C6: PARLIO RX with pulse_delimiter fails to capture data, while soft_delimiter works correctly

Dimandragon, 2025-08-08:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.3

### Espressif SoC revision.

ESP32-C6 (QFN40) (revision v0.2)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

nanoESP32-C6

### Power Supply used.

USB

### What is the expected behavior?

**Bug Description**

I am interfacing an ESP32-C6 with an AD7768 external ADC using the PARLIO peripheral. I have two firmware implementations for capturing data:

1.  **`soft_delimiter` approach:** This version treats the ADC's `DRDY` (Data Ready) signal as a 9th data line and performs a continuous dump of the bus. This method works perfectly and captures data that matches theoretical expectations. The logs confirm that the `DCLK` and `DRDY` signals are active and data is being transmitted by the ADC.

2.  **`pulse_delimiter` approach:** This version is intended to be more efficient. It uses the `DRDY` signal on the `valid_gpio_num` pin to trigger the capture of fixed-size data frames (32 bytes). However, this implementation consistently fails, resulting in buffers filled entirely with zeros.

The fact that the `soft_delimiter` method works correctly strongly suggests that the hardware setup, clock generation (MCLK, DCLK), and the ADC configuration are all correct. The issue appears to be specific to the `parlio_rx_pulse_delimiter` implementation or its configuration on the ESP32-C6.

**Hardware**

*   **MCU:** ESP32-C6
*   **External ADC:** Analog Devices AD7768
*   **ESP-IDF Version:** `release/v5.3`
*   **Operating System:** Linux Mint 22

**Expected Behavior**

When running `pulse_delimiter_fails.c`, the application should successfully receive 10 data frames, each 32 bytes long. The content of these frames should be the non-zero data from the ADC, which is transmitted following the falling edge of the `DRDY` signal.

### What is the actual behavior?

**Actual Behavior**

The application times out or completes, but all received frames consist entirely of zeros. The log output repeatedly shows the warning: `Warning! Frame #X is entirely zero.`

### Steps to reproduce.

**Steps to Reproduce**

1.  Connect an AD7768 ADC to an ESP32-C6 board using the GPIOs defined in the provided code snippets.
2.  Compile and flash the first code example (`soft_delimiter_works.c`).
3.  Observe the logs. You will see valid, non-zero data being captured, and the `DRDY` signal transitions will be visible in the 9th bit of the received words.
[... 565 lines cut ...]
W (976) ADC_DUMP_SIMPLIFIED_TEST: Warning! Frame #1 is entirely zero.
--------------------------------------------------
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
--------------------------------------------------
I (1016) ADC_DUMP_SIMPLIFIED_TEST: --- Dumping frame #2/10 (Address: 0x408158a8) ---
W (1016) ADC_DUMP_SIMPLIFIED_TEST: Warning! Frame #2 is entirely zero.
--------------------------------------------------
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
--------------------------------------------------
... (repeats for all 10 frames)
```

### More Information.

**Additional Context**

In the failing example, the configuration for `parlio_rx_pulse_delimiter` is as follows:
*   `valid_gpio_num` is set to the `DRDY` pin.
*   The data bus width is 8 bits.
*   `eof_data_len` is set to 32 bytes, which is the expected frame size.
*   `pulse_invert = 1` is set, because the data is valid after the `DRDY` signal goes from high to low.

This configuration seems correct according to the documentation, but it does not work in practice. The problem might be a bug in the PARLIO driver, the HAL, or an undocumented aspect of the configuration.

Dimandragon, 2025-08-11:
Hi @L-KAYA,

To further investigate and completely isolate the issue from the external ADC, I have created a self-contained Minimal, Reproducible Example (MRE) on a single ESP32-C6. This test uses the PARLIO TX peripheral to generate a "perfect" and predictable signal pattern, which is then physically looped back to the PARLIO RX peripheral on the same chip. This setup eliminates any potential variables related to external hardware, signal integrity, or timing mismatches with the ADC.

### Test Methodology

The MRE firmware operates as follows:

1.  **Signal Generation (PARLIO TX):** A task continuously transmits a specially crafted signal pattern on the TX pins. This pattern consists of:
    *   A **Reference Packet** (18 bytes long):
        *   A single `DRDY` pulse (LOW -> HIGH -> LOW).
        *   A known, sequential data payload from `0x01` to `0x0F` on the 4 data lines.
    *   An **Idle Period:** A 10ms pause where `DRDY` is held low. During this idle time, the data lines can either be static (all zeros) or can be made to change at a specific rate (`IDLE_FRAME_INCR`), which proved crucial for debugging.

2.  **Signal Reception (PARLIO RX):** The RX peripheral is configured to capture this looped-back signal. The firmware can be compiled in two modes (`TEST_MODE`):
    *   **Mode 0 (`soft_delimiter`):** Acts as a "logic analyzer" to capture the raw data stream, verifying that the TX side is generating the correct signal.
    *   **Mode 1 (`pulse_delimiter`):** The mode under investigation, which uses the `DRDY` pulse on `RX_PIN_DRDY_IN` to trigger the capture of a 16-byte frame.

Here is the complete MRE firmware used for the test:

<details>
<summary><b>MRE Firmware (parlio_tx_loopback_to_rx.c)</b></summary>

```c
#include <stdio.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "driver/parlio_rx.h"
#include "driver/parlio_tx.h"
#include "esp_heap_caps.h"
#include "soc/gpio_num.h"
#include "soc/soc_caps.h"
#include "esp_check.h"
#include "hal/parlio_types.h"

static const char *TAG = "PARLIO_MRE_FINAL_V24";

#define TEST_MODE   1 // 0: Soft Delimiter, 1: Pulse Delimiter

#define IDLE_PERIOD_MS          10
#define IDLE_FRAME_INCR 1 // Set to 0 for static zero, 256 for slow change, 1 for rapid change

// --- Loopback GPIO Mapping ---
#define TX_PIN_DCLK_OUT     GPIO_NUM_19
#define TX_PIN_DRDY_OUT     GPIO_NUM_10
#define TX_PIN_D0_OUT       GPIO_NUM_6
#define TX_PIN_D1_OUT       GPIO_NUM_15
#define TX_PIN_D2_OUT       GPIO_NUM_8
#define TX_PIN_D3_OUT       GPIO_NUM_4

#define RX_PIN_DCLK_IN      GPIO_NUM_18
#define RX_PIN_DRDY_IN      GPIO_NUM_21
#define RX_PIN_D0_IN        GPIO_NUM_2
#define RX_PIN_D1_IN        GPIO_NUM_3
#define RX_PIN_D2_IN        GPIO_NUM_5
#define RX_PIN_D3_IN        GPIO_NUM_7

// --- Signal Parameters ---
#define TX_DCLK_FREQ_HZ         250000
#define RX_SAMPLE_EDGE          PARLIO_SAMPLE_EDGE_POS

#define REF_PAYLOAD_SIZE_BYTES  16

#if (TEST_MODE == 0)
#define DUMP_DURATION_MS        100
#else
#define NUM_FRAMES_TO_CAPTURE   5
#endif
[... 261 lines cut ...]
         7 |  0   |  1  1  1  1  | 0x0F
         8 |  0   |  0  0  1  0  | 0x02
         9 |  0   |  0  1  0  1  | 0x05
        10 |  0   |  1  0  0  0  | 0x08
        11 |  0   |  1  0  1  1  | 0x0B
        12 |  0   |  1  1  1  0  | 0x0E
        13 |  0   |  0  0  0  1  | 0x01
        14 |  0   |  0  1  0  0  | 0x04
        15 |  0   |  0  1  1  1  | 0x07
    --- End of Captured Frame #5 ---
    ```

### **Conclusion**

This self-contained test strongly indicates a **significant and non-deterministic timing issue** within the `parlio_rx_pulse_delimiter` implementation on the ESP32-C6.

The evidence suggests the following sequence of events:
1.  The `DRDY` pulse is correctly detected by the peripheral, as a DMA transaction of the correct length is successfully initiated and completed.
2.  However, the actual **data sampling is delayed by a large and non-deterministic amount of time (observed to be around 2000 DCLK cycles).**
3.  Because of this massive delay, the peripheral misses the intended data packet entirely and instead captures whatever data happens to be on the bus much later.
4.  Furthermore, when the input data changes on every clock cycle, the sampling mechanism appears to **break down completely**, failing to latch data correctly on each DCLK edge, resulting in a corrupted and jumbled buffer.

This behavior points away from external hardware or GPIO configuration and towards a fundamental internal logic or timing bug in how the PARLIO peripheral handles the `pulse_delimiter` mode on the ESP32-C6.

My next step is to run this exact MRE on an ESP32-P4 to determine if this behavior is specific to the C6 architecture. I will report back with the findings.

L-KAYA, 2025-08-12:
Hi @Dimandragon ~ Thank you so much for the careful and considerable test!

After reading and tested the code you provided, I found the receive data is wrongly parsed. The data that received by parlio rx is tightly packed and without carry the `valid` signal. For example the test code used 4-bit data width, which means the data you received are supposed to be `0x01, 0x23, 0x45, 0x67...`, it is different from the TX data `0x00, 0x01, 0x02, 0x03...`

I printed the data of **Case C**:

<details>
<summary> print code </summary>

```c
void print_deserialized_data2(const char* title, uint8_t* buffer, size_t size, int start_sample_num) {
    printf("\n--- %s ---\n", title);
    printf("Sample | DRDY | D3 D2 D1 D0 | Hex \n");
    printf("-------|------|-------------|------\n");
    for (int i = 0; i < size * 2; i++) {
        uint8_t sample = i & 0x01 ? buffer[i / 2] & 0x0F : buffer[i / 2] >> 4;
        int drdy = 0;  // No drdy signal will be received
        int d3 = (sample >> 3) & 1; int d2 = (sample >> 2) & 1;
        int d1 = (sample >> 1) & 1; int d0 = (sample >> 0) & 1;
        printf("%6d |  %d   |  %d  %d  %d  %d  | 0x%02X\n", start_sample_num + i, drdy, d3, d2, d1, d0, sample & 0x0F);
    }
    printf("--- End of %s ---\n", title);
}
```

</details>

[image]

The result looks relatively correctly ordered, but you should also noticed that it skipped some data as well. I have not dug out the root cause yet.

And for the **Case A**, it always received `0` indeed, and just like your observation, the receive is not timeout, which means the `ready` signal should trigger correctly. 

About the assumption that there is some delay to receive the data after the triggering of the `ready` signal, it is also reasonable to me, I need some other tests to find out how the delay introduced.

L-KAYA, 2025-08-12:
It now can receive the correct data for one transaction, however, the following transactions always read `0x0 0x1`, even keep receive `0x0 0x1` after re-enabling, it seems relate to the `eof`, as long as `eof` takes place, the data go wrong.

## Major changes

1. TX buffer should align to 4-byte on C6
2. Do not invert the pulse because the TX module produce a positive pulse.

## Details

<details>
<summary>test log</summary>

```plain text
--- Captured Frame #1 ---
Sample | DRDY | D3 D2 D1 D0 | Hex 
-------|------|-------------|------
     0 |  0   |  0  0  0  0  | 0x00
     1 |  0   |  0  0  0  1  | 0x01
     2 |  0   |  0  0  1  0  | 0x02
     3 |  0   |  0  0  1  1  | 0x03
     4 |  0   |  0  1  0  0  | 0x04
     5 |  0   |  0  1  0  1  | 0x05
     6 |  0   |  0  1  1  0  | 0x06
     7 |  0   |  0  1  1  1  | 0x07
     8 |  0   |  1  0  0  0  | 0x08
     9 |  0   |  1  0  0  1  | 0x09
    10 |  0   |  1  0  1  0  | 0x0A
    11 |  0   |  1  0  1  1  | 0x0B
    12 |  0   |  1  1  0  0  | 0x0C
    13 |  0   |  1  1  0  1  | 0x0D
    14 |  0   |  1  1  1  0  | 0x0E
    15 |  0   |  1  1  1  1  | 0x0F
--- End of Captured Frame #1 ---

--- Captured Frame #2 ---
Sample | DRDY | D3 D2 D1 D0 | Hex 
-------|------|-------------|------
     0 |  0   |  0  0  0  0  | 0x00
     1 |  0   |  0  0  0  1  | 0x01
     2 |  0   |  0  0  0  0  | 0x00
     3 |  0   |  0  0  0  1  | 0x01
     4 |  0   |  0  0  0  0  | 0x00
     5 |  0   |  0  0  0  1  | 0x01
     6 |  0   |  0  0  0  0  | 0x00
     7 |  0   |  0  0  0  1  | 0x01
     8 |  0   |  0  0  0  0  | 0x00
     9 |  0   |  0  0  0  1  | 0x01
    10 |  0   |  0  0  0  0  | 0x00
    11 |  0   |  0  0  0  1  | 0x01
    12 |  0   |  0  0  0  0  | 0x00
    13 |  0   |  0  0  0  1  | 0x01
    14 |  0   |  0  0  0  0  | 0x00
    15 |  0   |  0  0  0  1  | 0x01
--- End of Captured Frame #2 ---

--- Captured Frame #3 ---
Sample | DRDY | D3 D2 D1 D0 | Hex 
-------|------|-------------|------
     0 |  0   |  0  0  0  0  | 0x00
     1 |  0   |  0  0  0  1  | 0x01
     2 |  0   |  0  0  0  0  | 0x00
     3 |  0   |  0  0  0  1  | 0x01
     4 |  0   |  0  0  0  0  | 0x00
     5 |  0   |  0  0  0  1  | 0x01
     6 |  0   |  0  0  0  0  | 0x00
     7 |  0   |  0  0  0  1  | 0x01
     8 |  0   |  0  0  0  0  | 0x00
     9 |  0   |  0  0  0  1  | 0x01
    10 |  0   |  0  0  0  0  | 0x00
    11 |  0   |  0  0  0  1  | 0x01
[... 280 lines cut ...]

    if (ret == ESP_OK) {
        ESP_LOGI(TAG, "RX completed successfully!");
        for (int i = 0; i < NUM_FRAMES_TO_CAPTURE; i++) {
            char title[64];
            sprintf(title, "Captured Frame #%d", i + 1);
            print_deserialized_data2(title, rx_buffer + i * rx_payload_size_bytes, rx_payload_size_bytes, 0);
        }
    } else {
        ESP_LOGE(TAG, "RX FAILED with error: %s", esp_err_to_name(ret));
    }
#endif

    ESP_LOGI(TAG, "Test finished. Halting.");
    vTaskDelete(NULL);
}

```
</details>

## Assumption

I think the Parlio RX hardware state machine did not reset correctly after finished one transaction.

And P4 behaves similar, but slightly different.

demik, 2025-09-13:
Hello,

Just to note that I've met this issue with P4 as well. Setup very similar to @Dimandragon "Case B". No timeout, `eof` always happens. After the first transaction, I'm seeing random delays and/or the `ready` signal is ignored (even if grounded)

The following transactions also seems to read something similar than the first one, and with multiple DMA descriptors used, it looks like the descriptors are mangled as well (but that may be a separate issue)

Tried disable/reenable as well without progress