espressif/esp-idf #17266: esp_transport memory leak on FIN connection from server

damian-kurek-wizzdev, 2025-08-08:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.2 and 5.5

### Espressif SoC revision.

esp32

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32 wroom 

### Power Supply used.

USB

### What is the expected behavior?

No memory leak 

### What is the actual behavior?

Check steps to reproduce

### Steps to reproduce.

I connect to AWS using a TLS connection on port 443 and then establish an MQTT connection with IoT Core. I wanted to test if it works and noticed that I run into throttling when doing it too often. When the throtling server sends me FIN. After that, MQTT is disconnected, and I close esp_transport by:
`    esp_transport_close(pNetworkContext->transport);
    esp_transport_list_destroy(pNetworkContext->transport_list);
`
After closing everything, I noticed a memory leak of around 300 bytes per FIN. I updated the esp-idf and added the SO_LINGER option to socket(I run this code after esp_transport_connect):
`    int socket = esp_transport_get_socket(pNetworkContext->transport);
    if (socket == -1)
    {
        LOG_ERROR("Failed to obtail socket, probably shouldn't happen");
    }
    struct linger lingerOption = {};
    lingerOption.l_onoff = 1; // Enable
    lingerOption.l_linger = 0; // Set time to 0 so it will be instantly deleted

    if (setsockopt(socket, SOL_SOCKET, SO_LINGER, &lingerOption, sizeof(lingerOption)) != 0)
    {
        LOG_ERROR("Failed to set SO_LINGER");
    }` 

It helped a bit, and now I get around 200 bytes of memory leak per fin.

 I set up a test firmware where I connect and disconnect once to clear all initialisation. I then run the following code in the operations loop:

- start heap tracking
- connect disconnect AWS 50 times
- Sleep for 20000 seconds, so everything should be deallocated already
- stop heap tracking 

I attached the heap tracking output. You can see 2 memory leaks, both from LWIP, first 208 bytes, the second 16 bytes

[memory_leak_on_fin_heap_tracking.txt](https://github.com/user-attachments/files/21686385/memory_leak_on_fin_heap_tracking.txt)

### Debug Logs.

```plain

```

damian-kurek-wizzdev, 2025-08-19:
Any update on this? Can I provide some more information or test something?

euripedesrocha, 2025-08-20:
Hi @damian-kurek-wizzdev, thanks for reporting this. Can you share a reproducer? This might be related to how LWIP manages the pcb in such cases. Depending on when you are measuring the memory leaks, it would appear that is a memory leak but is expected behavior of the system. With a reproducer, we'll be able to sort out if this is the case.

damian-kurek-wizzdev, 2025-08-20:
Hmm, I will figure out how to share the code with you, as it is part of a larger proprietary codebase and utilises our cloud, which I don't want to share keys publicly.
I will figure it out and share something with you when I'm ready.
Probably, I will copy the code to another repo and create a new AWS account where I can reproduce this issue. But it will take some time

damian-kurek-wizzdev, 2025-08-21:
No problem, I got the green light to do it. I will provide you reproducer soon. I just started working on it, but I might get interrupted with something (110% I will). I think I can do it in 2 weeks. I will contact you when I have the repo done. 
My DevOps has already set up an AWS account with IoT Core.
It does not look like a timeout issue, since after 2 hours, it was still going up

damian-kurek-wizzdev, 2025-08-28:
Ok, I figured it out.
There is no memory leak on 5.5.0.
I created the reproducer and found that it appears to be a memory leak because I attempt to connect too many times. It gets deallocated after TIME_WAIT = TCP_MSL * 2 (2 minutes), which occurs when I receive a FIN from the server after throttling.
So after I get thorling, I need to wait for more than TCP_MSL to get memory back

closed 2025-08-28 by damian-kurek-wizzdev