espressif/esp-idf #17267: Why is it so hard to OTA an encrypted binary?

mike6789, 2025-08-08:
### Is your feature request related to a problem?

I have enabled secure boot and flash encryption, but I need to OTA to my customers. I would like for this OTA file to be the same binary format that I flash via UART during manufacturing, but it seems like that is not possible.

The OTA expects a plaintext signed binary... I tried to hack the SDK to use raw writes into flash and bypass all of the magic number/header/hash verifications to get around this, but I kept hitting dead ends, no matter what I tried the app would just boot up into the pre-OTA version. I gave up...

I know there is some kind of THIRD encryption method for OTA on top of secure boot and flash encryption, but why is this so complicated???

Has anyone been able to OTA a pre-encrypted binary (with the flash encryption key) without doing the THIRD security method outlined here -- https://github.com/espressif/idf-extra-components/tree/master/esp_encrypted_img

Thank you

mahavirj, 2025-08-12:
@mike6789 

Please see prior discussion from https://github.com/espressif/esp-idf/issues/6172 on this topic. Documentation guide [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/system/esp_https_ota.html#ota-upgrades-with-pre-encrypted-firmware) also highlights requirements for an independent pre-encrypted firmware distribution scheme.

mahavirj, 2025-08-12:
Interim, this PR https://github.com/espressif/esp-idf/pull/7305 may help if you want to re-use flash encryption scheme for OTA. Please check the discussion in this PR, most important point is to manage separate OTA images depending on the flash offset.

Hope this helps!

mike6789, 2025-08-12:
Thank you for the replies @mahavirj ! In fact, by this time I was already able to figure out the RSA encryption with the component example and it works quite well, I will use that instead of pre-encrypting the binary with the flash encryption key -- this way, I can also have individual random flash encryption keys for each device :)