espressif/esp-idf #17419: ESP32-S3 GPIO15 briefly floats during wake from deep sleep when rtc_gpio_hold_en is used to hold it low. v5.4.2

ardugitter, 2025-08-11:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

Arduino IDE

### Power Supply used.

USB

### What is the expected behavior?

There is a clear bug when going from Arduino ESP32 library 3.2.0 to 3.2.1 which changes ESP-IDF version from 5.4.1 to 5.4.2.

This change means that GPIO15 glitches to a floating state during wake from deep sleep, even when specifically set to low using these just before entering deep sleep:

digitalWrite(GPIO15, LOW);
rtc_gpio_hold_en(GPIO15);

GPIO16 and GPIO 17 do not glitch to floating during wake from deep sleep. The bug is only GPIO15 and it only appears when going from Arduino ESP32 library 3.2.0 to 3.2.1, i.e. ESP-IDF version from 5.4.1 to 5.4.2.

### What is the actual behavior?

GPIO15 floats for <100ms during wake from deep sleep.

### Steps to reproduce.

Run this code to sleep the ESP32-S3:

digitalWrite(15, LOW);
rtc_gpio_hold_en(15);
uint64_t wakeup_mask = (1ULL << 10);
esp_sleep_enable_ext1_wakeup(wakeup_mask, ESP_EXT1_WAKEUP_ANY_HIGH);
esp_deep_sleep_start();

In setup() code (after waking from deep sleep), use this:

pinMode(15, OUTPUT);
digitalWrite(15, LOW);
rtc_gpio_hold_dis((gpio_num_t)15);

### Debug Logs.

```plain

```

atanisoft, 2025-08-11:
This likely would be better filed over on https://github.com/espressif/arduino-esp32/issues as there may be undocumented usages of this IO pin.

ardugitter, 2025-08-11:
Posted here: https://github.com/espressif/arduino-esp32/issues/11712

ardugitter, 2025-08-17:
This is almost certainly an issue in IDF release/v5.4 8ad0d3d8.

The previous release that was good was 2f7dcd86.

esp-wzh, 2025-08-19:
Apologies for the oversight and the delayed response. I have successfully reproduced the issue you reported. Could you please try this patch and confirm whether it resolves the problem? @ardugitter 

[fix-IDFGH-16262.patch.zip](https://github.com/user-attachments/files/21851859/fix-IDFGH-16262.patch.zip)

ardugitter, 2025-10-01:
Hi, when will this be merged into an ESP-IDF release?

The latest release 5.2.6 does not appear to include this fix and neither did the previous one.

Alvin1Zhang, 2025-11-19:
Thanks for reporting, fix on release/5.2 is available at https://github.com/espressif/esp-idf/commit/dc9b6fe9b856affc698ce32905cdd43bb3ad05c1, will update once v5.2.7 releases, thanks.

closed 2025-11-19 by Alvin1Zhang